inertialsense/inertial-sense-sdk #1: What are the options for the dynamic model?

superjax, 2017-05-21:
I'm trying to expose the flash configuration to ROS in my [ROS node](https://github.com/superjax/inertial_sense) and I'm looking at line [1194](https://github.com/inertialsense/InertialSenseSDK/blob/master/src/data_sets.h#L1194) of the SDK.  I can't find any information in the documentation about what values I should put in the `insDynModel`.  Could you clarify what this means?

waltjohnson, 2017-05-24:
We will be updating that comment for that insDynModel with the following
description:

/* INS dynamic platform model.  Determines performance characteristics of
system. 0=PORTABLE, 2=STATIONARY, 3=PEDESTRIAN, 4=AUTOMOTIVE, 5=SEA,
6=AIRBORNE_1G, 7=AIRBORNE_2G, 8=AIRBORNE_4G, 9=WRIST */

Cheers!

*Walt Johnson | CEO & Founder* | *Inertial Sense, LLC*

72 North 720 East | Salem, UT 84653

Office: 801-610-6359 | Mobile: 801-380-0819 | InertialSense.com
<http://www.inertialsense.com/>

Electronic Privacy Notice. This e-mail, and any attachments, contains
information that is, or may be, covered by electronic communications
privacy laws, and is also confidential and proprietary in nature.
Information contained in or attached to this email may be subject to export
controls, may be privileged, and is confidential and competition sensitive.
If you are not the intended recipient, please delete this email and notify
me immediately. You are not authorized to copy or use this email for any
purpose. If you are not the intended recipient, please be advised that you
are legally prohibited from retaining, using, copying, distributing, or
otherwise disclosing this information in any manner. Instead, please reply
to the sender that you have received this communication in error, and then
immediately delete it. Thank you in advance for your cooperation.

superjax, 2017-05-28:
Thanks Walt!  I added it to the node.

closed 2017-05-28 by superjax