inertialsense/inertial-sense-sdk #2: refLla not doing what I expected.

superjax, 2017-05-21:
I tried taking the gps measurement `gps.pos.lla` and setting it as the `refLla`.  My `ins1.ned` went from 

```
x: 4473423.0
y: -12409209.0
z: -1429.51501465
``` 

to 

```
x: 260779056.0
y: 2792222.25
z: 127.074996948
```

By the way, I'm in Provo, UT and I set 

```c

 flash_cfg_.refLla[0] = -40.2571739;
 flash_cfg_.refLla[1] = 111.6738041;
 flash_cfg_.refLla[2] = 1556.59;
```

The altitude did what I expected, but the `x` and `y` NED measurements seem really weird to me.  Any ideas?

waltjohnson, 2017-05-21:
Hi James!  Yes. There is a bug in the current uINS firmware. It is expecting the reference Lat/Lon to be in radians. But it should be in degrees as it is labeled. This will be fixed in the next release we hope to make available in a week or so.

superjax, 2017-05-21:
That did the trick!  Thanks!

closed 2017-05-21 by superjax