inertialsense/inertial-sense-sdk #7: Does Not Build in C++17

lambdaknight, 2018-10-01:
The SDK does not build in C++17. Thankfully, it's an easy fix. std::binary_function does not exist in C++17. So, in C++17, in src/ISDataMappings.h:
`struct sCaseInsensitiveCompare : std::binary_function<std::string, std::string, bool>`
should become
`struct sCaseInsensitiveCompare : std::function<bool(std::string, std::string)>`

and

`struct nocase_compare : public std::binary_function<unsigned char, unsigned char, bool>`
should become
`struct nocase_compare : public std::function<bool(unsigned char, unsigned char)>`

and `#include <functional>` needs to be added to the file.

jjxtra, 2018-10-12:
We are looking at fixing this for next SDK release.

jjxtra, 2019-01-14:
Please re-open this issue if you still see it in the latest release.

closed 2019-01-14 by jjxtra