inertialsense/inertial-sense-sdk #8: No instructions for Linux systems for updating bootloader (release 1.7.1)

chunhochow, 2018-11-08:
The readme says that the bootloader must be updated first before updating the firmware, but it only gives directions for EvalTool which requires Windows. How do I update the bootloader/firmware using Linux (CLTool?)?

orrinjelo, 2018-11-09:
I've found a couple ways to do this in Linux.
One is running EvalTool.exe through wine.  It seems to run fine, but I can't say with complete confidence that it will work as intended (running into some issues that may or may not be related).
The other is implementing your own code to call `bootloadUpdateBootloader()` from the SDK (found in inertiaSenseBootloader.c/h).  
It currently isn't implemented in CLTool, although it could be in the future.

siddirp, 2018-12-04:
In the latest version of the SDK there is a new function bootloadUpdateBootloader() that can be used to update the bootloader as mentioned by @orrinjelo. There is a known issue with this function when used with Linux embedded systems which can prevent the bootloader update to complete properly. We are currently working on a fix as well as implementing a way to update the bootloader within the CLTool.

siddirp, 2019-06-05:
Latest cltool allows users to update bootloader using -ub flag.

closed 2019-06-05 by siddirp