inertialsense/inertial-sense-sdk #9: MIN/MAX_REQUEST_PERIOD_MS vs bc_period_multiple inconsistancy

orrinjelo, 2018-11-15:
This will likely require some verification, so I'll try to explain the best I can here.  
From the InertialSense API code, it would appear that the call BroadcastBinaryData (InertialSense.cpp:545) takes in a broadcast period in milliseconds (periodMS), although the release notes reveal this is not the case.  Yet, we still see comments and variable names that match the old usage.
(IntertialSense.cpp:565-570)
`		for (int i = 0; i < (int)m_comManagerState.serialPorts.size(); i++)
		{
			// [C COMM INSTRUCTION]  3.) Request a specific data set from the uINS.  "periodMs" specifies the interval
			// between broadcasts and "periodMs=0" will disable broadcasts and transmit one single message. 
			comManagerGetData(i, dataId, 0, 0, periodMS);
		}`
Okay, not a big deal.  If we look from the com_manager side, we see that it's actually handled as a periodMultiple (com_manager.c:686):
`/**
*   @brief Request data
*   This function requests the specified data w/ offset and length
*   for partial reads.
*
*	@param[in] dataId       Data structure ID
*	@param[in] offset   Byte offset into data structure.  0 = data start.
*	@param[in] length   Byte length of data.  0 = entire structure.
*	@param[in] periodMultiple Broadcast period of requested data.  0 = single request.
*
*	@return 0 on successful request.  -1 on failure.
*/
void comManagerGetData(int pHandle, uint32_t dataId, int offset, int size, int periodMultiple)
{
	comManagerGetDataInstance(&g_cm, pHandle, dataId, offset, size, periodMultiple);
}`
However, I suspect that it isn't being handled correctly in all places, specifically comManagerGetDataRequest (com_manager.c:1311-1315):
`	// Constrain request broadcast period if necessary
	if (req->bc_period_multiple != 0)
	{
		_LIMIT2(req->bc_period_multiple, MIN_REQUEST_PERIOD_MS, MAX_REQUEST_PERIOD_MS);
	}
`
with
`#define MIN_REQUEST_PERIOD_MS       1               // (ms) 1 KHz
#define MAX_REQUEST_PERIOD_MS       100000          // (ms)`
Why are we comparing the period multiple with the period?

I'm not sure if this is strongly related, but I thought we were able to get messages faster than 10 Hz before in 1.6 (I could be wrong), but it seems here we are limited to 10 Hz.

orrinjelo, 2018-11-15:
I figured out our 10 Hz issue (related to _also setting the `startupNavDtMs` the same way we were doing with 1.6 and before_).  Still there is a lot of confusion with the use of period/stepPeriod/periodMultiple.  But I have enough understanding now to get my code working to the way it was before.

siddirp, 2018-12-04:
@orrinjelo Sorry for not getting back to you sooner. We had several requests by customers to be able to control the rate that data was streamed. Many people were getting data faster than they could handle it. We decided to implement the period multiple to allow users to slow the data rate as needed. Each data set has a base rate and the period multiple allows you receive the data at a slower rate as needed. As you mentioned there are a few configuration variables available to change the base data rate: startupImuDtMs, startupGPSDtMs, and startupNavDtMs.

siddirp, 2018-12-18:
@orrinjelo Thank you for pointing out the discrepancies in the broadcast period and the periodmultiple. This has been fixed and should be more clear in the next software release.

closed 2018-12-18 by siddirp