inertialsense/inertial-sense-sdk #13: Include pre-built cltool.exe with EvalTool installer

XPav, 2019-06-05:
Hi,

I was doing some work in the field yesterday, had EvalTool installed on a different laptop.   I realized I had to change a setting, and couldn't do that with EvalTool, and didn't have cltool.    Had to stop and go back the computer with cltool.   cltool should be included with the binary distribution.

siddirp, 2019-06-05:
@XPav What setting are you referring to? The EvalTool gives access to every setting whereas the CLTool is much more limited. The CLTool is distributed as a source code with the intent that users will customize it or use it as an example for their own development.

XPav, 2019-06-05:
I'm trying to  writing to DID_EVB_FLASH_CFG.  This might be unimplemented, so that's why it looks like writes do nothing?

siddirp, 2019-06-05:
From the Data Sets tab of the EvalTool you have access to every DID. If you go to DID_EVB_FLASH_CFG you can double click on a value and change it. I don't think that DID is implemented in the CLTool.

XPav, 2019-06-05:
DID_EVB_FLASH_CFG values don't stick, unlike DID_FLASH_CFG.  I can change a number, close/reopen EvalTool, and it's back to 0.    I'm going to close this anyway, because I think this is an EVB issue more than a EvalTool/cltool issue.   Thanks for the help.

closed 2019-06-05 by XPav

siddirp, 2019-06-05:
For the information of others with the same question:
The DID_EVB_FLASH_CFG lives on the EVB processor so you need to make sure you are using either the EVB USB port, or one of the serial headers that passes through the com bridge. If you have the EvalTool open and have connected to the EVB2 using the EVB USB port, if you go to DID_EVB_FLASH_CFG you should see the dataset populated. Also, if you push the CONFIG button you can see the top field, cbPreset, increment each time you click CONFIG. You can then change the radio power level to the desired level by just clicking on it and entering a new value.