inertialsense/inertial-sense-sdk #14: Eval tool is not installing properly

Ashik19, 2019-06-25:
Hi,

I have downloaded the latest EVAL tool version. When I am trying to update the firmware it's not working coming up with the following error

[image]

Can anyone please tell me how to resolve this problem?

Thanks

siddirp, 2019-06-25:
@Ashik19 It appears you're trying to update firmware that is not compatible with the current bootloader you have on the device. Follow the instructions here: https://docs.inertialsense.com/user-manual/system-config/system_configuration/#bootloader-update
After that try updating the latest firmware. The bootloader can be found in the 1.7.4 release downloads: https://github.com/inertialsense/InertialSenseSDK/releases/download/1.7.4/uINS-3_firmware_v1.7.4_b290_2019-01-24_163630.zip

siddirp, 2019-09-17:
@Ashik19 Is this still an issue? If so please let me know what questions you have. If not, could you please close this issue?

closed 2019-09-17 by Ashik19