inertialsense/inertial-sense-sdk #15: Wrong units for message description in EvalTools.

wonwon0, 2019-10-23:
[image]

For the angular rates, according to user manual, the angular rates are in rad/s not in degrees/s as shown in the EvalTools message description.

wonwon0, 2019-10-23:
As i understand, the data is displayed right in the GUI, it's just that it is confusing since when the data is recorded, it is in rad/s.

waltjohnson, 2019-10-23:
Those values are converted in the EvalTool from radians to degrees for
convenience.  Radians are more computationally friendly and degrees are
easier to read by the user.

You're best off referring to our online docs
<https://docs.inertialsense.com/user-manual/com-protocol/DID-descriptions/> or
data_sets.h in our SDK
<https://github.com/inertialsense/InertialSenseSDK/blob/master/src/data_sets.h>
for a description of the binary messages.

>

closed 2019-10-23 by wonwon0