inertialsense/inertial-sense-sdk #19: Getting free acceleration

prarobo, 2019-11-18:
Currently, I am using the ros wrapper to get imu data. It appears that the imu linear acceleration data includes acceleration due to gravity.

Is there a way to get free acceleration from the imu? i.e. linear measurements with acceleration due to gravity subtracted?

siddirp, 2019-11-19:
The IMU data is straight from the IMU. You'd need to remove gravity vector
yourself.

*Andrew Priddis | Sr Systems Engineer **|* *Inertial Sense, INC.*

Mobile: (801) 610-6359 | InertialSense.com <http://www.inertialsense.com/>

waltjohnson, 2019-11-19:
You would need to use the EKF attitude to rotate the gravity vector into
the IMU frame and then subtract that from the IMU output.

>

waltjohnson, 2019-11-19:
You can remove gravity from the IMU output using the EKF attitude as
follows:

```
#include "SDK/src/ISPose.h"
Vector gravityNED = { 0, 0, -9.80665 }; // m/s^2
Vector gravityBody;
Vector accMinusGravity;
quatConjRot( gravityBody, DID_INS_2.qn2b, gravityNED );
sub_Vec3_Vec3( accMinusGravity, DID_DUAL_IMU.I[0].acc, gravityBody );
```

>

closed 2020-01-20 by siddirp

andrewpriddis, 2024-04-30:
Upgrade to the GPX-1 for improved GPS accuracy and robust solutions!

Hello  

I'd love for you to take a second and check out the newly released GPX-1. It is a dual GNSS designed module for GPS positioning, GNSS compassing and RTK positioning in a single package. You will no longer need to fit two large receivers in your design. It is an L1/L5 multiband receiver to reject interference, improve fix times, and provide an overall robust solution. The introductory price of an OEM module is $325 with volume pricing available as well. The IG-2 is an integrated module with an IMX-5 and GPX-1 with a price of $788. The RUG4 is in final development and will be released later this year.

GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MVzZpzmKP-VW83vQX_8Hrb-4W4HyYrL5dxhKkN1PJbsT5jGjMW5BWr2F6lZ3kVW382xtx37ypDsW63Jj1s2f1T_7W4ZRV9d8GK3V4W4ZqH0N4VYyR4W5tWTrr7Kt6X2W61n71t3nQwdmW7kKvGP1ZK9F_W1k9TGz2jy32JW7nLfVy8SQ2BFW219PHT85dlCbW6zJDWr5kvkwcW6SNpgN6NV_sTW2KtkgB5sScZzW3fMd126f2VkmW7WbBd_55FMHZW2MzCHP6NhQvkW7TyBkJ7d5YL1W7c1N-r19M-3bMd9nqhL_h7tW8nR4D78-tP9dW2Qw8yd5jggh0W4fjxHm1Q5kb9W3C2pDb2pvBdnW7wNFyT7v7SZ4MYFMWRwxPpyW8-6p6z2H3xcWW9kGT8n5x78qWW7zkhLh4ZcsmHW5tgcYj2Mt-KJW2bj56B8lK3lvW3J1HKT14_0FvW2pkZgG6x9__rW7-zChp9klg7fW7X2hJ37_ZRXGf87TWpl04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MVzZpzmKP-VW83vQX_8Hrb-4W4HyYrL5dxhKkN1PJbsT5jGjMW5BWr2F6lZ3kVW382xtx37ypDsW63Jj1s2f1T_7W4ZRV9d8GK3V4W4ZqH0N4VYyR4W5tWTrr7Kt6X2W61n71t3nQwdmW7kKvGP1ZK9F_W1k9TGz2jy32JW7nLfVy8SQ2BFW219PHT85dlCbW6zJDWr5kvkwcW6SNpgN6NV_sTW2KtkgB5sScZzW3fMd126f2VkmW7WbBd_55FMHZW2MzCHP6NhQvkW7TyBkJ7d5YL1W7c1N-r19M-3bMd9nqhL_h7tW8nR4D78-tP9dW2Qw8yd5jggh0W4fjxHm1Q5kb9W3C2pDb2pvBdnW7wNFyT7v7SZ4MYFMWRwxPpyW8-6p6z2H3xcWW9kGT8n5x78qWW7zkhLh4ZcsmHW5tgcYj2Mt-KJW2bj56B8lK3lvW3J1HKT14_0FvW2pkZgG6x9__rW7-zChp9klg7fW7X2hJ37_ZRXGf87TWpl04 )

When paired with the IMX-5 you have a low cost tactical grade full Inertial Navigation System (INS). The solution can be further enhanced by enabling RTK positioning for cm level accuracy as well as enabling GNSS compassing for high accuracy heading. The IMX-5 is $275 with volume pricing available as well.

IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MVzZpzmKP-VW83vQX_8Hrb-4W4HyYrL5dxhKkN1PJbsz5jGjMW50kH_H6lZ3nSW528bLt5MLNw_W4BYMDq62mFzHW1MZgyw48cSYqW4r_Rvt3dw_HMW667TNt5vDB35W5KlTFV6ddmDxN4CBH2dhSCXhW3hSJ134gblTqW2xzM-r5hMNTqVBwttm7wJtg2W8dxRd97nH8HxW7T8RTH5K43T1N1pYJZgxmhjRW8tnwtD3Bv45lW6Rh4Xp8sbnkLW5Z_t_s7nW2BzW20LTF24NQ13rVLfDzl1TvqK0W8gV5tN3tJt9bW3CxgLG8gHf8VW52L_GV7l4YDmW4z1KWy4yzFJkW5YNrNJ4hwxwSMsnrkhs1kVtW1CTcDJ4-stxnW94RyW36H4prYW3qDGYQ2jR681W6Y7J1F12gb8kW2F8lTD42kPHkW8Qc8v74mCdzSW3NRjlq4LGLQCW8jlbw129XzLdf9gXBK804 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MVzZpzmKP-VW83vQX_8Hrb-4W4HyYrL5dxhKkN1PJbsz5jGjMW50kH_H6lZ3nSW528bLt5MLNw_W4BYMDq62mFzHW1MZgyw48cSYqW4r_Rvt3dw_HMW667TNt5vDB35W5KlTFV6ddmDxN4CBH2dhSCXhW3hSJ134gblTqW2xzM-r5hMNTqVBwttm7wJtg2W8dxRd97nH8HxW7T8RTH5K43T1N1pYJZgxmhjRW8tnwtD3Bv45lW6Rh4Xp8sbnkLW5Z_t_s7nW2BzW20LTF24NQ13rVLfDzl1TvqK0W8gV5tN3tJt9bW3CxgLG8gHf8VW52L_GV7l4YDmW4z1KWy4yzFJkW5YNrNJ4hwxwSMsnrkhs1kVtW1CTcDJ4-stxnW94RyW36H4prYW3qDGYQ2jR681W6Y7J1F12gb8kW2F8lTD42kPHkW8Qc8v74mCdzSW3NRjlq4LGLQCW8jlbw129XzLdf9gXBK804 )

We also still offer our ublox F9P based products that are integrated in systems all over the world.

I'd love to set up a meeting for this week. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MVzZpzmKP-VW83vQX_8Hrb-4W4HyYrL5dxhKkN1PJbv23l5QzW7Y8-PT6lZ3lZVdDxrF3YL9K3W4cJ9SW7qQyzgW7jHKgX49K8kcVdNCvC8cN2xZW1jFpnH6cNv_0W6Cfy1p5SGhp2W8GQw4K3Jpv4LW9lPsq74cs53XW7V4RKR8f3yYTW7vK3rV83RprLW3KgTMt7KjFLjW60z9Q35RwQk2W8X5M7r2D9CXdW8lNdsQ6-KJ3BW4F5PPH7wTxRkW21tSPz3PjXRBW4pxs3T3WGFFdN4CJP19Xt8n0W1ZVTzy2Dbt__W7NR2FP8J5_jbW2RkLDL8FGH0zW4RGrjt1Zx87_W4z92Bp7y6h7QW1WfT9p66RRCbW4Vrv0q6VwbfvW6wXVGz6D6n4lf6bzlLg04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-GCW38wgTx4kdKHJW2Pp0g13XH-G3W47hKR02RTyQgW1_tCNL4tkGfJW3b3N1-2THlyMW45vS7n4kz-kLW2-JQPD4htRjcW2-LY5q3g7v1QW4tw9Fz364xSbW2MB8yg3jnDGHW2vLbXC38lyTCW1ZkYv41Nn-y9W3_v7BQ4tx3RGW25lHBq1ZqsFTW3LMXPH32nqB_W2CWcV12w3ZYyW38rvj23HdhDsW2t1jqq41W19kW1LgYTf2HByy8W1XxZyZ2Y9NYYW38bxZ61Vj8tyW2Ypq-l2zFkCqW3K6Dnv2nTYc8W2vHpy-4hCSWkW2WJNL247p8jcW1LyzWd2HZy0SW3g6JLZ3dygQJW4rb10h3gh12vW45kgtY2zHN6VW4hmVTk3CfydVW36wrGc36jHFbW3gd63Z1Vdqp0W4prrnD3NDJgxW2xLsN241KvbpW2WGnMh1LFXKDW2-HDQn2Pz6HSW3_JN2x2WnNkTW2sJQk_21jc3VW4kbbV84msWwpW3z0SYF2RLxF0W2-dWcV3F4BHDW1-YsC138hx6RW347fLp45Qrd8W2vRCnv4rp9H-W2xLsCb2CX4WGW4rhSX92r0C3hW36yzFs4cRvYZW2w0Ksx3LDpMqW2MzLQw3jlx80W3h_qSv3LG78QW4rGm373bhPP8W2nDHmT2MBzc_f254ZwV04&_hsenc=p2ANqtz-_qGo1NF4CiZV2KrrpYlC7KLKo-3z62F-pDtBx7gVsYQFTcut7rsSgyOwlcn-jB3IR5VVVpN8zFLtt7xkqCtr_kYLIKU76n3WdNPr1PhKgaH28-9VFsLuwRjOM78kTpSybNU7au7sdSxpBjfz_kPxQTG7vZEQ&_hsmi=305054157 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-GCW38wgTx4kdKHJW2Pp0g13XH-G3W47hKR02RTyQgW1_tCNL4tkGfJW3b3N1-2THlyMW45vS7n4kz-kLW2-JQPD4htRjcW2-LY5q3g7v1QW4tw9Fz364xSbW2MB8yg3jnDGHW2vLbXC38lyTCW1ZkYv41Nn-y9W3_v7BQ4tx3RGW25lHBq1ZqsFTW3LMXPH32nqB_W2CWcV12w3ZYyW38rvj23HdhDsW2t1jqq41W19kW1LgYTf2HByy8W1XxZyZ2Y9NYYW38bxZ61Vj8tyW2Ypq-l2zFkCqW3K6Dnv2nTYc8W2vHpy-4hCSWkW2WJNL247p8jcW1LyzWd2HZy0SW3g6JLZ3dygQJW4rb10h3g [... cut ...] 6VW4hmVTk3CfydVW36wrGc36jHFbW3gd63Z1Vdqp0W4prrnD3NDJgxW2xLsN241KvbpW2WGnMh1LFXKDW2-HDQn2Pz6HSW3_JN2x2WnNkTW2sJQk_21jc3VW4kbbV84msWwpW3z0SYF2RLxF0W2-dWcV3F4BHDW1-YsC138hx6RW347fLp45Qrd8W2vRCnv4rp9H-W2xLsCb2CX4WGW4rhSX92r0C3hW36yzFs4cRvYZW2w0Ksx3LDpMqW2MzLQw3jlx80W3h_qSv3LG78QW4rGm373bhPP8W2nDHmT2MBzc_f254ZwV04&_hsenc=p2ANqtz-_qGo1NF4CiZV2KrrpYlC7KLKo-3z62F-pDtBx7gVsYQFTcut7rsSgyOwlcn-jB3IR5VVVpN8zFLtt7xkqCtr_kYLIKU76n3WdNPr1PhKgaH28-9VFsLuwRjOM78kTpSybNU7au7sdSxpBjfz_kPxQTG7vZEQ&_hsmi=305054157 )

andrewpriddis, 2025-01-02:
IK-1: For rapid evaluation and development.

 

We heard your feedback that evaluating and developing using an IMX-5 or GPX-1 module can be tedious. The parts are designed to be surface mounted for deep integration into a system. This makes accessing pins for evaluation and prototyping difficult. To address this concern we developed the IK-1. It has the following features to ease the initial interfacing:

- 0.1" pitch header for convenient interface.
- Interfaces with standard breadboard.
- Onboard Micro USB connector
- Onboard voltage regulation.
- Dual U.FL connectors for GPX GNSS antennas

There is a version for working with the IMX-5 as well as a version to work with the GPX-1

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllY3l5QzW8wLKSR6lZ3nJW6jmrDj6cD6XFW6BzkHX8RGD7TW4nRLQg3XlqBfW8BlYks83G-HVW3gR6C45xJy9KW6fr96v79RSqZW6TrHyh8cDK_hW39s6T01DNH68W89bFCQ11DsKxW2Zp3zh6z3WnsW1gJWfh2G23M1W1-JNkH6rkqrnW7g6vrx3syjdsW6-qh1d5YG9PKW4M-_xl71RTflW1c_BV-779kDTN27SQcgK4pGKW64gv918N8WFwW164l2-5zks7YW5zgp775YVjF4W7r5-Pb63p9SrN8_JYHp1lpByW4snPdW1QhkspW1nqzly6KtGh6W8FxkJT8h59w_MJYwXSk7m1vN86MWXvLhPmsW4n_Xml2hfV3wf1h_SQl04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllY3l5QzW8wLKSR6lZ3nJW6jmrDj6cD6XFW6BzkHX8RGD7TW4nRLQg3XlqBfW8BlYks83G-HVW3gR6C45xJy9KW6fr96v79RSqZW6TrHyh8cDK_hW39s6T01DNH68W89bFCQ11DsKxW2Zp3zh6z3WnsW1gJWfh2G23M1W1-JNkH6rkqrnW7g6vrx3syjdsW6-qh1d5YG9PKW4M-_xl71RTflW1c_BV-779kDTN27SQcgK4pGKW64gv918N8WFwW164l2-5zks7YW5zgp775YVjF4W7r5-Pb63p9SrN8_JYHp1lpByW4snPdW1QhkspW1nqzly6KtGh6W8FxkJT8h59w_MJYwXSk7m1vN86MWXvLhPmsW4n_Xml2hfV3wf1h_SQl04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllY3l5QzW8wLKSR6lZ3mLW5jz5JF2h-jJNW28DbTF2lTMpsV3Qsg08pttnmW5rR6--7Yk-wRW8hPfLb8RjKjDW4n1nSk1kXNt6W1_Zx826pkKzkW31bkXK6J_PWYN1G93jNpGryWW93mD2m5H21JTW3GRkBH1Fcc65W1B348R1H67rYW4-tG866zsz1GW22qp6S25hFYQV76XTR6wfH-dW5mSKPm4L8-4xW3ZK_3-3kCm4QW9c8lDH6PSzgxW2MKnLl5ddyRsW51QMjV4fFkxbW7gKhfP2JKF5LN4VgXfT6vcjSMfpcCMhYh0pW5dbMXG6w3mXGW21NmF-2G89h-W6rwhR17LDFd9W6qrYqD7cRRmYW30VYbX8t2MgMdv0BFs04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllY3l5QzW8wLKSR6lZ3mLW5jz5JF2h-jJNW28DbTF2lTMpsV3Qsg08pttnmW5rR6--7Yk-wRW8hPfLb8RjKjDW4n1nSk1kXNt6W1_Zx826pkKzkW31bkXK6J_PWYN1G93jNpGryWW93mD2m5H21JTW3GRkBH1Fcc65W1B348R1H67rYW4-tG866zsz1GW22qp6S25hFYQV76XTR6wfH-dW5mSKPm4L8-4xW3ZK_3-3kCm4QW9c8lDH6PSzgxW2MKnLl5ddyRsW51QMjV4fFkxbW7gKhfP2JKF5LN4VgXfT6vcjSMfpcCMhYh0pW5dbMXG6w3mXGW21NmF-2G89h-W6rwhR17LDFd9W6qrYqD7cRRmYW30VYbX8t2MgMdv0BFs04 )

Your work with the breakout boards is your first step to integrating  both units into a low cost INS solution. Be sure to review the datasheets:

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNlk85jGjMW50kH_H6lZ3pfW3YMS8z2Mng7TW7VmgLr5fnwLtW1rBTj51bR1lMW5X19Gb52xrS9W4YHL5W91jPdcW2W8DjY14jh0tVgHP9T1vvsQ1W7kB-Bh2grnr2W2TdktJ7QgSGDW6wzSvR4yftK_W8BqZKG92YhRXW6R2Z105sPFr6W7rHJwr361cxXW1BhS-w4kqVYkW3HZwfm6134LQVkTHYH4hVwT2W2QcgJ-7cxfQgW4R8V1m4Hx09vW7xtPXT8yJ84QW4Llzwb2f6cwtW7Q8Jq05wSHyJW71kNHC7TvMGbW1PSj8Y5KkPd7W6CL7gz2_X4ScW2JHtd-6MTmq4W2JykCq2_jF0rW7yj_sT862rsFW1L953K90gT17W8BXH__3nq5z-W3zPD6W5ZTChpN4x3qJt827tLW3YtY3N3Ll8_wf8vNjyP04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNlk85jGjMW50kH_H6lZ3pfW3YMS8z2Mng7TW7VmgLr5fnwLtW1rBTj51bR1lMW5X19Gb52xrS9W4YHL5W91jPdcW2W8DjY14jh0tVgHP9T1vvsQ1W7kB-Bh2grnr2W2TdktJ7QgSGDW6wzSvR4yftK_W8BqZKG92YhRXW6R2Z105sPFr6W7rHJwr361cxXW1BhS-w4kqVYkW3HZwfm6134LQVkTHYH4hVwT2W2QcgJ-7cxfQgW4R8V1m4Hx09vW7xtPXT8yJ84QW4Llzwb2f6cwtW7Q8Jq05wSHyJW71kNHC7TvMGbW1PSj8Y5KkPd7W6CL7gz2_X4ScW2JHtd-6MTmq4W2JykCq2_jF0rW7yj_sT862rsFW1L953K90gT17W8BXH__3nq5z-W3zPD6W5ZTChpN4x3qJt827tLW3YtY3N3Ll8_wf8vNjyP04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllY3l5QzW8wLKSR6lZ3mLW5jz5JF2h-jJNW28DbTF2lTMpsV3Qsg08pttnmW5rR6--7Yk-wRW8hPfLb8RjKjDW4n1nSk1kXNt6W1_Zx826pkKzkW31bkXK6J_PWYN1G93jNpGryWW93mD2m5H21JTW3GRkBH1Fcc65W1B348R1H67rYW4-tG866zsz1GW22qp6S25hFYQV76XTR6wfH-dW5mSKPm4L8-4xW3ZK_3-3kCm4QW9c8lDH6PSzgxW2MKnLl5ddyRsW51QMjV4fFkxbW7gKhfP2JKF5LN4VgXfT6vcjSMfpcCMhYh0pW5dbMXG6w3mXGW21NmF-2G89h-W6rwhR17LDFd9W6qrYqD7cRRmYW30VYbX8t2MgMdv0BFs04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNlks5jGjMW5BWr2F6lZ3nlW4ZBVGj8Z5g4FW43wMZ87dX31kW51gMD22sdYD5V6L6Vp7FFSpHW77BvlV5kw5FpW3500yg9kP-LKW6c716M6fSHP8W7wyK371c3_lSW90Sg3-8RcFvRMDfKdPQlng5W1Fl-k18Ss1C8W6nqnb09jRzCRVVdG7P8V2Q4TW48jd4l4wZFJNW6JNG1L8Hx2WgW5mqM-p7P5vLsN5t_wlFyy00QW6F1syw2blQ4NW2Y9sGq3Hcml5W71L34t1XsDW2W5NrQPv49XGS_W2sSYWq4NJnj2M9Ly9yHzgDMW14BQZ99kXdtgW2C-brp6Fq02sW18rthb90s1d-W2wS1rN1yxnb8W1wCrVY6PfvS2W46hzbR4CsZRhW3N0jSz838N7hW22J9BT5_LYTkW7FqzjV3M1jJGW4JY4JG9k3607W3B3cLH4HpLPtf90WYYd04 )

I'd love to set up a meeting in January. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

***@***.***

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW9HZ196YqQWW5BlCQq7WZcwSW3yn_R_5qkpWyMkNllF3l5QzW7Y8-PT6lZ3nDW7b9ypc2vMb_JW4tBbJy88DgH5W6XrJKN4VzHJgVq1d2K7PqgNjW9lPHMf8Bk3sHW8CKfr_1Z75ffW47tRFp1hJRSKW16C7dF6sZHhlW3QGsXV4hx8ccW8QJhYV5Jc5xrW5ywDXC13ys5PW3zK7B96-7vmpW267r286hD1h_W4ZrQ6q47yJq6W2g0xrt6rkVbGVSsXSG3G [... cut ...] 9GW2CCTYC2xZywNW3P5XDk49TM_zW1BgzQ9329YwtW2CZxrq4pG72GW4ktzNg2PpmMxW3NVSX-32xHWFW41zBlw45BQZwW1QcFmd3XN7v1W2C-nwQ4tdGSwW20ZrHv2CCrB-W45Gxsy1-_jXFW3S_81S1NxZfTW2RLv014hy7nTW3XJN8p2vBp-4W32CsCp1Zcf75W30yjnq38613pW4tvqc01Xn7trW1Sxtdx20VJwtW4pDY4P3g0nzJW2WjDhN4fh-R3W3jl_fk3DMGkJW2x_z9W3JKcDWW3dwGX934vVswf30BNym04&_hsenc=p2ANqtz-8_I1ul-zOnN7kLnIrbMaI-vkLOQX4_xj7_tUzR61sQlcX2iUcQWd8UISCEZx3gYQ7phVnwTWFN6X4ChOxIUtSVnMJTQvwKvGc8oO_abW338YkoYXHEOmFIh5hUjFY3hwYjPNuUxtm_cY_Y4Et9Pjem7VhmgA&_hsmi=340288392 )

andrewpriddis, 2025-06-27:
Order now to avoid tariff-induced price hikes in July!

Hello  ,

Hope you’ve been doing well! I wanted to give you a quick heads-up — especially if you're planning any new builds or procurement before Q3.

Due to rising tariffs and component costs, Inertial Sense pricing will increase in July. But if you order before June 30, you can lock in our current 2024 pricing on key products like the IMX-5 and GPX-1.

Whether you're sourcing for:

✔️ Defense UAVs, loitering munitions, or targeting systems

✔️ SATCOM, radar, or LEO antenna platforms

✔️ Or just need rugged, U.S.-made tactical-grade INS/IMU performance —

Now is a great time to stock up or move forward on your project!

📦 Explore Our Top Products:

Have Questions? I’m Happy to Help:

📞 Call me at 801-610–6359

💬 Chat with me (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXd8K54WzVMMW3vN6vG7P-p5SVtG5yr5yj-1KN4xH9fx3prCCW95jsWP6lZ3kHVr2hv461CsphW93xykj7MxNjwW1PYDbK11QjZzW3vx1dR65x6y1W67hq8p1yFtcvW1KrLrZ91qrJHVqjFkh4Y30swVJqvHd2hxyFcW5SXG-L47p90dW8JbSjW7_XTg2W39xxc44-CVwKW35L5r86vlwD1N1xF3J7xy2QmMFZ5s860fmMW8c3Yhw39qs6rW1NFdXW7W9Y7zW7VWTYq53KHYjW4fWcsm96FzZHW63zZJp9kRfWsW54cpT24B6w5sW3hV7GL4vY23CW1rj30m2lpq_wW2GPZFt3hL4bDW3FgnJ41yBV8DVmgZ3553TjgwN84mKtYGQs3HW1T-Vzq4sQh1mW25MVNJ7XFzsDW7z6scJ6YmP1_W7--TQL8SWzDjdywqKd04 ) live on our website

👉 Or place your order directly: inertialsense.com/shop (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXd8K54WzVMMW3vN6vG7P-p5SVtG5yr5yj-1KN4xH9cs5m_5PW50kH_H6lZ3nWW6tNnrG2BSxKnW7PlmsS6_hkH7W92LHY94K6t9kW46mPZs6p1C16W8BVSxB4Q77TxW2nnWHN1KX-7TW3YbN3Z4r7nLGW6qYM1F75ZpsSW37Wntw2stwKVVqzHy53-HX6mN67Rq3_xLN_yW4dDKsH7WKZX1W3znJ0j83w1kJW19FDQy3w0fjmW34m_PV5GjwMHW1JRNzY7P6DsNVhLqTs5fr1DVN6LmMTlyM9BHW2h3Q1K3nVNfSV1mg_z49xJTcW67tkY560V6qjW42pz__7by1X4W6MlnKr706y0WN1bR5VtzKClbW7HPldY9df8JCMb60b7MLR78W5ghXK26dgmPVW2J5FvF2DXz6lN7rpcbn9XBbgW4NsrZc5c2w7HW2dl7zd5csRMbW2fNmyH3TPQmzdnJnpb04 )

📧 Or just reply to this email — I’ll get back to you personally.

Thanks again for working with us — and if there's anything I can do to support your mission, don’t hesitate to reach out.

Best regards,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXd8K54WzVMMW3vN6vG7P-p5SVtG5yr5yj-1KN4xH9fx3prCCW95jsWP6lZ3mmW2pjLbN7lgLgrW1nv7xC9m0pCbN1wkKJL4kMs0V34CZJ5ZTjxVW8x3mkl5msv3PW32Xg2V2CcH2XW9j558s11x9CwW3mJCMh9kW9vCN4RXys5gXYlQW31G8J52CgCZ9W8pWD9246JvRQW861rzY2-GH5WW3TXm0S464Md0W5rXrYM5jl2K3W4yJN-48y56s6W6kHtn16rGJtZW1bkQpX5F6nbNW3yFgPD3qW2k4W9hW38N5FH_7QW5DD74v3N5clTW1MFvvs19J6F6W3rbmh97Flx4WW9cQHTK1scxB8W1J3Qmn3P4LkJW41nnvS4GhLRKW2Msv-s47Bqj4W2CgclD1m8KBtN12jPtJtdNRrW7PCt-44nZQzxVlQ1xf1vn5lbdjM2RW04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-zQW1Bwnpx49QWGpW2vBpN-1SfTZfW1QccJz2vSWPlW1ZbRNs2CMfBGW1YYKhz2qD21pW30KW8p3bbRM8W3QxVMQ1L67LGW3C9NLX3bmbftW1_8yMs3zdxgVW3XMlp51LrjmsW34GCDw43WfK5W47B8Fd3z8MCKW4kdhM-3blNNyW212SMV41yKc2W3Vxtf_1Nls1zW3C7N7h1Nvn-8W2-L3RT3SXXMTW47m2vV4chN8TW3CglP23f_-9XW3NRVPQ41Y3TWW30r25H2zvK_YW23nZSj3dyPWYW4hGnrm34gjM4W4pCKdJ3VHKJvW36dq0-2KNR18W2YKw1k3P0tl2W1N3-CB2vYWG-W4tkDwM3VYs0PW3Cbfn73LYqY8W2r4lqX1_5-lLW41rrHN3bjCMrW45zZ3s328KzHW38598d3gbSxfW3jkLt91Z8KQxW1Q2XxF2Yn7NCW23hfm02RrjnPW2RQblL4rqWB4W2PxT6w1W-YvqW3ZNPJw21qvhWW43xdnQ43BZ8jW2Ws_2D3K7-ldW309mVm4kBSygW3z8JK13yN6szW3djWmp2TdFywW3F9dG73hYS-RW3dbdCV38jqGzW2RxjJK1Bhj5PW1Z0l3c1Bw2vYW3Q_Ryt41_0ZKW2G0d3N2HWFNcW41K1S-4cQJwbW3Sy31J1S3XTwf49LfGh04&_hsenc=p2ANqtz-90tMPxWhVrajsmdjudblMCyU2YzzsLrdVxxaJ8pagRSK6nU4rakSKxU4RER_rRgqhBwHxAnmmdNAAsoVn0vinXbN0naHyTaEu11vQWURipZ5CmadyhuQt8HNAuqPeuBJnOHazPfykVeix9d2qWzy4aSoM11g&_hsmi=368897815 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-zQW1Bwnpx49QWGpW2vBpN-1SfTZfW1QccJz2vSWPlW1ZbRNs2CMfBGW1YYKhz2qD21pW30KW8p3bbRM8W3QxVMQ1L67LGW3C9NLX3bmbftW1_8yMs3zdxgVW3XMlp51LrjmsW34GCDw43WfK5W47B8Fd3z8MCKW4kdhM-3blNNyW212SMV41yKc2W3Vxtf_1Nls1zW3C7N7h1Nvn-8W2-L3RT3SXXMTW47m2vV4chN8TW3CglP23f_-9XW3NRVPQ41Y3TWW30r25H2zvK_YW23nZSj3dyPWYW4hGnrm34gjM4W4pCKdJ3VHKJvW36dq0-2KNR18W2YKw1k3P0tl2W1N3-CB2vYWG-W4tkDwM3VYs0PW3Cbfn73LYqY8W2r4lqX1_5-lLW41rrHN3bjCMrW45zZ3s328KzHW38598d3gbSxfW3jkLt91Z8KQxW1Q2XxF2Yn7NCW23hfm02RrjnPW2RQblL4rqWB4W2PxT6w1W-YvqW3ZNPJw21qvhWW43xdnQ43BZ8jW2Ws_2D3K7-ldW309mVm4kBSygW3z8JK13yN6szW3djWmp2TdFywW3F9dG73hYS-RW3dbdCV38jqGzW2RxjJK1Bhj5PW1Z0l3c1Bw2vYW3Q_Ryt41_0ZKW2G0d3N2HWFNcW41K1S-4cQJwbW3Sy31J1S3XTwf49LfGh04&_hsenc=p2ANqtz-90tMPxWhVrajsmdjudblMCyU2YzzsLrdVxxaJ8pagRSK6nU4rakSKxU4RER_rRgqhBwHxAnmmdNAAsoVn0vinXbN0naHyTaEu11vQWURipZ5CmadyhuQt8HNAuqPeuBJnOHazPfykVeix9d2qWzy4aSoM11g&_hsmi=368897815 )