inertialsense/inertial-sense-sdk #20: Need access to DID_INL2_STATUS but inl2_status_t struct is not defined.

Atlis, 2019-12-06:
Hello,

I need access to `DID_INL2_STATUS`, but `inl2_status_t` struct is not defined in the SDK. Therefore, I can't map the buffer onto the struct.
Is it possible to provide the `inl2_status_t` struct?

Thanks!

waltjohnson, 2019-12-07:
I hope this helps.

typedef struct PACKED
{
int ahrs;
int zero_accel;
int zero_angrate;
int accel_motion;
int rot_motion;
int zero_vel;
int ahrs_gps_cnt; // Counter of sequential valid GPS data (for switching
from AHRS to navigation)
float att_err;
int att_coarse; // Flag whether initial attitude error converged
int att_aligned; // Flag whether initial attitude error converged
int att_aligning;
int start_proc_done; // Cold/hot start procedure completed
int mag_cal_good;
int mag_cal_done;
int stat_magfield;
} inl2_status_t;

Atlis, 2019-12-07:
Thanks Walt!
That's what I needed!

closed 2019-12-07 by Atlis