inertialsense/inertial-sense-sdk #21: integration with pixhawk

Ador2, 2020-01-14:
I connected uINS with pixhawk2 Cube on the GPS1 port but it doesn't detect it, I've clone inertialsnese branch of px4 firmware too.
is there a way to connect uINS to pixhawk??

siddirp, 2020-01-20:
We don't officially support the pixhawk at this time. Work was done to create Beta firmware in the past, but because of lack of interest it was dropped. Only GPS position data was consumable by the Pixhawk in the past. Current users implement their own bridge from uINS to GPS input to the pixhawk.

Ador2, 2020-01-21:
well, what about arduino??

siddirp, 2020-01-21:
We have an application note that describes interfacing with an Arduino: https://docs.inertialsense.com/user-manual/SDK/CommunicationsArduino/

Ador2, 2020-01-22:
actually I tried that as described and with Arduino due, but it does not work, Arduino due  outputs nothing on serial monitor.
I executed ImportSdkFiles.sh, here is the code I used:
```
/*
  Read InertialSense
  This is a basic example of how to read data from the uINS using the InertialSense serial protocol
  The circuit:
  - inertialSense port H4 attached to Serial port 1
  - Serial Monitor open on Serial port 0
  created 26 Jan 2018
  by James Jackson
  This example code is in the public domain.
*/

#include "src/ISsdk/ISComm.h"   // Run ImportSdkFiles(.bat or .sh) to copy needed SDK files into this project.
#include <stddef.h>

// you can make this 512 size if memory is really tight
static uint8_t s_buffer[1024];
static is_comm_instance_t comm;

static void handleINSMessage(ins_1_t* ins)
{
  SerialUSB.print("Lat: ");
  SerialUSB.print((float)ins->lla[0], 6);
  SerialUSB.print("\t");
  SerialUSB.print(", Lon: ");
  SerialUSB.print((float)ins->lla[1], 6);
  SerialUSB.print("\t");
  SerialUSB.print(", Alt: ");
  SerialUSB.print((float)ins->lla[2], 2);
  SerialUSB.print("\t");
  SerialUSB.print(", roll: ");
  SerialUSB.print(ins->theta[0]*C_RAD2DEG_F);
  SerialUSB.print("\t");
  SerialUSB.print(", pitch: ");
  SerialUSB.print(ins->theta[1]*C_RAD2DEG_F);
  SerialUSB.print("\t");
  SerialUSB.print(", yaw: ");
  SerialUSB.print("\t");
  SerialUSB.println(ins->theta[2]*C_RAD2DEG_F);
}

void setup() {
  // initialize both serial ports:
  SerialUSB.begin(115200);
  Serial1.begin(115200);
 
  if(sizeof(double) != 8) {
    SerialUSB.println("Inertial Sense SDK requires 64 bit double support");
    while(true) {};
  }

  SerialUSB.println("initializing");

  // This buffer is going to be used to hold messages
  // as they come in
  comm.buffer = s_buffer;
  comm.bufferSize = sizeof(s_buffer);

  // initialize comm interface - call this before doing any comm functions
  is_comm_init(&comm);

  // Stop all the broadcasts on the device
  int messageSize = is_comm_stop_broadcasts_all_ports(&comm);
  Serial1.write(comm.buffer, messageSize); // Transmit the message to the inertialsense device

  // Ask for ins_1 message 20 times per second.  Ask for the whole thing, so
  // set 0's for the offset and size
  messageSize = is_comm_get_data(&comm, DID_INS_1, 0, sizeof(ins_1_t), 50);
  Serial1.write(comm.buffer, messageSize); // Transmit the message to the inertialsense device

}

void loop() {
  
  // read from port 1, and see if we have a complete inertialsense packet
  if (Serial1.available()) {
    uint8_t inByte = Serial1.read();

    // This function returns the DID of the message that was just parsed, we can then point the buffer to
    // The right function to handle the message.  We can use a cast to interpret the s_buffer as the
    // kind of message that we received
    uint32_t message_type = is_comm_parse(&comm, inByte);
    switch (message_type)
    {
      case DID_NULL:
        break;
      case DID_INS_1:
        handleINSMessage((ins_1_t*)(comm.buffer));
        break;
      default:
        SerialUSB.print("Got an unexpected message DID: ");
        SerialUSB.println(message_type, DEC);
    }
  }
}
```

siddirp, 2020-01-22:
I’ll check it out. I ran it recently but there may have been SDK updates
since then.

Ador2, 2020-01-22:
Thanks Andrew!
I'm waiting for your comment.

siddirp, 2020-01-24:
Hi Ador,
I just tested the IS Arduino example project successfully. It seems the
difference between the code I ran and your code is the USB serial port you
use to read out data. Have you made sure that the baud rate of the uINS is
the same as the Arduino baud rate?

*Andrew Priddis | Sr Systems Engineer **|* *Inertial Sense, INC.*

Mobile: (801) 610-6359 | InertialSense.com <http://www.inertialsense.com/>

Ador2, 2020-01-25:
Hello Andrew,
yes I made sure that the baud is the same, could you shared your schematic and the SMD jumper for the uINS? what serial/pins did you used from uINS?

siddirp, 2020-01-28:
I used the instructions in the online documentation here:
https://docs.inertialsense.com/user-manual/SDK/CommunicationsArduino/

*Andrew Priddis | Sr Systems Engineer **|* *Inertial Sense, INC.*

Mobile: (801) 610-6359 | InertialSense.com <http://www.inertialsense.com/>

Ador2, 2020-01-31:
This is odd, i tried the same thing but it doesn't work, I've noticed in the doc you're using another uNIS I'm using rugged uINS 
[image]

siddirp, 2020-01-31:
It should work the same. You’ll just need to connect the serial pins to the
serial pins of the Arduino. The default configuration is RS232 so if you
want to use TTL levels to connect rugged directly to the Arduino you’ll
need to either use an adapter or reconfigure the rugged to TTL levels.
https://docs.inertialsense.com/user-manual/hardware/rugged1/#jumpers

closed 2020-03-06 by Ador2