inertialsense/inertial-sense-sdk #24: Documentation on DID_WHEEL_CONFIG and DID_WHEEL_ENCODER is incomplete

mjshiggins, 2020-05-07:
While there are descriptions of these two objects on the SDK DID page, some important information is missing:

**DID_WHEEL_CONFIG**
1. bits - Config bits / eWheelCfgBits are not really explained, and the modes indicated by the enums (WHEEL_CFG_BITS_ENABLE_KINEMATIC_CONST & WHEEL_CFG_BITS_ENABLE_ENCODER & WHEEL_CFG_BITS_ENABLE_MASK) aren't clear enough to be self-explanatory
2. Wheel frame: While it's clear that the translation and rotation need to be set from the IMU to the left wheel frame, it's not entirely clear what the wheel frame is (nor is it standard).
3. Distance / Diameter: Are these assumed to be in meters?

**DID_WHEEL_ENCODER**
1. status: What is the status word supposed to be? No other parts of the API use something like this, so it's not possible to know how to set this.

siddirp, 2020-05-11:
Wheel encoder input is currently not supported. The data sets are in place, but the full implementation hasn't been built out.

closed 2020-05-11 by siddirp

mjshiggins, 2020-05-13:
@siddirp I would suggest that this is then made clear in the documentation. As it reads, this feature seems active.

siddirp, 2020-05-14:
Thanks for the suggestion. We have adjusted the documentation.