inertialsense/inertial-sense-sdk #25: RTK via NTRIP / CLTool not working

mjshiggins, 2020-05-13:
Following the documentation, I cannot initiate RTK mode on the uINS using the same NTRIP server (our exact username and pw are omitted). We have been granted access to this server, and I can confirm using other tools like BNC that our login details allow us to receive RTK corrections. Additionally, this was tested with the INS outside with a clear view of the sky for 30 minutes, with no change.

Using the following command results in immediate failure due to out of date documentation (no -msgPPD option exists):

`./cltool -c=/dev/ttyUSB0 -flashConfig=rtkCfgBits=0x01 -baud=460800 -svr=RTCM3:rtgpsout.unavco.org:2101:P213_RTCM3:OMITTED_USERNAME:OMITTED_PASSWORD -msgPPD`

Updating the command to use `-msgPresetPPD` doesn't seem to show any relevant RTK information, but it does show a successful flash. Both `-msgRtkPos` and `-msgRtkRel` remain blank. I have used the ROS driver as well, and can confirm that no RTK related messages are being sent back from the unit, so it's not really possible for me to understand the status / what's going wrong. The only feedback that I receive, as far as I can tell, is a successful connection message (which does fail if I try to connect to a non-ntrip server, and interestingly also returns a success if I don't pass in the username and password):

`[INFO] [ins.inertial_sense]: Successfully connected to RTCM3:rtgpsout.unavco.org:2101 RTK server
`

I'm wondering if we are missing some undocumented requirements for getting an RTK fix or consuming that information. Please let me know.

This is retrieved using the CLTool:
`Current flash config
cBrdConfig = 0
checksum = 3616527607
gps1AntOffset[0] = 0
gps1AntOffset[1] = -0.150000006
gps1AntOffset[2] = -1.29999995
gps2AntOffset[0] = 0
gps2AntOffset[1] = 0.5
gps2AntOffset[2] = -1.29999995
gpsMinimumElevation = 0.261799395
gpsTimeSyncPeriodMs = 1000
insDynModel = 4
insOffset[0] = 0
insOffset[1] = 0
insOffset[2] = 0
insRotation[0] = 0
insRotation[1] = 0
insRotation[2] = 0
ioConfig = 0
key = 24
lastLla[0] = OMITTED
lastLla[1] = OMITTED
lastLla[2] = OMITTED
lastLlaTimeOfWeekMs = 247154000
lastLlaUpdateDistance = 1000
lastLlaWeek = 2098
magDeclination = 0.230414361
magInclination = 1.063622
refLla[0] = OMITTED
refLla[1] = OMITTED
refLla[2] = 5
RTKCfgBits = 1
sensorConfig = 15
ser0BaudRate = 460800
ser1BaudRate = 921600
size = 240
startupGPSDtMs = 200
startupImuDtMs = 1
startupNavDtMs = 10
sysCfgBits = 258
wheelConfig.bits = 0
wheelConfig.diameter = 0.720000029
wheelConfig.distance = 1.79999995
wheelConfig.e_i2l[0] = 0
wheelConfig.e_i2l[1] = 0
wheelConfig.e_i2l[2] = 0
wheelConfig.t_i2l[0] = -0.100000001
wheelConfig.t_i2l[1] = -0.649999976
wheelConfig.t_i2l[2] = 0
zeroVelOffset[0] = 0
zeroVelOffset[1] = 0
zeroVelOffset[2] = 0
zeroVelRotation[0] = 0
zeroVelRotation[1] = 0
zeroVelRotation[2] = 0`

mjshiggins, 2020-05-14:
More info from the GPS tool. Looks like it might still be initializing, even though it has a 3D fix? Is there any way to tell if it's processing corrections?

DID_GPS1_POS: 355819200ms
	Sats: 19,  Status: 0x401f0313 (3D)
GPS_STATUS_INIT_STATUS_READING 	0x40000000
GPS_STATUS_FLAGS_RTK_POSITION_ENABLED 	0x00100000
Unknown
GPS_STATUS_FIX_3D 	0x00000300
GPS_STATUS_NUM_SATS_USED_MASK 	0x000000FF

siddirp, 2020-05-14:
Once any base correction data is received, the RTK_REL data will be displayed. Until then, there is nothing to stream. The CLTool is provided in the SDK to be modified as needed. It may be helpful to create a more verbose output of the information you are looking for. Have you tried using the EvalTool? It is a good way for users to get started as well as get familiar with the behavior of the uINS. It's easier to see the status of connections. When corrections are processing you'll see a field call age of differential. It counts the time since last packet was received from the base. If that value continually resets you'll know they are being processed.

mjshiggins, 2020-05-14:
@siddirp Thanks for the response. I am currently unable to use the evaltool (remote hardware), but is it correct to say that it still wouldn't tell me any info if zero corrections are being received (apart from the "succesfully connected" response with the `IS_.OpenServerConnection(RTK_connection)` functino call)? There's a small disconnect I suppose between testing the stream in a different tool and then plugging that into the device. If possible, it would be great to debug this issue directly with your team to protect our credentials for the NTRIP server.

mjshiggins, 2020-05-20:
@siddirp Following up here. Still cant get the device to accept an NTRIP stream.

siddirp, 2020-05-26:
Sorry for the delay. Let me get with one of our testers and see what I can figure out. Can you send me an email to support@inertialsense.com so we can work directly?

closed 2020-08-25 by siddirp