inertialsense/inertial-sense-sdk #27: Firmware/Bootloader update Failed_1.8.1

xinai13994, 2020-06-24:
I have a INS Rugged unit used in work. The unit was working fine on 1.7.6 version. I tried to update the firmware to the newest (uINS 1.8.1) with the newly released Bootloader Version (SAMx70 v4b), the platform used to conduct upgrade is EvalTool 1.8.1.

However, the update failed on first attempt, then the serial connection not working ever since the failed attempt. Returned information in Firmware update window (Setting tab in EvalTool) is: "Starting bootloader...Unable to find bootloader...Attempting to reload bootloader...COM3: Firmware update ERROR!...Update failed to communicate"

Please advise.
[image]

siddirp, 2020-06-24:
Hi @xinai13994 Please just check the box to reopen the COM port and try again. Feel free to send an email to support@inertialsense.com if you need more assistance.

natfallow, 2020-06-30:
Hi @xinai13994, I found that I had to first update the bootloader to v4b using the 1.7.6 version of evalTool - it seems version 1.8.1 cannot detect earlier bootloader versions. Your serial connection is likely no longer working because it is stuck in bootloader mode, which you should be able to resolve by restoring firmware version 1.7.6 (using evalTool 1.7.6).

waltjohnson, 2020-08-11:
This issue is resolved in the latest bootloader.  Let us know if you have any further issues here.

closed 2020-08-11 by waltjohnson