inertialsense/inertial-sense-sdk #30: Arduino example doesn't compile with 1.8.2

kiranshila, 2020-07-10:
Due to ISComm refactor, the arduino example does not compile

kiranshila, 2020-07-10:
In fact, this seems to be the case for all the examples

waltjohnson, 2020-07-10:
Kiran.  I just cloned and compiled the InertialSenseSDK repo and all of the
example projects built just fine for me.  Can you post what errors you are
seeing?

>

waltjohnson, 2020-07-10:
@kiranshila It looks like we intended to removed GpsParser.cpp from SDK repo, but didn't.  Do you know where that file gets used?   Deleting it seems to be fine as it is not a dependency for any other file in the SDK.

waltjohnson, 2020-07-10:
I removed the GpsParser.cpp and GpsParser.h from the SDK.  Let me know if I missed something here, but it shouldn't be needed.

closed 2020-07-10 by waltjohnson

kiranshila, 2020-07-10:
@waltjohnson How are you building the Arduino project?
```
src/IS_Example/main.cpp: In function 'void setup()':
src/IS_Example/main.cpp:46:8: error: 'struct is_comm_instance_t' has no member named 'buffer'; did you mean 'buf'?
Compiling .pio/build/is_example/FrameworkArduino/IPAddress.cpp.o
   comm.buffer = s_buffer;
        ^~~~~~
        buf
src/IS_Example/main.cpp:47:8: error: 'struct is_comm_instance_t' has no member named 'bufferSize'
   comm.bufferSize = sizeof(s_buffer);
        ^~~~~~~~~~
src/IS_Example/main.cpp:50:21: error: too few arguments to function 'void is_comm_init(is_comm_instance_t*, uint8_t*, int)'
   is_comm_init(&comm);
                     ^
In file included from src/IS_Example/main.cpp:3:0:
lib/INS_Arduino_SDK/ISsdk/ISComm.h:531:6: note: declared here
 void is_comm_init(is_comm_instance_t* instance, uint8_t *buffer, int bufferSize);
      ^~~~~~~~~~~~
src/IS_Example/main.cpp:55:12: error: 'struct is_comm_instance_t' has no member named 'buffer'; did you mean 'buf'?
       comm.buffer,
            ^~~~~~
            buf
src/IS_Example/main.cpp:62:12: error: 'struct is_comm_instance_t' has no member named 'buffer'; did you mean 'buf'?
       comm.buffer,
            ^~~~~~
            buf
src/IS_Example/main.cpp: In function 'void loop()':
src/IS_Example/main.cpp:76:56: error: too many arguments to function 'protocol_type_t is_comm_parse(is_comm_instance_t*)'
     uint32_t message_type = is_comm_parse(&comm, inByte);
                                                        ^
In file included from src/IS_Example/main.cpp:3:0:
lib/INS_Arduino_SDK/ISsdk/ISComm.h:605:17: note: declared here
 protocol_type_t is_comm_parse(is_comm_instance_t* instance);
                 ^~~~~~~~~~~~~
src/IS_Example/main.cpp:81:41: error: 'struct is_comm_instance_t' has no member named 'buffer'; did you mean 'buf'?
       handleINSMessage((ins_1_t *)(comm.buffer));
```

kiranshila, 2020-07-10:
As I mentioned, this seems to be inconsistent with your refactor of ISComm.

waltjohnson, 2020-07-10:
Thanks, Kiran.  We will fix this issue.

kiranshila, 2020-07-10:
> Thanks, Kiran. We will fix this issue.

I'm almost done with the PR, testing on hardware right now

waltjohnson, 2020-07-10:
Thanks, Kiran.  We'll look for your PR.

>

kiranshila, 2020-07-10:
Yeah I'm still not getting any data back from the unit.
I'm setting the message broadcasts with 
```
  messageSize = is_comm_get_data(&comm, DID_INS_1, 0, sizeof(ins_1_t), 50);
  INS_SERIAL.write(
      comm.buf.start,
      messageSize); // Transmit the message to the inertialsense device
```
But I am not receiving a single byte back from the unit

kiranshila, 2020-07-13:
Any movement on this?

coyt, 2020-07-16:
Bump. 

We'd very much like to update our uINS modules to the latest firmware because of several other problems - mainly the AR ratio not being shown properly on new boot-ups in FW1.8.0 - looks like you fixed that in the latest firmware. 

We can't update though until this issue with the Arduino code / libs is working :( We tried to update a week or two ago and nothing worked during a field test of our gear because we didn't realize the embedded code needed to be completely different.

siddirp, 2020-07-17:
Hi @kiranshila, and @coyt,
I spent some time today fixing the the Arduino example code, but I've not had a chance to test it yet. Once the testing is completed I'll let you know!

coyt, 2020-07-18:
Awesome! Thanks for the support. 

If you think it's ready for testing, we're happy to update one of our units and test out the fixes first thing Monday.

siddirp, 2020-07-20:
Can you send me an email at support@inertialsense.com and we can talk about
details?

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
andrew.priddis@inertialsense.com
*inertialsense.com* <http://www.inertialsense.com/>