inertialsense/inertial-sense-sdk #33: Two units with identical configuration gives different results

AlexisTM, 2020-08-06:


siddirp, 2020-08-06:
@AlexisTM - Your other question indicated you were trying to use Compassing and RTK positioning simultaneously. Is this the case in this setup? To discuss further could you send an email to support@inertialsense.com.

AlexisTM, 2020-08-06:
In this setup, it was either one or the other, the firmware [1.8.2] correctly disabled POSITION if I tried to enabled 0x09.

In this case, the `RTKCfgBits` are 0x80008

AlexisTM, 2020-08-07:
@siddirp 
The µINS3 is from the µINS Dual GNSS Compass Development Kit

siddirp, 2020-08-07:
Are you seeing "Compassing Fix" status? Most likely you have interference
problems and aren't completely getting into a fixed state.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
andrew.priddis@inertialsense.com
*inertialsense.com* <http://www.inertialsense.com/>

AlexisTM, 2020-08-08:
Both the rugged unit and the EVB-2 should be having the same perturbations, yet no received satellite goes into Float/Fix state at all on the rugged unit, even in ideal situation (open-sky, open-field). I go not a single GPS signal to Float/Fix for both compassing and position mode on the rugged unit.

siddirp, 2020-08-08:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN203 )

Inertial Sense (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN213 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN223 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN233 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN243 )

Schedule Meeting (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN253 )

Read Blog Posts (https://info.inertialsense.com/e2t/sc2/Mm-nFbkB_0QW7cKx4l7tV6QgW7mFN9g8QqZWMW8hWtXD8QqH96f1HVPN263 )

Inertial Sense, 384 S 400 W #125th, Lindon, UT 84042, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hrytW1Ztq4c1XmmZlW1Synkn3W3636W3HbJ363KbsvFW3R0gs93_-09-W47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8zMnRfW4Rc5GT6tf82_W4_ZJwY8YNKs3V28Q_q8Xp6wxMGKd_P20N5RW3PHcCm8jR_5YW3d2RgN2R0wzY4CplG1&v=3&_hsenc=p2ANqtz-_G8dTM7_HFH_QfvWhL2TyRlpfVdmR2lgsweaDN5JGDD-hR_GGefETepYRXvc1xK93ZpW4HmtXP3-bT_DHvQ3UY4LZ4X13d3MFOtN5YrzqaOySqS23OFo-xDWfiMCb74pfzruSxlQ231QtC2TYKkjE-g26HRw&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hrytW1Ztq4c1XmmZlW1Synkn3W3636W3HbJ363KbsvFW3R0gs93_-09-W47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8zMnRfW4Rc5GT6tf82_W4_ZJwY8YNKs3V28Q_q8Xp6wxMGKd_P20N5RW3PHcCm8jR_5YW3d2RgN2R0wzY4CplG1&v=3&_hsenc=p2ANqtz-_G8dTM7_HFH_QfvWhL2TyRlpfVdmR2lgsweaDN5JGDD-hR_GGefETepYRXvc1xK93ZpW4HmtXP3-bT_DHvQ3UY4LZ4X13d3MFOtN5YrzqaOySqS23OFo-xDWfiMCb74pfzruSxlQ231QtC2TYKkjE-g26HRw&_hsmi=86142041 )

waltjohnson, 2020-08-11:
Alexis.  Have you made any progress on this issue?

AlexisTM, 2020-08-11:
On the last in-field tests, I finally got similar results but the ambiguity resolution ratio never went higher than 2 or so for both units.

waltjohnson, 2020-08-11:
I suspect there is some sort of interference.  This can be due to mutli-pathing of the satellite signals or some other noise source that is impacting the RTK solution.  Are you using [GPS antenna ground planes](https://docs.inertialsense.com/user-manual/gnss/gnss_antennas/#gnss-antenna-ground-plane)?  What is the distance between your antennas?

AlexisTM, 2020-08-11:
I am not currently using a ground plane, I will add one on each antenna.

The distance between antennas is 1m for both.

siddirp, 2020-08-12:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8M04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8N04 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8P04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8Q04 )

Schedule Meeting (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8R04 )

Read Blog Posts (https://info.inertialsense.com/e2t/sc2/MmZ-8yjZ5PLW8YKH3_7DqBsHW6xgdls7sc4ZSV28PGW7Zp2pWdBzP8S04 )

Inertial Sense, 384 S 400 W #125th, Lindon, UT 84042, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21hrytW1Ztq4c1XmmZlW1Synkn3W3636W3HbJ363KbsvFW3R0gs93_-09-W47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz5wqwC8W64QbLK4RszpdW505_f17XxhwGVWzpl95Sb1gvMXk_7QbwcydW8d8krT3fLFQvW2DzBP36Z0BCz3pWY2&v=3&_hsenc=p2ANqtz-_f4UQozYw1EWzpvuOgweL1o_2wir4tOAN9d0XfrQMkpe8lVIivgFYsQQ2Tmemkx7ecZz5FPmZFif2TPybrA9ULylL1mqZ2i0iovLaD32snsOQMXI6j4tG_jSNjlkwqJozAnAJ9PrB9FVwH5t8P51RFWqcpcw&_hsmi=86142139 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21hrytW1Ztq4c1XmmZlW1Synkn3W3636W3HbJ363KbsvFW3R0gs93_-09-W47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz5wqwC8W64QbLK4RszpdW505_f17XxhwGVWzpl95Sb1gvMXk_7QbwcydW8d8krT3fLFQvW2DzBP36Z0BCz3pWY2&v=3&_hsenc=p2ANqtz-_f4UQozYw1EWzpvuOgweL1o_2wir4tOAN9d0XfrQMkpe8lVIivgFYsQQ2Tmemkx7ecZz5FPmZFif2TPybrA9ULylL1mqZ2i0iovLaD32snsOQMXI6j4tG_jSNjlkwqJozAnAJ9PrB9FVwH5t8P51RFWqcpcw&_hsmi=86142139 )

waltjohnson, 2020-08-25:
@AlexisTM Hi Alex.  Did this get resolved?  Did you add antenna ground planes fix your problem?

AlexisTM, 2020-08-26:
I added the ground planes, which indeed improved the GPS signal. 

Yet, I could not get an RTK fix nor robust RTK compassing, with the ambiguity resolution ratio never reaching over 1 or 2. 
When compassing (baseline of 1m), the measured distance was oscillating from 0.6m to 1.8m.

Here-under, the top plots are from the µINS3 (top left) and EVB-2 (top right). The data comes from `DID_INS_1` and is converted to from NED to ENU. The bottom (orange) plot is from our reference. 

[image]

Here-under, the `GPS_POS_1` position output from the EVB-2 (pink), the µINS3 (purple) and our reference (yellow) for the same track. The graph looks stretched on the x-axis as it is longitude & latitude.

[image]

We can see one big jumps of position in the estimation for the µINS3 and a very very bad estimation for the EVB-2.

It is to note that the yaw (heading) of the EVB-2 was far (20°) from the estimation from both the µINS, our reference and a competitor with at worse almost 30° off. This can easily explain the bad EKF output.

waltjohnson, 2020-08-26:
Hi @AlexisTM.  Are these logs recorded with RTK positioning or RTK compassing mode enabled?  If the AR ratio is below 3 you will likely have RTK problems.  If you have ground planes under the GPS antennas and are still having issues, I would suspect you are getting interference from other objects in close proximity.  What is the the GPS satellite CNO level?  This should be above 40 dBHz for RTK.

https://docs.inertialsense.com/user-manual/troubleshooting/gnss/#satellite-cno-strength-rfi-and-emi

Any chance you can record PPD (post process data) log?  This will gives us best visibility into issues you are seeing. 

https://docs.inertialsense.com/user-manual/logging-plotting/data_logging/#post-process-data-ppd-logging-instructions

AlexisTM, 2020-08-29:
As I could not get a proper fix, this was without RTK and without compassing (Same condition as the reference).

closed 2023-04-17 by steezeman