inertialsense/inertial-sense-sdk #37: Add the device serial to the USB descriptor

AlexisTM, 2020-08-12:
Hi,

I would like to differentiate between two devices using UDEV rules to prevent having to open the device to get information.

In the ideal case, it would be great to have the serial attribute of the device forwarded to uniquely identify them to allow symlinks such as `/dev/inertialsense/40756` and `/dev/inertialsense/40840`, without opening the serial port.

This is useful in the case where we want to run different software over two devices where each device. The only method as of now is to open both serial devices, try to connect and get a serial and close them. When starting the second node, it will impact the first node which is already connected. 

Of course, in production, if the case happens, this would be handled by a single software which can then do the discovery safely.

As of now, I am unlocked thanks to the  `urbnum` which is respectively 6 and 9 for the EVB and µINS. The following udev rule creates a device `/dev/ttyIsUins` and `/dev/ttyIsEvb`.

```python
SUBSYSTEMS=="usb", KERNEL=="ttyACM[0-9]*", ACTION=="add", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="6124", ATTRS{urbnum}=="9", MODE="666", SYMLINK+="ttyIsUins", GROUP="dialout"
SUBSYSTEMS=="usb", KERNEL=="ttyACM[0-9]*", ACTION=="add", ATTRS{idVendor}=="03eb", ATTRS{idProduct}=="6124", ATTRS{urbnum}=="6", MODE="666", SYMLINK+="ttyIsEvb", GROUP="dialout"
```

waltjohnson, 2020-08-12:
Alexis.  Do you know which field in the USB descriptor we might use for the serial number?  I'll check with our embedded dev team, but any suggestions you might have are welcome.  It looks like we could make the product ID different between the EVB and uINS.

AlexisTM, 2020-08-13:
@waltjohnson I believe the best would be the `serial` field, accessible in the udev rule with: `ATTRS{serial}=="40840"`

waltjohnson, 2020-08-25:
@AlexisTM  Hi Alexis.  We have looked into adding the device serial number to the USB descriptor.  It would be fairly easy to implement.  We have even tested it out.  However, it breaks our existing bootloader update process, causing the serial port number of our device to switch.  Can you work around this issue for now?  We can always further invest this if needed, but for now it appears to add significant complication.

AlexisTM, 2020-08-26:
Yes we can work around it, it is mainly for simplicity for the users later on; allowing to connect multiple devices for testing purpose.

closed 2023-04-17 by steezeman