inertialsense/inertial-sense-sdk #41: wrong compiler arguments for C source files when trying to install logInspector

ryanofarizona, 2020-09-18:
After doing `pip3 install logInspector` to get the dependencies, I did `cd logInspector` followed by `python3 setup.py build_ext --inplace`.  I get the following error when trying to compile the first C file (not C++) in the list:
```
gcc -pthread -B /warehouse/users/rlh/conda/envs/sparky/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -Iinclude -I../src -I../../src -I/warehouse/users/rlh/conda/envs/sparky/lib/python3.7/site-packages/pybind11/include -I/warehouse/users/rlh/conda/envs/sparky/lib/python3.7/site-packages/pybind11/include -I/warehouse/users/rlh/conda/envs/sparky/include/python3.7m -c ../../src/com_manager.c -o build/temp.linux-x86_64-3.7/../../src/com_manager.o -O3 -DVERSION_INFO="0.0.1" -std=c++11
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C [enabled by default]
../../src/com_manager.c: In function ‘stepComManagerSendMessagesInstance’:
../../src/com_manager.c:413:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (broadcast_msg_t* bcPtr = cmInstance->broadcastMessages, *ptrEnd = (cmInstance->broadcastMessages + MAX_NUM_BCAST_MSGS); bcPtr < ptrEnd; bcPtr++)
  ^
../../src/com_manager.c:413:2: note: use option -std=c99 or -std=gnu99 to compile your code
../../src/com_manager.c: In function ‘comManagerSendInstance’:
../../src/com_manager.c:641:2: warning: missing braces around initializer [-Wmissing-braces]
  pkt_info_t pkt = { 0 };
  ^
../../src/com_manager.c:641:2: warning: (near initialization for ‘pkt.hdr’) [-Wmissing-braces]
../../src/com_manager.c: In function ‘comManagerGetDataRequestInstance’:
../../src/com_manager.c:932:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (broadcast_msg_t* bcPtr = cmInstance->broadcastMessages, *ptrEnd = (cmInstance->broadcastMessages + MAX_NUM_BCAST_MSGS); bcPtr < ptrEnd; bcPtr++)
  ^
../../src/com_manager.c:944:3: error: ‘for’ loop initial declarations are only allowed in C99 mode
   for (broadcast_msg_t* bcPtr = cmInstance->broadcastMessages, *ptrEnd = (cmInstance->broadcastMessages + MAX_NUM_BCAST_MSGS); bcPtr < ptrEnd; bcPtr++)
   ^
../../src/com_manager.c: In function ‘comManagerDisableBroadcastsInstance’:
../../src/com_manager.c:1051:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (broadcast_msg_t* bcPtr = cmInstance->broadcastMessages, *ptrEnd = (cmInstance->broadcastMessages + MAX_NUM_BCAST_MSGS); bcPtr < ptrEnd; bcPtr++)
  ^
../../src/com_manager.c: In function ‘disableDidBroadcast’:
../../src/com_manager.c:1062:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (broadcast_msg_t* bcPtr = cmInstance->broadcastMessages, *ptrEnd = (cmInstance->broadcastMessages + MAX_NUM_BCAST_MSGS); bcPtr < ptrEnd; bcPtr++)
  ^
../../src/com_manager.c: In function ‘sendAck’:
../../src/com_manager.c:1212:2: warning: missing braces around initializer [-Wmissing-braces]
  p_ack_t ack = { 0 };
  ^
../../src/com_manager.c:1212:2: warning: (near initialization for ‘ack.hdr’) [-Wmissing-braces]
../../src/com_manager.c: In function ‘comManagerValidateBaudRate’:
../../src/com_manager.c:1576:2: error: ‘for’ loop initial declarations are only allowed in C99 mode
  for (size_t i = 0; i < _ARRAY_ELEMENT_COUNT(g_validBaudRates); i++)
  ^
error: command 'gcc' failed with exit status 1
```
I can verify that it compiles manually if I change `-std=c++11` to `-std=c99`, but it tries to re-compile it with the wrong flags when I try `python3 setup.py build_ext --inplace` again.

siddirp, 2020-09-21:
There are build errors that are known. Can you still run logInspector? I think in the Readme it states that you will see build errors that should be ignored.

ryanofarizona, 2020-09-21:
No. The Readme mentioned errors when doing `pip3 install logInspector` which I did see but ignored.  I am getting the error above when trying to do the next step `python setup.py build_ext --inplace`.  I tried to proceed to the next steps by creating a directory at `/home/[username]/Documents/Inertiial_Sense/config.yaml` with entries similar to what is in the Readme, but when I try to run `python logInspector.py`, I get the following error:
```
Traceback (most recent call last):
  File "logInspector.py", line 19, in <module>
    from logReader import Log
  File "/my/path/to/work/area/inertial-sense-sdk/python/logInspector/logReader.py", line 14, in <module>
    from log_reader import LogReader
ModuleNotFoundError: No module named 'log_reader'

closed 2023-04-17 by steezeman