inertialsense/inertial-sense-sdk #227: [Evaltool] 3D model doesn't show & appimage or flatpak instead deb for Linux

AchmadFathoni, 2022-04-27:
I tried the 1.8.6.b Evaltool in Linux and found that 3D model doesn't show in INS tab
[image]

I guess there are some missing dependencies. What if the Evaltool is distributed as bundled app that contain dependency such as appimage of flatpak for Linux? Or release the source so Linux user and compile it themself.

AchmadFathoni, 2022-07-05:
Well at least I still can run Evaltool.exe on Linux using WINE.

siddirp, 2022-07-05:
Is the missing 3D model the only issue you see? I will register a bug for that, but that is probably not going to be seen as a high priority item. The calibration path error only occurs if you have internal mode enabled (which you do) and that directory does not exist. There are 2 solutions: turn off internal mode, or create a folder in the path described.

AchmadFathoni, 2022-07-06:
> Is the missing 3D model the only issue you see? I will register a bug for that, but that is probably not going to be seen as a high priority item. The calibration path error only occurs if you have [...]

Thank you but I think it's OK. Running Evaltool on WINE fixes the missing 3D model problem. Just one question: There is no Evaltool.deb on 1.8.6.d release page, is it deliberate?

siddirp, 2022-07-06:
It is there. You may have just overlooked it?
[image: b93d5984-ddd2-4366-bab0-4f5ad16aefca.png]

AchmadFathoni, 2022-07-06:
> It is there. You may have just overlooked it? [image: b93d5984-ddd2-4366-bab0-4f5ad16aefca.png]

They just uploaded it 16 minutes ago. 

[image]

siddirp, 2022-07-06:
> It is there. You may have just overlooked it? [image:
> b93d5984-ddd2-4366-bab0-4f5ad16aefca.png]
>
> They just uploaded it 16 minutes ago.
>
> [image: Screenshot_20220706_231152]
> <https://user-images.githubusercontent.com/26775746/177596153-3dcdeec3-c86f-4440-a30f-71db42c6b9ae.jpg>

I didn't realize that was happening while I was replying to you. 😀 Also, I
didn't see the timestamp. I just popped open the release page and saw it
there. Sorry about that. The most important thing is that you have it now.
Please reach out if you have any questions.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
Mobile: (801) 610-6359
*inertialsense.com* <http://www.inertialsense.com/>

AchmadFathoni, 2022-07-06:
Hey, now the 3D model is there in EvalTool_1.8.6.d.deb. Closing.

siddirp, 2022-07-07:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_Yf5js7VV3Zsc37CgS9YW37n5nh3jj9K7W7qlp7l7X6x17W7r2bl56mQNCyW1ZTVGZ17sgVnW9lqb5y5GmpbDW30kSQs29dXWKW2cVkvm5fFH5vW7crFYZ2FpVx5N545-qQwtHNhW5hRVNn42PY1pN1rzcdFqyvC7W8Vyyq-5kdZgWV4w6P89d5J83VTvC_149c9zzW7lHP8f7PC0L_W3BlpTs7CytK0W3xYYyx5SHNHrW3My4qV3ZQFf1W6kYGQx58GwTGW412l9b3Qm9T9W653jSm48bjm5W4twS_G1Sp4WWW7lX65_6R1rpkW2Z2jjZ2ngp7YW5PK9SD2fb33TW3SGTpG4wLjqkW8VYy8V7z8H3yW92PWrt73PkLtW3TwcKj8NyB1lW9knv-C32F3dkN1svdCkhhvLTW8MH3KB26PzbZW3Jw-5X7SLqGdW5hNVf27PKynXW1bfvsD3NhKjFW8JcVH_19bbYJW8fypkm4hwwsBW22JNgp2RH_LJW2bDnYW48Vz6dW4fpmY54hmbBMW7-DPzQ4vRVbWW734sVf5dpFdx31PN1 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_Yf3kWFJV1-WJV7CgMNHN740RfWd2YwgW6DL9P24sJbZ9Vp5nsq50RyhfW8hVlr448DhtrW7fp-S98X5Pf6W86JDd018DzxPW3f4WSq4wqr4vW7922Nw2r98TLVS3VDm3ZdsfmW6v1SwG5HhbWNW15T2RM2s5pFDW7pYpjr6LbxpLW6g6sm22Cx391W85YMps54T4lCW68-G7r3QHjy9N7jg6dbXXp8qW6dT89K1Cm-FqW1xVQ0r44B-9qW5jPZkl4M3NZLW6LWh-L4VCZcJW7dybDz2GSFyVVccKLn12xxy1W5qpJ7j20PZSNW3Gwhld9cb_JmN1K8h1pnglf0W5NtH8p2YFkVx375F1 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_Yy3kWF_V1-WJV7CgHB9W3766Jn3czTsYW1h1Z7x5bNYMbW5-K9646vDmFQW6m3WwG71GfZKVMdRVQ2Dt2MqW9gLx7B8_65GjW7JjQDw9gkyPxW6P7D8T29jFHGW8n9_zm1bcnFYW77zL3x5thCsvW6J7sGM1kc1WsN2pfrf2MG4jpW76cJRv5PZWH8VNPWV23Mlt31W6N1d-K7pK_jbW1L__L-33JZ3fW2fVLbl3WpVW-W8BkzFZ41ZPJvW244jW24y8FFcW2-Sg5m13PvscV4Lxvv77FR5dW6qxL6f4gN2gzW8_LbSR7ZXMs8W1fq6Xf82fkT8W1NYGkz5KC8r6W2TvXFd6xywQkW1GMV1b57KxdwW123hZQ5bR52w33kX1 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_WN7ND_VV5X_Kf7CgQJWW4rlklG8B_-D_W4B1YKR8LmMwgW2-Ncyz30Mtm0VdC9PB99C9_kVtsyGh1x09WrMFXxNSs5rXkW2rNpWr1qcj18W4Vz4HG6hzymbW4gt5qw5gTdqmN1R5d9FZrT5qW2pHSH44nb_-zW7J7ldc6ykR4jW47xvcp6Fs0qzVsKqsY5MpG4CW8T1tcH98WJG6N8CvPy1M3BDNW83_VRL8wkMR5W5LlDg42RqrHPW7RcK9C75TV9hW35rhwG85NKmrW1fhrG67YB2c-VjXWj34_q0JMW10JBhx2Dn4NBW3XKysn19DgrGW2vhb0z76q2_vW6skMNZ2GQf9QW5dtshl7bZQ_qW8KH7c-20GnysW6wRJ7-4-1ZZ-W2d4vJV3_gzSKVJcrm27k2fkCW78w_l_7c6rtKW7lc0Tv8ln2xTW5BDFYz8gVpYvW18DHlH7hNPVqW79dYjb8hVrnWW3F7sLc4fhyc5Vl44Vr6C5HYVW2hB05y1zL-6XW4nf17d41qCHSW2HkJKp3SqjG9W4vYp1B2l8DmKVNJ8_g8PNHcmW5bZGJL50P2HhW12kPv98_4gJ1W136PZ_39L-XkW65SL6384_9KXW9hs8vS4Xf3G3382x1 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_WN5js6pV3Zsc37CgJZwVJNJVz6Zgm93W86Kg0t9g1dH5W3Pq6ws932C_7W3HWVDH3BBBdvV2bgWy2pKyZmW41Tmp94NB7l9W1N23ck8w5SVpW3c8z_Y286LhLW5QHKdw5Sdc8WW6B14wg4fgxtXVxMJwG57J7g6W1gc3xd5yYhjGW9b579c28Fl-WVvfSnY7hcJ_fW15PqMx1jy_qcW3TH1t41kTNDsVw0bMk6GbNTxN7jVDx4q7wCSV_RSZq6y9YqxW6npt0f2kLbqKW3VXwlD2qrngYW6qzC0K5wmFYZVgpVB-4XDrLyW3xL7Yd1LZWjDW6f0qMP8m6wLtW4B24pp4ByRpwW2tc4ks3dT_FwW159lcb21M8G3W25HdmT8j7wbSW5yF4Xj54BrmxW6bksxX56td70W7fMfFd3F0n2N31311 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_Yy3kWF_V1-WJV7CgY66W7p46bp2RdtljW7KgWsf1h51y4TlpxV8QJ0y5W3dv1x68L7R6PN7nzkTp__5kkW9gfxfG85ZLchW2HTRHh5PkvbsW4bLH5L2YWSLwW1JSY7H1XrBprW8Hwhqp4BjvS6N77pXgxXGGL6W7f9-Zs51-X2wW6l1qjn7bS4vVW8WpPCD2_DtPMW3ZWMGr4xm_KnW2tHsD42ClmGTW4Ht3xs2jvBRXW8Kg-2p90CSnxW7q41n-77vRLfW7vmxJC5cD30YW1wL3ms6p-kKvW1swTt4315DQ7N3j_x4syM-2lW6gj7y58B3WLQVt2G8r8TfgqpW6vtMxn6fhHXyW5xWnfb7z9GLVW7mvJTg54RFxd3gnZ1 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX497D2ZV4FZN6PbBQZW6Y8fW4125np4MbY0xN1f4_Yf3kWFJV1-WJV7CgBL5W8jmH49421GwWN495bY-kTy6rW1Fq_535NHLJ3W8MPlHQ5NMFTbW7SsqRv8skchMW6Dnv6h7Bv1gXW11_Ry030qlYGW2K6MC28NmF7rW8vTB946kl44rVWBsNw1tH6c8W8fDy6V8Ft1nVN1Yp5hZpKQcQW25r74v4ZWp4gW8w55DN1xyHWnVr4pW778GrGsW5JmZ9L5QdJxsW39V-Yw6yZ4BmW1tbgX58LJjrHW7BF19v4XW3HgW8qcn872kfFBbN6P_tfrJzKhHW7pv3bN1WPzb1W7bBMfy4-dCBJW5fjrFN7PnjGYW2ML1Mt4yrsGgW75_VZh67jXKK3g2z1 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21h05MW3_Wg9C4pFm3LW1Xv2Tk3R6wm-W3XWfkb3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8KX8vhW4XGh9Z5yXvt3W5fyMXY8z-13tV28Q9q76lkljMGKd_P20N6QW8MSG_B2jT7CjW9hlkX-6b1k7y51VhdY1&v=3&_hsenc=p2ANqtz-9Djy62fW4Y8FudW1mq7-w4C-6d5LFLSs5k_pBz51VCGf8HBcgLztIVj_FwXi__a-1ud7k9XNoBt1AHGGwLu_FQ5g2cmkWZZ3nn2SKcvDsR_fYCI3mnxo5FTUoPBDqU6jgwYCpqBpQQ3v1GfSjJqUq42tEpEA&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21h05MW3_Wg9C4pFm3LW1Xv2Tk3R6wm-W3XWfkb3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8KX8vhW4XGh9Z5yXvt3W5fyMXY8z-13tV28Q9q76lkljMGKd_P20N6QW8MSG_B2jT7CjW9hlkX-6b1k7y51VhdY1&v=3&_hsenc=p2ANqtz-9Djy62fW4Y8FudW1mq7-w4C-6d5LFLSs5k_pBz51VCGf8HBcgLztIVj_FwXi__a-1ud7k9XNoBt1AHGGwLu_FQ5g2cmkWZZ3nn2SKcvDsR_fYCI3mnxo5FTUoPBDqU6jgwYCpqBpQQ3v1GfSjJqUq42tEpEA&_hsmi=86142041 )

siddirp, 2022-07-07:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0Hf5js7VV3Zsc37CgN_mW8RCqs12vVx_MW6Nkzmp1__cw-W6HGXV81vtQxyW3TQRg74qqlwMN1l2XN4DsbnFW1LT47-7GC9jRW8lJPxs6qQZHhW6q15Xd5L2vgVW8qr8Yf7CfBpTW729KsK7jZs6CW3lMsjv7wtqG6W1rPDqF7VPFCdW8w-f5C5McfbbW4MXlbv7hsDNlW65LckJ4zG0b8W2zXlvG6Jf23HW462CrM4jNnQ_N71x6243YKKvW2sVxcW3qnBZLW43GRbC22L4q6W1Vvj_822DZtzW6wMHNj2T2db9W4dj_j38sNkllW2Ppvh76fk3XNN9jt967P40VWVwC2MG2Thf9nMHfDgKZpTr-W550nnD6V36C9W4FS55-54LfWRW1Xvn2v4gRBVwW6PYp681jKD71W4dMLGd59LR2-W8zqgnX7XzKN6W5KD19y45yG7kW2MQHNY8z_0s8W3znCzF8Z4LvVW8Q7rg481cGckW7lYKJq2T-YPwN7XtCxnvPkWlW72B6HQ3slng1W5CzKjY4lQnLqW72cjgz8fcBzt31vS1 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0Hf3kWFJV1-WJV7CgFrGW1Jjd8C4pR5BFW2S2Cd65_nJnnVZJrDy1-hVFgV9t28N24RBlYW8tCzB932xD_TW2rCVSs2X0MlRW7rB70n5dFtLcW6_H7rx7QbwDDW6kJKfK51MpLLW2swZtW87jX4GW1Mr9WV4ZfnZ2W94Cnlc11QCBRW6Wl7h02tJswWW8PQXV22ljRbVW7mV4lT1wnl6WW2bgr-W1hqbzHVS1-Qv5pRWMQW2l_PF17_8tZnVvhGF51PPMJqW6zJ3DD3-Dg6YW5kT2cM8Pr212W5gXMkd2zw5swW4f9SXs2jTcK4W43HD-T8VX1SbW4n96815WrdX3W7dKd5m7sgNP835FM1 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0Hy3kWF_V1-WJV7CgD3hN4VFjJ95mPM6W6gj4Zl1jl40pW5Zr-8-2RQ_xgW4v2mt68GJtb5N8JGHcJLmxjCW6q8Tkb9bjFHrVR3N5G5KFjh-W4XKRV571BD1rVsZF_G7XCzLGW8DGYzd7BGtfvW3yTzrq42kg9dW5RPqbH8PPl9BMmjzFvd58cNW6VwwCd2879t4W7_Yr4-8VHH-RW5cP1Xz4JPX14W6Q-Zkr3QgfVZW8GC2gy2jf_VXW7-cL6f7fKFBkW6Q-GL01lCbPKW7K90rh3LVddTW5JSC5M7tK4NhW8Xk8Xg3wfdG9W2DD_hh3z5r5GW5Tddwc4-pRNcW8xYH9n5ZF-qdW7bhsjP13q3spW3rppV254cpmK3cVM1 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0FN7ND_VV5X_Kf7CgH01W84kVcv2Yrr11W64zKF2573TJTW23VWcC1BGJV2W6Gk0zK3VLngKVN9_NR7MMjL3W7BLLgQ3fPjnjW6Gzcqm2m7K2HN84cTH-FGsD_W8jp2x327ffKrW4_sBF87p3qPzW90RQVH1qvwtGW1gc5h456-2ktW7yPFrR3Q0zXvW6xQLG_7l6gM_W2GhG4z4L-LPPW7NX9Xb3vcFTLW1m4str4ld2CkW3xDdyH2K8xVxW59qKjX55XR_HW5yDtt45RPN33W750hGT62_CRCW4d7bTJ11nJ05W5mDyjP83pFN_W4pwxXc6cbvS1W2JQMx-9j1R7XW5HzBZD5Dkx_VN8dHWzmnWB0YW5c1xMr96SymlW6-810D3zv3y8W1yrcbf8JMr37W8zQ6YT1f9js5W5Ff0XS1dJNcPW1lsj6C6FX8d_W6Yb0h33Mnjk1W4jYTTw4YFnd4VZwgvf5LzgvTW1mjKtN69yB2XW2wRXG45Ww35nW4NKWpL7H-RygW3jtdNz8xD_xPW6Kk4r-7yHQX2W3mtGGw5FmBD5VGbfWL4HbkbbW2Sw-Hx4N15c-W13Wr9J49HF5XW7QH9R77vHckHN2jzrmQ-FHSvW7L31Qf68nG4v2Tq1 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0FN5js6pV3Zsc37CgKDQW1Kd6DJ79CYmWW53gSnS4z_qTgW4CmZLR5VV_8RW6bmmK73R9hgfN7nd9xzTRWfVW44yhBz795w2tN12FL2dXkc7DW5vhw2X8twwkwW79G9_75LDFwVW362MQY4YqwkZW9216bH1bt6wrW1MrVBz1xZ4hXVC9PpQ8ZTrTvW7FD5b65t91hKW96c9nY1N4zM5W7f3VD_2JCPHgW5S4tLB4GYmx0W5ysyW8448FsSN2b1d5Jy6J1-W3H8g-Y8b8LlmW1Bwyl51X7V9yW3LklSq3psjFMV6vm-d8FJz1HW92Qtmx2_tDzjVslh4q5kl7ZvN7NmyfSGXtT1N8nwPT6jfrF4W6-JYCp6txVxTW1fftJf7vBn1PW2dZ0Lh6QgSGGW5r2DpW8FM50SW8Y167M6gBS5632yW1 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0Hy3kWF_V1-WJV7CgD3SW4sNP116mMnBvW5m89FL18byZMW5TXjdz7ycP0PW56w1Mr7zxX0VN7q9xZRkbX_vW48cgjt3tV8lJW3Z5-0q9jLF8_W12R5YJ2j-zDCW7HJm9F4mcwY0W54NjDn4m_JJsW2w0rLj6L1BTkW1q3g-23-6gypN61PnVvdmfybW7ZK6qN4BRDlZW3WnY2Q8H8ld4W19tRNX24NhG1W2zXGzL6C_bCYW39hJRD5K20wNVW7-8-92h_WDW20qyF_3P4QLXW3mJLn07DQTDGW5wwgG51qVFYKW19MHMp4trMVYW5w3mlJ5QtPcjW5smSvl4w_R6SVlsW3c8Fw34fW4KT8Y_7jpjBpW4CJRvm6S4lgK3mFd1 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWPD607JZMlGW23R9MY2v-yv4W1KLHGL4Mb_cwN97h0Hf3kWFJV1-WJV7CgXyyW6Pwny_2hTNjbW5Xf7k-2FwF_bW5l_Q8Y72JngVW8NWxkR573lCqW3sM18397pnz3Vpr6jc6_cN3WW7g-bTb1qDdFrW3W6ZHN13-LMPV-MKVy1RCpY4W7d9g_g1brYsgW3Tqlsc1zjyBqN5y10K7R03xDW91nVhj1fBxMxW7Jxt4R90BsY7W2wG_j51mvckpW4z-1v13LgLsbN3bD8Hm3vG_MW7YH6yr6Zxd_SW1XXCTS7Qk_sWW3PVW8W4XhKrHW5Vfv-G2mv5NLW4VF04922g_VzW1486-l3DmDBhW3DngfZ8zZQsBW3Z6ywx1mwfzSW8KmGyL4bYWyb3brq1 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21gXQmW45M8kT3X_dB6W43QCsw4cJTjYW3Z_tRK3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv6qBNbvW7gcMJ96fpV1zW5cH74N5FW8Q8V28YWw7NTzVgMGKd_P20N6PW5jqbSj4-rCj-W5MpM-36jfbnv42ywM1&v=3&_hsenc=p2ANqtz-9y3UdWHpzpJ3B1HSAr3u1lNV1iVJPbwogFUh9PVWkuAcPORByZFCWTpBWyWZ78eI1DmMNvMJP2MqyTZNCE3Mv3LNFvwbkVtmrsrXAVjhQtXXcuFkCySwK06Tq9QbnmRJKpFYlAdKjcqDqsiOEuXYcD9jw1ug&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21gXQmW45M8kT3X_dB6W43QCsw4cJTjYW3Z_tRK3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv6qBNbvW7gcMJ96fpV1zW5cH74N5FW8Q8V28YWw7NTzVgMGKd_P20N6PW5jqbSj4-rCj-W5MpM-36jfbnv42ywM1&v=3&_hsenc=p2ANqtz-9y3UdWHpzpJ3B1HSAr3u1lNV1iVJPbwogFUh9PVWkuAcPORByZFCWTpBWyWZ78eI1DmMNvMJP2MqyTZNCE3Mv3LNFvwbkVtmrsrXAVjhQtXXcuFkCySwK06Tq9QbnmRJKpFYlAdKjcqDqsiOEuXYcD9jw1ug&_hsmi=86142041 )

siddirp, 2022-07-11:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7DQ3kWFJV1-WJV7CgH-WW1gx-3Y5RwhXPW1dxNtY6TK59yW4BPBmd2vF8Y5W36DjXn7TY-f9W8YbnKD2PtvyLW4K9mRP7cYLQNW77bkx_5nxKGzW8nCYqr2dnDTGW1mpmRp2nQszSW99p6td5JxXc8W411PHl63hZslW4NR0qj2hd4b8W1HcymW82ypMRW7R09XV7N0rG0W87Xk9G5D1Z95VrTs501ffDJWW6Kyn155gmVmkW8n0XJV4THHs9W2V6nzF6dQpPmW2k5xyz10JKKpV53dvM3cDWHGW5b7R895WqGvtW3MlLPQ2BWddlW7-0MC22qPhZzW6rXQxX2WkX8BVrJwc44xPcC834xG1 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7Fp3kWGhV1-WJV7CgTzhN3Pp6h5hkfKwN98_N8c-7ZF5W54bsfV65BWFXW95SSR65jmQKKW3fms896XLHCnW4J-xZy3QnP1TW6Qj4yJ1ktTmMW6flxrq5N_Jx_W2X6_py21hVbZN6yTgKnt1ST6W2dc37X6DvH9DW6vSh0Y8FFXLGW2rl2P072NncMW14S9PR5rXynXW2bkZb11kNcgpW6HWpGy5q4Lz8VPmh648j4HhcW5pzpY2719lyNW4l5JB55MVXqfW2-1wDB8rnFzHW6cdFPs63Q0VpW8svlYt15q7pdW8gD1RH3FMR5vN5-mwVJ_qj6_W8bYmbp2BT-gBW1wDsWY95SVypW48qdcG2W23jWW38vZQw7p8m1cW2f3fjW7PvdGLW2pqS2g5llzc53dG31 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7Ck7ND_VV5X_Kf7CgR7sN1qjwF_Nf2CKW3ZPCYQ4G8gg1W3QzV681JHtpjW8KX1CH1h409JW1s3wNJ1hqnspW1tdZlq7BrHjcW1p1jsj3DnVYdW2fBpVl1h1Gx7W1C_z932ZcV_mN3XgXJr3LBgZW8WGQ-78qhw3dW7_c5q18h0wwbW1jftRx207SjrN5mNy3qLLTDpW5vcWg_4QJV3LW2VVzrg10S05wW2P3ybX7X_GrjW73F--x4WttyqW6Hmx5v4bq1CkW6ttq3J3sB3jSW2CrXy51gKL-kN8Jc-Bmfk0wrW6M6Yb45KBHGMW5GmkSw8YK54fN3msqjx4J3xCW397f6D1kjSjcW5PSYdm2Nm_pgW7ggh-Y8R4cYqW8GlkYb6HnMdFW8NDjQw53cvdRV1lDqV3RJsx3W1BB1046q4MyVW40HGbF6H6xh6W1CGKXv4Hm_pzN1LD2Y971TYGW1Qvf5x9dCtvqW4dMvg-3d0WJZW3LqmBh8gyPRSN1yHMsW3c4HdW7Y1dzF4JsSRMW3MmY9x4x676zW1C17Lg5W1KgbW4MzFn1637NFmW5nSj_p3h2YgNVb-FK91R7Q5ZW4nnTMw5Lpc8DW6fwjVB8rz7vZW8S0cWV1M-XZn3ntK1 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7Ck5js6pV3Zsc37CgWbZW2TVJK82rHmHyVQY2Rz14LfZxVgyj6881lLj_N8tbLZqyFVwZW8Rz9RX868hhKW5sBZJ351f-1FW8GVPLH6BBX90W2J-N6H27q_6YW193JfQ8jbxyNW5mSGm74XS-mpW1QHx2S69H7DFMN1qwQRysdVW43gldg839NgFW6gNbSK17f8mDW85q88_6dY8S_VTgYGJ3pQNH1W1VRSx06TVH4XW4XvRc45TlH0rN2LyqZXyZ2VKW8s9MsC633-QYW9b--jj1FWqfLW2ffsxZ5fbvbxW7s4tlv8NymXVW9fmxN63vRmnjVTVktT1xddxwW8C_26-5ZGjybW4Gv4qQ4_XLlWW1NB8gx6NFcC9W6Mw5rD3vVYDTN4QzbJLHY5R0W5xKLQz6JkGpJW90ZFRz8yCD5C38Pg1 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7F53kWF_V1-WJV7CgH2YW1WQ9cn4ZZN1sW5-P_gF4q334XN3_nv0PCRKWWW5nWdNs83D-csV9TNhr8Gsj80N2nz24MNkff9W35SHnK9lyR6-W74VnRs8MCB-yW7fbdxj79ZN9lW8tKLbg6pdTJPW3Jxl1H75M3ntW5f6wnN12vRjrW4q1MRK3dRLYPW2dCXCc8ljlsgW5Sx5Zw6W0qSgW6bWtht5Hk5pZW4W4rPq5DklzdVSzpsN48jxsgW8ylwRN3X88wQW284cDc95zlKDN2dc9vsYXtzyW7ND_Jj8ms20vW1fFmft1JCjNVMdVWStQbBKZVnXCD-8v_FTVW5VwxBK7jvYlRW1Q554l5f-l8bW5xk6bG4ZyHsR3qqq1 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWRw2DwCY0lW7wvV7h4QPwNKW72Qs8N4Mmv_6N3kz7DQ3kWFJV1-WJV7CgN0jW7DBS4k8lqXZ1W5J0zhn6TxgfDW1ScFVY8tDdkPN81RNbFp1bFwW6fmvGr67166lW2gL8px6p00SCW2lhLMs8bPVdgW5pDlpk1n10XbW4_mBhv5lbk71W1sqgV93sV-sxVKd7H15mGSwFW6Vcfgf82g0vQW7pyV0X4m2N44W7dJq7p3lc44nVVnN231cbN86W7pVMQT4GlFb2W2bxZ4l7VyLGJW46TXbt9b4Q4hW1pV_v-27nd-MW7tpjpY6MxK12W8G0qfH6Km25RW76pszm22V1rNW2trctv3rbR0yW5MGbMt8bk5DCW68bG_354jWhqW8l7MDz7gFmnD35Lf1 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21h05MW3_Wg9C4pFm3LW1Xv2Tk3R6wm-W3XWfkb3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv74H4DfW6RmfKb4Pv0WyW5t4vfN5Xg3tdV28Wbd69kwWxMGKd_P20N5gW3NwzMT93lZLgW6tNrh978kwlh510ypV1&v=3&_hsenc=p2ANqtz--ZxmgRXZrVdbrOCihvQVdauUPGQM02UDoEpwV9bF1nf4fARBavF8N_K3rfZlTswJ8SaoBJtNZYw4qmZ61ZMbDQow29RrLcnPOc9W_4_cOInoy5McFR-t2oGKkyO8GgdGWZE23bRZOJClFZf0M5VFTmklkppg&_hsmi=86142139 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21h05MW3_Wg9C4pFm3LW1Xv2Tk3R6wm-W3XWfkb3KbsvFW3R0gDH3FgZ9bW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv74H4DfW6RmfKb4Pv0WyW5t4vfN5Xg3tdV28Wbd69kwWxMGKd_P20N5gW3NwzMT93lZLgW6tNrh978kwlh510ypV1&v=3&_hsenc=p2ANqtz--ZxmgRXZrVdbrOCihvQVdauUPGQM02UDoEpwV9bF1nf4fARBavF8N_K3rfZlTswJ8SaoBJtNZYw4qmZ61ZMbDQow29RrLcnPOc9W_4_cOInoy5McFR-t2oGKkyO8GgdGWZE23bRZOJClFZf0M5VFTmklkppg&_hsmi=86142139 )

andrewpriddis, 2024-04-30:
Upgrade to the GPX-1 for improved GPS accuracy and robust solutions!

Hello  

I'd love for you to take a second and check out the newly released GPX-1. It is a dual GNSS designed module for GPS positioning, GNSS compassing and RTK positioning in a single package. You will no longer need to fit two large receivers in your design. It is an L1/L5 multiband receiver to reject interference, improve fix times, and provide an overall robust solution. The introductory price of an OEM module is $325 with volume pricing available as well. The IG-2 is an integrated module with an IMX-5 and GPX-1 with a price of $788. The RUG4 is in final development and will be released later this year.

GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXjyXV1JpgW6W3jDWdc3fF1t5W3S9mTt5dxhMfN1xVmYz5jGjMW5BWr2F6lZ3pRW12Vrwc1bFc2mW97gVXF8ZNdq_VrV8s46ph_RcW59SYs68Qxg2XW99JRVZ1VbSgMW9fF-R_73_PXKW9dNdJw8dBgbTW9g6gpF85HtcKW6RTvgC4QVmnPW7dlBP18WffVnW5RSS2L349SFXW8W2wxV4xbFt2W63xy2X6KkK09W1pG3zn976mG-W6BPJpv7hmh76W3HDz276r39pvW64MDKw1WpsnrW8ymqp_8xH6c8W57mtNX1cy06KW28Q6HT1dT4zfW1lvVL85XZJsjW3VxYlj8tP0x5W7s0rYs3qbLYvVfSPJJ3F6zkZW25BQBb5p_ZCrW3D77xv1R6-KxW6mVZhz3ZlTgVW2sZbLf36xPf4W6n1Nwv2K1H_vW1K_fXB3FF_v5VS4ZpX1VXq01W8rvz7b80X6CZW8xYbQl3jtV3rVPJnrK7_kxP5f5Q_Vsd04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXjyXV1JpgW6W3jDWdc3fF1t5W3S9mTt5dxhMfN1xVmYz5jGjMW5BWr2F6lZ3pRW12Vrwc1bFc2mW97gVXF8ZNdq_VrV8s46ph_RcW59SYs68Qxg2XW99JRVZ1VbSgMW9fF-R_73_PXKW9dNdJw8dBgbTW9g6gpF85HtcKW6RTvgC4QVmnPW7dlBP18WffVnW5RSS2L349SFXW8W2wxV4xbFt2W63xy2X6KkK09W1pG3zn976mG-W6BPJpv7hmh76W3HDz276r39pvW64MDKw1WpsnrW8ymqp_8xH6c8W57mtNX1cy06KW28Q6HT1dT4zfW1lvVL85XZJsjW3VxYlj8tP0x5W7s0rYs3qbLYvVfSPJJ3F6zkZW25BQBb5p_ZCrW3D77xv1R6-KxW6mVZhz3ZlTgVW2sZbLf36xPf4W6n1Nwv2K1H_vW1K_fXB3FF_v5VS4ZpX1VXq01W8rvz7b80X6CZW8xYbQl3jtV3rVPJnrK7_kxP5f5Q_Vsd04 )

When paired with the IMX-5 you have a low cost tactical grade full Inertial Navigation System (INS). The solution can be further enhanced by enabling RTK positioning for cm level accuracy as well as enabling GNSS compassing for high accuracy heading. The IMX-5 is $275 with volume pricing available as well.

IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXjyXV1JpgW6W3jDWdc3fF1t5W3S9mTt5dxhMfN1xVmYg5jGjMW50kH_H6lZ3pZW7TQYJ69cPxMzW7vTnw87n2knbW7MxPZ61lBqfCW6zR1cY5cnPL6W2fHYZR7QS7NKW6BRsTk4JnzT2W7jQWzK5Km_xDW71-j078qdM-0W6vxQ337GgcjBV_QzwV8TFKbGW1_Wdxh3FDnmbW8Kqw-d3NdJYWW32-sRv1zfycwN8PMGlbmdYdRW17BW7q5X63LhW1ck5Xq2PPHlbV2qxgY4y59tdW20rVsQ6bMVZ2W2tkyfK5hnm7TW6W57SW5J6FXTW6Ss8Hl1xdcsfN733yVPGMhS_W2mtQtB3WTfdbW7-h8sJ1Dg7GnW1nHbkx68tBmZV-VNb87DmNKPW8SHjr683YTSbW7TWm2g59VGz1W2NPW_M9kW21qW87_5zM3x3Y50VB9GBx95YmvnW4flYLY1T6L8hf6XT_kq04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXjyXV1JpgW6W3jDWdc3fF1t5W3S9mTt5dxhMfN1xVmYg5jGjMW50kH_H6lZ3pZW7TQYJ69cPxMzW7vTnw87n2knbW7MxPZ61lBqfCW6zR1cY5cnPL6W2fHYZR7QS7NKW6BRsTk4JnzT2W7jQWzK5Km_xDW71-j078qdM-0W6vxQ337GgcjBV_QzwV8TFKbGW1_Wdxh3FDnmbW8Kqw-d3NdJYWW32-sRv1zfycwN8PMGlbmdYdRW17BW7q5X63LhW1ck5Xq2PPHlbV2qxgY4y59tdW20rVsQ6bMVZ2W2tkyfK5hnm7TW6W57SW5J6FXTW6Ss8Hl1xdcsfN733yVPGMhS_W2mtQtB3WTfdbW7-h8sJ1Dg7GnW1nHbkx68tBmZV-VNb87DmNKPW8SHjr683YTSbW7TWm2g59VGz1W2NPW_M9kW21qW87_5zM3x3Y50VB9GBx95YmvnW4flYLY1T6L8hf6XT_kq04 )

We also still offer our ublox F9P based products that are integrated in systems all over the world.

I'd love to set up a meeting for this week. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXjyXV1JpgW6W3jDWdc3fF1t5W3S9mTt5dxhMfN1xVmZM3l5QzW7Y8-PT6lZ3pFVJJWXc2cx2rKW11t9VQ3rwLnRW84Ytqj78D943W6vQddl2N3MvcW764qMN5KHHYBW2WdQ5M1vNL7KW8C-lRZ4SzQlPW4D5Qk03tH98tW396QJz82LjXNW1ZFJP17LCVN6W3v38HQ4_b0sZW1Dzbg6693ZyKW6QChb08rSZJ3W4wQY-w6knfr6W7fgKQ42vyz60W2TpH0c19n2LcN5--5CKGYzL0W5jYdF45D16h5W5QWqMK4fMRThW8Cm7Hc51FvX8W1mzQjt650rslW4_QkLR1p33CsN55pSt6_-kV-N7Rq7fgZ09-bW5PcQB56279ggN4gqZz5LkY_5f5YTmWH04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-SqW3zdW2b2-MKfHW3JRrlj2HDFTCW4fzx_X3P23xWW2t04d52-zPDrW34h6g-21cgydW4cLtSL2sMLygW2CMHqG3XPSwbW47B81D2WvyfdW1L7Y233675nvW3hXcGk3djTCWW4fjTG12Yxw_9W21hmNP3_SLM4W22TgNs3C0dj6W3K2Xdj3db8M2W1LyZDf3g1JYmW230Htg2vXG7GW2p6H823hZlcJW21nSP_1BKJ-fW3NGkB-3Fds79W2sPqcV4ffXB0W1VpcqZ3bywzBW3858_33DZXQ7W3dsp4v2MT750W47RyHV2n_-D8W1VnLxv36yyTJW2vMXVt47C0hRW45WvT92WFyPmW3f-Mly2WnjGYW2WNZ8H4ffvXRW1W_NZX1Z6g7-W2TS8G03z49hJW1X6npw3j615gW41BQ0W2MCs7VW4ffypg1X5_fhW1Bg_KC1ZpfC7W3dbZYh2FM3LYW21m79n2HH7DVW3dwDgN41PBBgW2CFwfS1Q48wKW1WYswC3K8QmMW4kM8TG1VsVM8W4rCs923_wt2yW3ZXHGk2qLhqFW2z-bxL2TwF8rW4pnZl21NnbXQW211dkN22XffcW3M4nSx2zZhh3W2qW2CB3NMj8vW3V_30l3T4zQxW4pp1_l3h-zGSW3Zw9JT25hTJYW3d13Q03QBhP8f3M9mc704&_hsenc=p2ANqtz-8ZOEw2pwDvz__zDHahVqyYL3fdPAfatg3_yUQ8vGxGD3Ao_mQKRXONmieoSrFTojI7y8eDcjRLtrF5PltxXQxygRVUS-KEHu2Mr1Zk4dBDdJmCfePleTyEN4XS7JYbuuYQnZLtI2uRtyyhlWzOYgVpYyBPPA&_hsmi=305054157 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-SqW3zdW2b2-MKfHW3JRrlj2HDFTCW4fzx_X3P23xWW2t04d52-zPDrW34h6g-21cgydW4cLtSL2sMLygW2CMHqG3XPSwbW47B81D2WvyfdW1L7Y233675nvW3hXcGk3djTCWW4fjTG12Yxw_9W21hmNP3_SLM4W22TgNs3C0dj6W3K2Xdj3db8M2W1LyZDf3g1JYmW230Htg2vXG7GW2p6H823hZlcJW21nSP_1BKJ-fW3NGkB-3Fds79W2sPqcV4ffXB0W1VpcqZ3bywzBW3858_33DZXQ7W3dsp4v2MT750W47RyHV2n_-D8W1VnLxv36yyTJW2vMXVt47C0hRW45WvT92WFyPmW3f-Ml [... cut ...] XRW1W_NZX1Z6g7-W2TS8G03z49hJW1X6npw3j615gW41BQ0W2MCs7VW4ffypg1X5_fhW1Bg_KC1ZpfC7W3dbZYh2FM3LYW21m79n2HH7DVW3dwDgN41PBBgW2CFwfS1Q48wKW1WYswC3K8QmMW4kM8TG1VsVM8W4rCs923_wt2yW3ZXHGk2qLhqFW2z-bxL2TwF8rW4pnZl21NnbXQW211dkN22XffcW3M4nSx2zZhh3W2qW2CB3NMj8vW3V_30l3T4zQxW4pp1_l3h-zGSW3Zw9JT25hTJYW3d13Q03QBhP8f3M9mc704&_hsenc=p2ANqtz-8ZOEw2pwDvz__zDHahVqyYL3fdPAfatg3_yUQ8vGxGD3Ao_mQKRXONmieoSrFTojI7y8eDcjRLtrF5PltxXQxygRVUS-KEHu2Mr1Zk4dBDdJmCfePleTyEN4XS7JYbuuYQnZLtI2uRtyyhlWzOYgVpYyBPPA&_hsmi=305054157 )

andrewpriddis, 2024-04-30:
Upgrade to the GPX-1 for improved GPS accuracy and robust solutions!

Hello  

I'd love for you to take a second and check out the newly released GPX-1. It is a dual GNSS designed module for GPS positioning, GNSS compassing and RTK positioning in a single package. You will no longer need to fit two large receivers in your design. It is an L1/L5 multiband receiver to reject interference, improve fix times, and provide an overall robust solution. The introductory price of an OEM module is $325 with volume pricing available as well. The IG-2 is an integrated module with an IMX-5 and GPX-1 with a price of $788. The RUG4 is in final development and will be released later this year.

GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWHBrZVg0DdW3-m5ZQ7JLPtpW4Mw1s55dxhMxN1f4_X45jGjMW5BWr2F6lZ3pCW83ntW68ln_36W4lwGLm3b9_XXVt5Z124qv-DdW1CrfXv1jK_zLW2mzPmj5BhPngW8GCBLY3fhZQLVgPFCr48Gw4dW15m0nq3Zvn_XW1NT15s1cPJ7FW8PH3bd41bFGxW9lVNzt8bZN5qN3DYzXB8HD8VW8pg_dR7VZp_WW356jjH73LN92W8n26Vl5_VZg_W4BKM9575ynQtW5qPsrw3Y2D64W1lrj1M5q6Z47W77xYxT88KhGdW5QTr5314fWYzVcd-mx3b2s0PW54-Tks6H0R3MW1bgsN22Mhbh0W8lJkmp6GZmT9VcLN8z6-0bw5W9m398l7vhWQbW65HCWF1hpNhrW8j7s6K30hf5vW5VqcrV3nHMYwW4dWDtd5sWprQW2x30Fs7xt2WhW8MJ97H2Qbj85W2yQ6SQ4QmPcBW2RW7R61yQG0pf4LwvLK04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWHBrZVg0DdW3-m5ZQ7JLPtpW4Mw1s55dxhMxN1f4_X45jGjMW5BWr2F6lZ3pCW83ntW68ln_36W4lwGLm3b9_XXVt5Z124qv-DdW1CrfXv1jK_zLW2mzPmj5BhPngW8GCBLY3fhZQLVgPFCr48Gw4dW15m0nq3Zvn_XW1NT15s1cPJ7FW8PH3bd41bFGxW9lVNzt8bZN5qN3DYzXB8HD8VW8pg_dR7VZp_WW356jjH73LN92W8n26Vl5_VZg_W4BKM9575ynQtW5qPsrw3Y2D64W1lrj1M5q6Z47W77xYxT88KhGdW5QTr5314fWYzVcd-mx3b2s0PW54-Tks6H0R3MW1bgsN22Mhbh0W8lJkmp6GZmT9VcLN8z6-0bw5W9m398l7vhWQbW65HCWF1hpNhrW8j7s6K30hf5vW5VqcrV3nHMYwW4dWDtd5sWprQW2x30Fs7xt2WhW8MJ97H2Qbj85W2yQ6SQ4QmPcBW2RW7R61yQG0pf4LwvLK04 )

When paired with the IMX-5 you have a low cost tactical grade full Inertial Navigation System (INS). The solution can be further enhanced by enabling RTK positioning for cm level accuracy as well as enabling GNSS compassing for high accuracy heading. The IMX-5 is $275 with volume pricing available as well.

IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWHBrZVg0DdW3-m5ZQ7JLPtpW4Mw1s55dxhMxN1f4_WP5jGjMW50kH_H6lZ3p_W9g2WfJ5rgDpvVDj64C2nTl6TW80DHJc6lWTlVW8ZtXnY5TXLN3W9jvlJV8lqlXCW54-99n1-g-8MN11FDzXNXQ0BN5MNFCrX7xRRW5pVSfC6n20HMW3Hv8Bp3Mfqn8W7MTMLh7ZVxLVW74Z4ml1QzpSJMCJtt6-nSr0W2Y6Z-F1s0wHWMlkVKM6P5lBW4KZQZH2Rgr5PW299MdT6gqCF3W1VjsR38Khl1mW35qFHW8ZVMTpW7ynxRJ226snpW6N6gS71Fzp1rW613CWM70C03jW5C9dy32nXHHwVLVz667NPDJCW48Bld_1lH2LhW5b-mCk6B9PXCW8nZDRT8prNYkW6NQCP198dLV3W2fN1Lx3sH2mzW2Pk3YX2Xw_ghW6L9fdc8pb0lpN2g7B0TD4tnGf6q2j9b04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWHBrZVg0DdW3-m5ZQ7JLPtpW4Mw1s55dxhMxN1f4_WP5jGjMW50kH_H6lZ3p_W9g2WfJ5rgDpvVDj64C2nTl6TW80DHJc6lWTlVW8ZtXnY5TXLN3W9jvlJV8lqlXCW54-99n1-g-8MN11FDzXNXQ0BN5MNFCrX7xRRW5pVSfC6n20HMW3Hv8Bp3Mfqn8W7MTMLh7ZVxLVW74Z4ml1QzpSJMCJtt6-nSr0W2Y6Z-F1s0wHWMlkVKM6P5lBW4KZQZH2Rgr5PW299MdT6gqCF3W1VjsR38Khl1mW35qFHW8ZVMTpW7ynxRJ226snpW6N6gS71Fzp1rW613CWM70C03jW5C9dy32nXHHwVLVz667NPDJCW48Bld_1lH2LhW5b-mCk6B9PXCW8nZDRT8prNYkW6NQCP198dLV3W2fN1Lx3sH2mzW2Pk3YX2Xw_ghW6L9fdc8pb0lpN2g7B0TD4tnGf6q2j9b04 )

We also still offer our ublox F9P based products that are integrated in systems all over the world.

I'd love to set up a meeting for this week. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MWHBrZVg0DdW3-m5ZQ7JLPtpW4Mw1s55dxhMxN1f4_Yg3l5QzW7Y8-PT6lZ3pxW52nFtV5txH83W3lFmR_987dNRVj6G_M4RjTqvW8M0qFy6HdkwRW5JRsFt6d4SG1W1YghrM9jTMN6W7m7ZS_83nlN1W1tZsnR779d4CW1LX3nm2qHHwVW8wRPC88-31ytW6vfN2Q5brfNsVBxrzQ5ZRk1HW71MxbC5Q5wbgW8mXRHC6N7PpWW2DcYQv8N_53_W3gVTDT6Yvs5vW2DDCzY8qy7VyW8NZKWt4b9w_XW4zMHFm2WV_yGW5S3GDT6VfJgmW3S4r227240C8W2LNx2s6Qzvg1W22XVsJ2lT2q8W8h_Y-Q8Y7Q8zVND5Vr7cs8mXW5sg4gl1_GtMtd2w06n04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-RVW4tdKhW4hcHf1W34mjPl2F-_rmW2Yx2Pp2vD2-XW3_XY8V1Bg7wNW3P8QPx2RCnPtW3LYSSy43Qd8CW47W9RK1Bfh-xW3ZxvbF21ph2MW4ffw1r2FC_YHW2CTZz01By1VBW2RNZKC45xrZ4W3g8Lhx2zxVw2W43CT8B2YPcsMW2KlpQC1VzK_VW1SB0Qf1V3m_yW4rnMfy47BxM0W2zKPMW1QhRdpW3dsQM92FZL27W3zjjBW2p3kWNW4kf7Cp3ZsN4cW4crD_530n8h3W2Fvkc22MN-CJW1Nn-5t4ktYxDW2-lL6_1_t7pmW47Rxhh4ppRzpW1LDgJL3QK00kW3JWfRS45GxmRW1VtPc638smy6W1XskZX34z9pBW215T9c3BM5zCW2xJqTl4tjtV4W3H5G8p2122cZW2Kp1981Vq4FyW3QDT2z2TP7K3W1_8t9T3M0hNSW2FwdDy2RrKwKW1BF6cQ1V1dhcW1VjbpW2-KbNsW2CFtHQ4cPsPhW3SLX2F3jnG2TW219y1v2p6HzTW25fkB41_jrxZW4tdLNF2KG8r5W2Hy0XD1Qs_j5W1Srxmj2p5qDWW2YrZcx3ZxTSdW2311Vv3jnBmkW3_KhpN34jsVNW2-pfzk3jvQP9W1X6VPz49KYgRW43HYSF1ZwT15W2sR4_H2FKRJdf23hzsy04&_hsenc=p2ANqtz-9jZU8VW1FjqqB7AOXADlOuCnFDSLoEr1VC4hU0UdFnvEkoZXdmAPQ2lb3PEg-t0EEiASCpBN2zWg1-MRlTc2QN1Ah4OH3YhUOHw4HqIxl8-9OoQkfG0KoXs9AjReohWuD2K5RCaft_Fg9-BdlFUQFMPpzSsQ&_hsmi=305054157 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-RVW4tdKhW4hcHf1W34mjPl2F-_rmW2Yx2Pp2vD2-XW3_XY8V1Bg7wNW3P8QPx2RCnPtW3LYSSy43Qd8CW47W9RK1Bfh-xW3ZxvbF21ph2MW4ffw1r2FC_YHW2CTZz01By1VBW2RNZKC45xrZ4W3g8Lhx2zxVw2W43CT8B2YPcsMW2KlpQC1VzK_VW1SB0Qf1V3m_yW4rnMfy47BxM0W2zKPMW1QhRdpW3dsQM92FZL27W3zjjBW2p3kWNW4kf7Cp3ZsN4cW4crD_530n8h3W2Fvkc22MN-CJW1Nn-5t4ktYxDW2-lL6_1_t7pmW47Rxhh4ppRzpW1LDgJL3QK00kW3JWfRS45GxmRW1VtPc638smy6W1Xsk [... cut ...] pBW215T9c3BM5zCW2xJqTl4tjtV4W3H5G8p2122cZW2Kp1981Vq4FyW3QDT2z2TP7K3W1_8t9T3M0hNSW2FwdDy2RrKwKW1BF6cQ1V1dhcW1VjbpW2-KbNsW2CFtHQ4cPsPhW3SLX2F3jnG2TW219y1v2p6HzTW25fkB41_jrxZW4tdLNF2KG8r5W2Hy0XD1Qs_j5W1Srxmj2p5qDWW2YrZcx3ZxTSdW2311Vv3jnBmkW3_KhpN34jsVNW2-pfzk3jvQP9W1X6VPz49KYgRW43HYSF1ZwT15W2sR4_H2FKRJdf23hzsy04&_hsenc=p2ANqtz-9jZU8VW1FjqqB7AOXADlOuCnFDSLoEr1VC4hU0UdFnvEkoZXdmAPQ2lb3PEg-t0EEiASCpBN2zWg1-MRlTc2QN1Ah4OH3YhUOHw4HqIxl8-9OoQkfG0KoXs9AjReohWuD2K5RCaft_Fg9-BdlFUQFMPpzSsQ&_hsmi=305054157 )

andrewpriddis, 2025-01-02:
IK-1: For rapid evaluation and development.

 

We heard your feedback that evaluating and developing using an IMX-5 or GPX-1 module can be tedious. The parts are designed to be surface mounted for deep integration into a system. This makes accessing pins for evaluation and prototyping difficult. To address this concern we developed the IK-1. It has the following features to ease the initial interfacing:

- 0.1" pitch header for convenient interface.
- Interfaces with standard breadboard.
- Onboard Micro USB connector
- Onboard voltage regulation.
- Dual U.FL connectors for GPX GNSS antennas

There is a version for working with the IMX-5 as well as a version to work with the GPX-1

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgqF3l5QzW8wLKSR6lZ3mBW3DtwYx66kCb-W7YGf0Q49HwcBW10D_w544pJMhW849-JR1TyRDwW5JRQsN1xwTvjW104xJ57XZKVQW5hpgvX82T3_0W7mhT1p2h4YzXW67Xt-G5910WkW9gFC9k1gcq_xVD45fj8KdxdZW39-r4h46w8QzM9Qx0jpnL1yW7vGtnT1mzHNQW484Xd36RnBJ-W2_BBKj5g8Ww3W85KCMT6cxsWJN4zkF27f5ZrYW7sG_tZ8R7FjbW2X9T3b8Cn1GnW5WbNc26S6VtKW2PFYgQ8hSfMNVsT6bj1ChqSsW6GVZSl7Gyx0BW7mjDDX5MyM_rW3Vw9RT8SzbSFW4R-Mrr97SHcbW4l9vDM17BKwjf7fwHrz04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgqF3l5QzW8wLKSR6lZ3mBW3DtwYx66kCb-W7YGf0Q49HwcBW10D_w544pJMhW849-JR1TyRDwW5JRQsN1xwTvjW104xJ57XZKVQW5hpgvX82T3_0W7mhT1p2h4YzXW67Xt-G5910WkW9gFC9k1gcq_xVD45fj8KdxdZW39-r4h46w8QzM9Qx0jpnL1yW7vGtnT1mzHNQW484Xd36RnBJ-W2_BBKj5g8Ww3W85KCMT6cxsWJN4zkF27f5ZrYW7sG_tZ8R7FjbW2X9T3b8Cn1GnW5WbNc26S6VtKW2PFYgQ8hSfMNVsT6bj1ChqSsW6GVZSl7Gyx0BW7mjDDX5MyM_rW3Vw9RT8SzbSFW4R-Mrr97SHcbW4l9vDM17BKwjf7fwHrz04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgqF3l5QzW8wLKSR6lZ3lgW1cl7639d7lCgW1G7pkL2Qx5zvN3HFxxqWv16BW6vx4vn32WSWCVGTxFL7N57P_W8tFkPv4vrtq-W68tp9y6K-1-TW3bNv265ckHQlW5f2w697Ck7mvW7bR20l6MvD5GW9cTjZX1HGcZVN4q5kKxQBjhRW8Q-q8Y2r8pM4N3M5-x1JV8CsN4R-9n7SkH4JW3nxbWL2tQG_NN6SFmyTd-KJkW6NlsTb3CZ0mNW4SzgzP5qllcFW4KYSq_1DZgGKW7npFfv2Bk9NCN88syBpmZtdfW70LLZw2Jpr3jW441l163WmB-mW6n_l9B5SjRRRW4xmYYj2wctxqW2t8wLF8117r9W5YVxTm9jqdZnf1qsyYC04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgqF3l5QzW8wLKSR6lZ3lgW1cl7639d7lCgW1G7pkL2Qx5zvN3HFxxqWv16BW6vx4vn32WSWCVGTxFL7N57P_W8tFkPv4vrtq-W68tp9y6K-1-TW3bNv265ckHQlW5f2w697Ck7mvW7bR20l6MvD5GW9cTjZX1HGcZVN4q5kKxQBjhRW8Q-q8Y2r8pM4N3M5-x1JV8CsN4R-9n7SkH4JW3nxbWL2tQG_NN6SFmyTd-KJkW6NlsTb3CZ0mNW4SzgzP5qllcFW4KYSq_1DZgGKW7npFfv2Bk9NCN88syBpmZtdfW70LLZw2Jpr3jW441l163WmB-mW6n_l9B5SjRRRW4xmYYj2wctxqW2t8wLF8117r9W5YVxTm9jqdZnf1qsyYC04 )

Your work with the breakout boards is your first step to integrating  both units into a low cost INS solution. Be sure to review the datasheets:

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgnT5jGjMW50kH_H6lZ3n3W3GrdtG8XZLKzW2VfzY93L4chhW7HFrnp222Gp9W70B8pz6YmznhW76T28F6y19cZW2dt6K_4kBTDnW1jQfWb30LMGRN7TM2v65gwLsVY1g9V8wMVf5W7kPZpV85DbntN4Lwzrvr8d3gW4xgqmn4VLJ6_W2G1Qlz7WpGRZW6NSKZ610g9_-W3FgHjJ1pSHmYN446kzwzt0GKW2Kb9P735WVK6W5yfx3f61gcSPW4BKfZh60ncdKW2f5wTW5BDCm5W1ZFbSG5Yy6BCW262LKW2pD5blW7G-K_w464D29N26n4Tw63FhlW6_X9bx67wX8wW8_Y1Vw3bhn4HVcSTKy2hQPRWW1xnmTz5J8h6WVpy9ld9g2ynBW98ffS-2-d_DqN6sSx6cGG4NpW6PzK6G91qgBMf4Jp9BP04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgnT5jGjMW50kH_H6lZ3n3W3GrdtG8XZLKzW2VfzY93L4chhW7HFrnp222Gp9W70B8pz6YmznhW76T28F6y19cZW2dt6K_4kBTDnW1jQfWb30LMGRN7TM2v65gwLsVY1g9V8wMVf5W7kPZpV85DbntN4Lwzrvr8d3gW4xgqmn4VLJ6_W2G1Qlz7WpGRZW6NSKZ610g9_-W3FgHjJ1pSHmYN446kzwzt0GKW2Kb9P735WVK6W5yfx3f61gcSPW4BKfZh60ncdKW2f5wTW5BDCm5W1ZFbSG5Yy6BCW262LKW2pD5blW7G-K_w464D29N26n4Tw63FhlW6_X9bx67wX8wW8_Y1Vw3bhn4HVcSTKy2hQPRWW1xnmTz5J8h6WVpy9ld9g2ynBW98ffS-2-d_DqN6sSx6cGG4NpW6PzK6G91qgBMf4Jp9BP04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgqF3l5QzW8wLKSR6lZ3lgW1cl7639d7lCgW1G7pkL2Qx5zvN3HFxxqWv16BW6vx4vn32WSWCVGTxFL7N57P_W8tFkPv4vrtq-W68tp9y6K-1-TW3bNv265ckHQlW5f2w697Ck7mvW7bR20l6MvD5GW9cTjZX1HGcZVN4q5kKxQBjhRW8Q-q8Y2r8pM4N3M5-x1JV8CsN4R-9n7SkH4JW3nxbWL2tQG_NN6SFmyTd-KJkW6NlsTb3CZ0mNW4SzgzP5qllcFW4KYSq_1DZgGKW7npFfv2Bk9NCN88syBpmZtdfW70LLZw2Jpr3jW441l163WmB-mW6n_l9B5SjRRRW4xmYYj2wctxqW2t8wLF8117r9W5YVxTm9jqdZnf1qsyYC04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgp85jGjMW5BWr2F6lZ3n_N5bgG8fsNNNwW5mZv126bGPdqW6HSN6N8hjQJnW2bjrys9cxHjYW5HR4GC8vxLrLW1JdgfB4kKqGyW6LjZFN5DWKf_W7VW5nf3Y-JX9VkvkZ11Gq2vTN6Zs6F5GhYSNMbprQZMJXqCW2g_fvZ7cSB1jW7ZS2RJ81rNxLW1DKYpX8PWT1SW2d9Jqg1dYg3jVvlmMj5dVPHzN2TwfzjsD70JW54qMlS5lhfTyW2lTx3h235fVjW3SsC8G51lYyNW2L5nF16khB-GW8KkG343PdbyVW1HK--h4zXcgLW3Ffhzk83PqT3N1sg1tNZ8Qf1W2S5QtZ5KMhS5W4XMWkZ4PdzTsW8Z8ybc23H1PYW1rS51t6PfTlkW7J4kjN1PDjDFW15ckRn5kgVBZW2T45tn93NhHHV2qR_T3pN8LXW67M0LM19LdMBf3X6Yqn04 )

I'd love to set up a meeting in January. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

***@***.***

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVskZX40RV_yW87tjRQ7jrynJW1HN78y5qkpWFN13dgql3l5QzW7Y8-PT6lZ3p4W3zpN4337XRhLW2jqhb226GnFzW44KtpM3G1MMsW748Mtg1wJNpxW2tzl898jZLjQW33p3L03S58KpW23fKlv2ps2j0W3nB8461ZQJDdW8nmd113JzQxmW5K0_tl3ccXB8W5TCxyZ8VH5g6W6ZWjjl3CTf5bN7h3PXNsFSnfN3F5gygXhCHgW68fVPP5sQXRmW6hQJZY8tmbx [... cut ...] StW3bhQht3K8Qh1W2q-xhN2-By0qW2MV0vS32jbYkW3QMwWt41SYQ1W2MXTLR2WGkc-W1BLw7S1NDwdrW3W37WN365nSBW3D-lWs3DN5zLW34D1GJ2385BGW2p3PMy1XpqQyW3LSX7n3C12LnW1Vk3DT1S13CVW4clf_J34qhsmW2RzJBs43P29qW2zJ9YV2zVZLCW4hfMh41Sxs_6W3jbdd42HXXtZW1VgwR021jhbtW49KYy-4ms3bTW3hZln-3bxJGlW1Bhlp42Wg6gjW2Pvr1W30kb33W45NmbB2WH6rvf3GQPzN04&_hsenc=p2ANqtz-86NPKuwP0n8gTxDOvYrGTrParkSV76au4na5Is53xktzOwJCLA_8L2usbUaMjWnXbbCpqOJAaeh3o33Sc3D24Cyw7tgPxaz31BLzOvh-DN8YchF_u--jDI_GdXXvgBUi1THAHpj-vMdrCNc3cgr2xqJuxjGQ&_hsmi=340288392 )

closed 2025-01-02 by AchmadFathoni