inertialsense/inertial-sense-sdk #270: RTK outages

FloopCZ, 2022-08-08:
Dear InertialSense,

We are using multiple of your units on moving vehicles with unstable network connection and we have an issue with the RTK client.

According to GPS status (`DID_GPS1_POS.gpsPos.status`), the InertialSense SDK drops from RTK level (Fix/Float) to 3D Fix level for long time periods and it does not attempt to reconnect. Unfortunately, I was not able to find a method providing the timestamp of the last received data from RTK or the connection status. What approach would you recommend to ensure a robust connection?

Yours Sincerely, Filip

siddirp, 2022-08-09:
Filip,
The statement "unstable network connection" is the entire crux of the issue. RTK base corrections need to be reliably received. Within a second would be optimal. The RTK solution can only be sustained for maybe a minute (in ideal conditions) without consistent base correction messages. Monitor DID_GPS1_RTK_POS_REL.differentialAge. This is a counter (in seconds) of the time since the last base correction message was received.

I don't know your exact application but it is vital that you implement a reliable communication method. This can be done using radios, GSM modems, Wifi using a computer in the loop, Wifi using the EVB2 Wifi module, etc.

FloopCZ, 2022-08-10:
Dear Andrew,

Thank you for reaching out. We are using GSM modem, but with any kind of technology, we sometimes lose connection for a brief moments - e.g., tunnels, underground parking, etc. I really like the idea of `differentialAge` and I will add that one to our monitoring.

I have browsed through the code and I have a few remarks.
1) When TCP connection to RTCM server fails and `ISTcpClient` e.g., fails to read, it disconnects from the RTCM server (see e.g. [here](https://github.com/inertialsense/inertial-sense-sdk/blob/a04077f075d432dec6dbccfabefdddf68ace2030/src/ISTcpClient.cpp#L317))
2) I have not found a way to find out about the socket error and RTCM disconnecting using the public API of the `InertialSense` class. I could, theoretically use `GetClientServerByteCount()`, but since I do not know when it should be zero, it is a bit difficult to use. I could also use `m_clientStream` member.

So my specific questions would be - would it be a good idea to add some kind of reconnection support, where `InertialSense` automatically reconnects in case of a socket close? It would not be that simple though, as the reconnection should probably happen asynchronously and under mutex, because it should not block the broadcast processing. How does `EvalTool` do that?

Thank you for your time.