inertialsense/inertial-sense-sdk #325: Number of used satellites with/without RTK

FloopCZ, 2022-11-24:
In our deployment, we are observing a curious pattern. During RTK outages, the number of "used satellites" increases (to about 15-20) and when the RTK connection restores, the number of used satellites decreases (to about 5-15). Is this the expected behavior? Thank you.

[image]

Legends description:
RTK differential age: `d.gpsRtkRel.differentialAge`
Accuracy: `d.gpsPos.hAcc`
Status: `d.gpsPos.status & GPS_STATUS_FIX_MASK`
Satellites: `d.gpsPos.satsUsed`

waltjohnson, 2022-11-28:
Yes.  The RTK solution has a more stringent requirement for satellites used
in the solution when in RTK fix.  Thus, the non-RTK solution will use more
satellites.  However, 10 or less satellites is somewhat low for either
solution.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Inertial Sense Support Te**am*
3000 S Sierra Vista Way Suite 2, Provo, UT 84606
801.610.6771 (Office)
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>
Notice: This transmittal is a confidential client communication or may
otherwise be privileged or confidential. If you are not the intended
recipient, you are hereby notified that you have received this transmittal
in error and that any review, dissemination, distribution or copying of
this transmittal is strictly prohibited. If you have received this
communication in error, please notify me immediately at 1.801.609.3697 and
immediately delete this message and all its attachments, if any.

FloopCZ, 2022-11-30:
Ok, thank you for your confirmation. The low number of satellites is a problem we have difficulty solving. We have multiple cars driving through a city and even though we have the antenna placed on the roof of the vehicle, there are parts of the city where we rarely have RTK Fix. Do you have some tips on setting and antenna placement?

[image]

Also, if we set `insDynModel` to `INS_DYN_MODEL_GROUND_VEHICLE` the accuracy is much worse, so we use `INS_DYN_MODEL_AIRBORNE_4G` instead.

steezeman, 2023-04-17:
@FloopCZ please feel free to reopen if you're still having issues, I'm cleaning house on issues.