inertialsense/inertial-sense-sdk #407: Confusion about External GPS TIMEPULSE source in ioConfig

AchmadFathoni, 2023-07-05:
https://github.com/inertialsense/inertial-sense-sdk/blob/dafe7597851b1fb3be95699f3d8a82aa6c7b09f1/src/data_sets.h#L2211-L2215

My device is my RUG-2-G2.

According to the comment, value 0 means internal and 1 means disabled. But it contradicts with the code below that comment which state value 0 is `IO_CFG_GPS_TIMEPUSE_SOURCE_DISABLED` and 1 is `IO_CFG_GPS_TIMEPUSE_SOURCE_GPS1_PPS_PIN20`. 

If I use `IO_CFG_GPS_TIMEPUSE_SOURCE_GPS1_PPS_PIN20` or `IO_CFG_GPS_TIMEPUSE_SOURCE_GPS2_PPS`. Does that mean I use internal source for my PPS? 

I just want to use internal source for my PPS time pulse output at PIN 5 i, which value should I use?

waltjohnson, 2023-07-06:
Achmad,

Please ignore the comment and follow the enum values.  The comment is
wrong.  0 = disabled.

 /** GPS TIMEPULSE source */
IO_CFG_GPS_TIMEPUSE_SOURCE_BITMASK = (int)0x0000E000,
IO_CFG_GPS_TIMEPUSE_SOURCE_OFFSET = (int)13,
IO_CFG_GPS_TIMEPUSE_SOURCE_MASK = (int)0x00000007,
IO_CFG_GPS_TIMEPUSE_SOURCE_DISABLED = (int)0,
IO_CFG_GPS_TIMEPUSE_SOURCE_GPS1_PPS_PIN20 = (int)1,
IO_CFG_GPS_TIMEPUSE_SOURCE_GPS2_PPS = (int)2,
IO_CFG_GPS_TIMEPUSE_SOURCE_STROBE_G2_PIN6 = (int)3,
IO_CFG_GPS_TIMEPUSE_SOURCE_STROBE_G5_PIN9 = (int)4,
IO_CFG_GPS_TIMEPUSE_SOURCE_STROBE_G8_PIN12 = (int)5,
IO_CFG_GPS_TIMEPUSE_SOURCE_STROBE_G9_PIN13 = (int)6,

Use 1 for internal source (IO_CFG_GPS_TIMEPUSE_SOURCE_GPS1_PPS_PIN20 =
(int)1,).

[image: Inertial Sense] <http://www.inertialsense.com/>
*Walt Johnson*
CTO & Founder
3000 S Sierra Vista Way Suite 2, Provo, UT 84606
801.610.6771 (Office) *|* 801.380.0819 (Cell)
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

Notice: This transmittal is a confidential client communication or may
otherwise be privileged or confidential. If you are not the intended
recipient, you are hereby notified that you have received this transmittal
in error and that any review, dissemination, distribution or copying of
this transmittal is strictly prohibited. If you have received this
communication in error, please notify me immediately at 1.801.609.3697 and
immediately delete this message and all its attachments, if any.

AchmadFathoni, 2023-07-06:
Noted

closed 2023-07-06 by AchmadFathoni