inertialsense/inertial-sense-sdk #465: Cannot run EvalTool debian on Ubuntu, v1.10

jkaob, 2023-10-09:
Hi, after running the installation script and installing dependencies
`libqt5opengl5, libqt5webenginewidgets5, libqt5charts5, libqt5sql5, libglu1-mesa`, and rebooting,
the EvalTool debian is still not running:

```
$ sudo EvalTool 
EvalTool: symbol lookup error: EvalTool: undefined symbol: _ZN8QtCharts6QChart8setThemeENS0_10ChartThemeE, version Qt_5
$ sudo apt-get install libqt5opengl5 libqt5webenginewidgets5 libqt5charts5 libqt5sql5 libglu1-mesa
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libglu1-mesa is already the newest version (9.0.2-1).
libglu1-mesa set to manually installed.
libqt5charts5 is already the newest version (5.15.3-1).
libqt5webenginewidgets5 is already the newest version (5.15.9+dfsg-1).
libqt5opengl5 is already the newest version (5.15.3+dfsg-2ubuntu0.2).
libqt5opengl5 set to manually installed.
libqt5sql5 is already the newest version (5.15.3+dfsg-2ubuntu0.2).
libqt5sql5 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
```

Seems like some compatability issues with qt5. Are there any specific versions of the packages required?

Using Ubuntu 22.04.

Thank you,

andrewpriddis, 2023-10-09:
Ubuntu 22.04 is not supported at this time.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

andrewpriddis, 2023-10-10:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNy-v5jGjMW7Y9pgv6lZ3ltVQV4zT3XpMQ0W5JJq2k4xGCGYW5SlQBx4bk3tgVgNL0k3xfhFxW6xr29k9c2l7SN3YvdV3qTg8PW2bpdFd72kpXrW3GLR9B94DJGZW12ksMw6LB9ldW9gqHYY3BJc3GW8_BVZ86brrbNW2yyV_x7w_yN4W63Whqs4ngN1bW5zXjxc4GSqt3W3BqXsz6M7vPJW6rjKQR33lJYxN5gPtgfC730-W2_wL8p3T0ZzzW8GrFvd6TS2PcW8mD42z4Wm6d7W1GnykP5mVY76W3HCd6N9hHv-DW69vHJn1160gyN8VdvR3gpWk1W2DYsWK6JyHh7W43LybH2r2YKtW93p98J2jdR2kW3mTwz46yC_KpW4gw6Sp5WkhVkV4bYgb7nG1lSW6sdxtD8Z1fyLW2Pf8554R65cPW646sfm6h2xGTW2kzkn_1t0Ld7W2jPCJr4QCPkpW7w6pC41RLYVGW3yqch55qwhYtW57XDqR8-ms8nW4DY_4Z98TQr-W7ytdNr8b99gfW8Mn9JJ5vtCynW686RCC1-cY-Ydv6l3K04 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNy-v3l5QzW7Y8-PT6lZ3kHW8rWbZR6ZPWf5W2N-xbF29BZndW9j2rbq3wVcdBW7hMg0x8dygKYW8WX5-D8vnKSKV-cqXY7FFDq5W5QymVB8_tbWMW5kZNFq2Y7y6DW3S-jWr1tn09KW52t-5C87cnQQW1xNLc94p2S1vVRVl9g214bwlW3t0L_z67WKftW18RMx04P8d6pW1Rl3qq6ntq0XN9h86Rk-H3w1W2K9ZvR1D3DX-W3YT7Nh5j7xRmW4RXmKT3335f6W4XNC_M36JT3RVNVqlw50slT2W34HRd96yBv5XW7L8vvJ4JMMbRVd24dt3FL089W5jxY5N23yZP1Vqfv-J9ckQ-wf70D85v04 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNy-P3l5QzW8wLKSR6lZ3p8W59WP9n2d0zKhW3XZV8B3ydPWTW7G08rX9lmND-W7qZ8l-6Ps3N3W3zMT-V136V61W5sk_Yb92sgM8W475KQp4mm8zTN1hjj8TPGNzKW6STytZ5HL3y8W8ft4NN67SPRzW35tZ_5537r1KN10KVGPKn6w0W5Gh2z-4TDqLnW1nnLS31WcQqbW5cbdgf1wrwM-W35yXPW45r8W0V458v21w-Z68W9fmrxK747rtvW6tBLD71jV1qZN6F8RqzXTD9gW7Sx8jJ4Cg0h7W8J_vr-68JgGDN7SsWFx3W63wW4qF23P3T_XX5W7LC5Qs43bXzFN3Mkl-5y5pNZW8hMMlY35pRf6W8TPSWM1BCdpPf5_6KLT04 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNyZj7NSbdW5BWSxg6lZ3q5W4FBR682MGPJ8W1xgJpK6qHy40W1sftLp5Gz4yZW2JLMPt1NHfQ_W2CWCGY2NnBQFW5GpJD35qPCSDW1Hxk801Zn0JKW1LPtZg2q--zXW3N74Yx94LWDcW57mFyc4RWPKzW5hD6z71GMqZzW6BfN9d99gbgzW7554ZJ3m1L2lN4Gy1mdV0NH-W6w4PJh1hccZHW57JS291LG5wbV2X_0079WnGRW28dDbH9jj2wMW30cVB12RMkTMW6s-FWr2K7r33N1WvR3BWKk8cW8MG-Zw3QJ13SW6S6FM465CjGLW3N37-j1BStVCW55BPcF2yxntZW7XwBvd5hYv-ZW21-hyC6-v95JW4V-K298fzTsbN8qhkrt7mYpQW8bv8vJ6pxXShW67C210834_nPW4bNMNt3d-QDWW41QNcM89hbydW7pB3qL55bxW4W6ChKxM5plSdHW5JypGr2hjCLnW69mss63YSMBXVNw9gs1DgqSMW66X3Nc1JSc1_W4VK4WF5X7wDYW98k7XS4-t7v-W83Zv6p5JLCKgVKLyRh95TF1YN724vjV3XRvlW5Kls_k4T5Q26W7n42L_6-flCwVQ7rVs2D4kGcW2NYz-73ycJ_fW1WyqRj59m_PgW5xx5Js5F04d_dKr9g004 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNyZ05jGjMW50kH_H6lZ3n0W1Mz0FP5cP9lRW7DsGy09g5VmZW5NWm_43cKplQN5gBpWWS_dxwW1BTcrd5rfZcGW5-Nk8n2CZYswW8b3sX92rjXnfW1PTn9v1R0Z7xW2FstVj8Zh2-KW3jhCn715Br6HW73McpK2FsKd0W2l25jr55TSwnW2lqD9Z96PKBFW7_1FNL8J2DCtW5VTJHL4RxZHyW4F-FWW3zGY5BW7kcVmj49-TbvW8MWNzs969WkJW7xD4zR6YpJDVW6ksXgm4ZqT1dW5RJc5-47KDbkN7qxMQm78rHcW6pxfJR7_jfpTW9jLHzT5gQJymW3xwlLT8LqGq5M1r18qLp8fgW417BZ-1XDXw2W5QHm-l2VRQZrW6pvWTJ9bz54gN3KVvpJVSkCXW2tVyln3FmdTZVBqYfR5NmJPFd6Cg-W04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNy-P3l5QzW8wLKSR6lZ3nrW22VX4p2qsdv6V1vnc92lzqgpW4MsqNg9d8NJ9W2c7j845HrHNxVWmyd31JY5fDN5094mYRbfNLW6kBsJb1nH6ZtW33mlZl6JXdd4W5VhsYT6wyDmYW65MnMM91-LvRVXLr0y3jlxyCW6bDtwG1VhL7dW5qywcX19ytvCW5Skq2X3NB3GnW6ymWF66LVrGcW9dRDdK5l1XpDW2WT3Gk1XPkh0W3dqR0m6q0QS-W1VBYS-29qrMSW2GJKW_6xLhKBW2gQvJF98fX3hW8J_CP551XPX1W39xG4335P370V4VCWD7nWC6PW7dj2lG3QtDD3W6vSh5q8fzW-PW52Gc-r1bPYgDW6ND0tS7QnzQlf2qwqqx04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj0DD6BJRtMW2BftFN3PljghW8DFYQL54t6QkN1HNy-v3l5QzW7Y8-PT6lZ3pWMwCP2Mlb249W379Cxf4-Zj_RW1Nsppc5qvZdrW7P0MDc4XLknRW5ph_5C5QDrJKW8wZFZk79D9tqW7H4lkN52PPQJW5dtsdd234wJgW2RR1N191s0D7N3NC3LXD6hGVW89b9Gq4Pn7fBW3_dRh84ngZlSW7s4BK_4ccFBFV2ZVP-7tm79jVwrZrP1fsqk2W5zMDn078P-ZJW3tm9ZX9hYqTqW8bSMbj8P2QT3W2XC8Zd1hBBy1VxD_jk8DXqhlW5Mv3w-86P_GHW1WMLMH6Xx8xYW3NKLJS8Gcr3xW3sDhC_7ZdbklW3scyNc6-y0nnW1_qRhQ8T9Mcpf674gjs04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hTQZW3CdGGD3JJqwYW4hJT_H3M9qFQW3K7ZWT3KbsvFW3R0gVJ41WVxzW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8GrsHXW96DSgV5n2MZnW57SNK-90HZR8V28Rfs930jtwMGKd_P20N8NVM3R1M2m_M2cW8hbPh45N3sY03hyz1&v=3&_hsenc=p2ANqtz-9Sw5z4XH_WXrmnlfbdYSsS6QLt_vIN7E0gTyoZaeQkN2MSWvV0it7FcY28nGPMNKQRMyFOhVEu5uaVGA25o7cqxOIcD3D2L5Qp0GhcG56z9u4_Nz3VsuNyWTjPU38Dv5ddYKXyDU2wdSKyLRJagDCPk7qqdA&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hTQZW3CdGGD3JJqwYW4hJT_H3M9qFQW3K7ZWT3KbsvFW3R0gVJ41WVxzW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv8GrsHXW96DSgV5n2MZnW57SNK-90HZR8V28Rfs930jtwMGKd_P20N8NVM3R1M2m_M2cW8hbPh45N3sY03hyz1&v=3&_hsenc=p2ANqtz-9Sw5z4XH_WXrmnlfbdYSsS6QLt_vIN7E0gTyoZaeQkN2MSWvV0it7FcY28nGPMNKQRMyFOhVEu5uaVGA25o7cqxOIcD3D2L5Qp0GhcG56z9u4_Nz3VsuNyWTjPU38Dv5ddYKXyDU2wdSKyLRJagDCPk7qqdA&_hsmi=86142041 )

jkaob, 2023-10-12:
Ok, thanks.
Any plan for when the ROS driver for firmware v1.10 will be released?

closed 2023-10-12 by jkaob

andrewpriddis, 2023-10-14:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VHP3l5QzW7Y8-PT6lZ3m6W7MXLyQ8j_QL8W7bG3RK1pnzFxVZRZKK7Xy3TYW88L7SQ8W3Vg5W16Gmgw7SH3DlW2P80BG6Ll3SRW8KRqHt66vgq7Vr1KGT331SrNW7r_CMm6rLPQCW7wwbrq80STxDW68cCMD2KhF1YW71Kx2F7FBNqBW5vgBPq22nnPQW7trjjD8-8pj8W97QL3w39qjP0W1cv76w7QH5SHW6QlV5S7sw_GJW2wmhMm4gsdY-W1hmWsN5jCfJdW4BYBHd3mGwkjW4hxcNd2vTRHWW2v2g_R1w3XYWW9bpvBk77mqDzVPtcWm4-Dv67W7lzXcz2DfnVyW4dLLL54YxJfJf2ZTW-804 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VGj5jGjMW50kH_H6lZ3l4W9hmrtJ4tZtTbW3zx9y_51zzb3VDKH-S5C-J9TW96SVls5JR3rzW5cSQkp6l5KWfN5-skbbX1sR4W7GsPqd7db-2fW9h1Qwh5VFHsfVNKBgS89ZtmwN1-JKLLLstLvV-74Nx5V0dNbVRwyJ8594gVCW1JskTm84M0sDW6ZwtTx7WkHK_W3MHD6_1Ym_19W4-rsKD5DMMsZW6Sqmgp36GnXLW1h5bb477gQFCW1fHQx04bRrH0W28lkYK2XC-KlW1bbNGR5MDqJxN4z80_xLWQMSW3YwxLN3vyV3BW8YsWHf2mn-cWN53NF4tXPlz_W4Gyz7S8DQkT1W1qM2Fg1D3ClBW3wldrY5T0MD0W6dlvqL24NPWVN7lJw8XPJtFtW4mbf2s33mvB2W4WDNKy5sphCSf774_Zv04 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VGC7NSbdW5BWSxg6lZ3pQW8WmLHP6KkScHW64n8T47QZNlNW1GH0_427rTcBN7QqVrjzJX6cW3CM1gl3hgbLbW4VPt_n52Nn93W56D__06nXsxFW8Zcl7s5W-mXkW6wDDgd4jrN5WW1FP1Fg7Sk2_ZN5YXzhC9-GTvW60pwMH4KSQHWW5XfqTb7LmqYlN6wTM00ybcRtW8RgNbC2Tyc3KW6_DWVD1NT44sW2W__dh3shKk_W6MgNkv7xdS1BN35rh6cj-kXnW6nfQC35Sd16WN4bKQBpGGG-xW3r9FGf3wCN7YW6pKNwz6dN5GdW6kkdYw4ZfQw8W5-v84T6DHL0zW95xwdM8yYKd2W349R2k9d5h_MW69PL6J7jpWTkVpTx9795F80HM5x0s4-vqR_W6hhBZg8RnkYnW5vcFBz3n-d3FVxvzlK2DtK_2W5HHfDD4zz_9dW6BlH-285kyg1W2Lm2_z58G8tYV7BZRD7cXrmPW4LyG9Y3shWYdW2mXFd57vKMk2W1YPTHS7QncWDW8zj33H3N01CXVrF_353YPy7TW7NkZWH9dV_KPW8XZlvX6gCtyxW5FFlCL67r597W4ZX-Vc2vlm0zW72RXw51mM7BpW3NNP5q4pKzF0W93C5376k9fHzVj1FJS4J-xR_f81q54n04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VGj5jGjMW50kH_H6lZ3p6W56Wh-08lwGNSW8qBN-g8tB_QRW6k077g3YJlM4W6dYJLY81WPphW7ZBTKx7FW0_4W2ZQwYf4r2XCVW27Fwms1g8mQcW66STRQ3qTVMvW8KvTSK14mGrQW8Dytmx7vxGPfW8cw_TM2sLV2sW1_Smqx3YsFBGW367n4Z3jrcPzW6d2JrV17Gkj4N6llzBVt1HWbW27FLT_5FhxmsW1zbbwD4nxhglW2_Wtc71ggj1fVSxb2q1PJsXTW4bPNmN2s6PkpW1xh9HR5nCVTRW5P5y7s78R8kBN2tX8hgl408YW7s6WZf8vNzq0VK3HlC6Gpq27W2CfH9Z4tqqsmF8kbYx94LVSW3TSJQZ5s1wlrW23mKY97pPMjnW3kxFhj6GyfpPW212PKj7ckGyxF6m0QxfVH3yf8PNY9v04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VJ43l5QzW8wLKSR6lZ3pDW5JZkNm4XNmb2W11xNGN31GdL3N4yMCC4dFThHV1JLVp1T1w27W2xSQWD1qvqKDW92bSzV89GwxkW3kVVy04WqxBGW7FrVx87DVv7HN1933dD4Mn4ZW48YR193GR7rSVjjDK14_qKHfW3Kybv_3dXZwwW6DDG055frTpSW54-7Qq4k7MSMN1rB8Vq3j4PFVBDTGH542fSMW5C6Pj29hlRccW8s__sx93pPWqW119lC27w9TqcV83zyk1r6dV1W4t5GY96lKxdPW3H9HNz6rQCb4W2g_xFk1RgQBZW2QB2Hk6JcZClW6Wz68H4NwyTxW1J9k6F4qRRW7W1Wqbmh6kTXJGN7_0VXp8xXQnf6srlKd04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX6QHK5VxNqVW6lMGYc7MG-3bW4qM5Xr54DJNqN3X6VHP3l5QzW7Y8-PT6lZ3lNW9jW-Ll4j36TpW6XPQ707cT8NyN9cMfrgmzckSW2PzlDl5WWWXyW730vSD5JDfBYMTtFPLcmPxtW4_lSGb4bJ4QgW1c63DH8QcplmW2WNHpb5m_VdbN33nCY_dhpJqW1m48L498P7JfW3BTnY15XDJsFN7L_1th_f0vgN6Wt68j2Mjy6W1VWTNG8YdlcfW5bQJ2j4jygkyW7wnj5s7wGCZdW5khBgJ4YBGk1N7_vsfbsg5JkW3vZkQ_1Wh4fyW1fkfjq7QxQgPW2-TsR53zrtFMW4575gx4JkbzfV_F1jT7CqyKgW6FSWzT2cJRSmW1XPY3X3Y42KQdlJ5Xz04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hTQZW3CdGGD3JJqwYW4hJT_H3M9qFQW3K7ZWT3KbsvFW3R0gVJ41WVxzW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv96Sv__W78HSfM6G5VzpW5fQ9FK7NmG_mV28XCd85_w1LMGKd_P20N88VvNZS74DMPjqW7D8wll7Hm8yd4cphl1&v=3&_hsenc=p2ANqtz-9IlFsFggeVT3muZRFeO2piDt73poyfIqBGQEHCtqmBIwDbxVuUjdtx2wtjFZX9bmhieN-ZsvPcwhX9hw47QhGZlpU2HUg5DNn8OUSrlcKSljboglc1GPeBiB7a-hFJ14LxY3fYtw8SoawBmztf7YWJ7U0njQ&_hsmi=86142139 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21hTQZW3CdGGD3JJqwYW4hJT_H3M9qFQW3K7ZWT3KbsvFW3R0gVJ41WVxzW47szjy45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv96Sv__W78HSfM6G5VzpW5fQ9FK7NmG_mV28XCd85_w1LMGKd_P20N88VvNZS74DMPjqW7D8wll7Hm8yd4cphl1&v=3&_hsenc=p2ANqtz-9IlFsFggeVT3muZRFeO2piDt73poyfIqBGQEHCtqmBIwDbxVuUjdtx2wtjFZX9bmhieN-ZsvPcwhX9hw47QhGZlpU2HUg5DNn8OUSrlcKSljboglc1GPeBiB7a-hFJ14LxY3fYtw8SoawBmztf7YWJ7U0njQ&_hsmi=86142139 )

andrewpriddis, 2024-04-30:
Upgrade to the GPX-1 for improved GPS accuracy and robust solutions!

Hello  

I'd love for you to take a second and check out the newly released GPX-1. It is a dual GNSS designed module for GPS positioning, GNSS compassing and RTK positioning in a single package. You will no longer need to fit two large receivers in your design. It is an L1/L5 multiband receiver to reject interference, improve fix times, and provide an overall robust solution. The introductory price of an OEM module is $325 with volume pricing available as well. The IG-2 is an integrated module with an IMX-5 and GPX-1 with a price of $788. The RUG4 is in final development and will be released later this year.

GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVTLSv31XXRlW8D27kY7vHRSvW65wf3Z5dxhLHN1MKFZH5jGjMW5BWr2F6lZ3l1W71l_KQ3vll-tW6Bgb6x2tXZVTW24-GX12zj8_6W4xn0NT5PfHjsW447BkZ14KlzjVQwn-65pVFfNW8Bk1kK4cBw91W4zMfv686LbRhM9ZZPpqLVT4N2R4P26bZ047N9b2Nr-QNv7KW7mxD4M2rTpMCVQkpgk3pRLz-W8Bnp5Y906-QLW3MjV6t8NvqjRW2kxrLC3PKgRMW1f92BX8v8B6wN72zXDFzlqr9VQ2hd58qnzPmVvm9T54b-wF0W4wfV5B4MVXzNW90dfSs67mWXPW6_YwVC59rskWW8Jj-8w6qvMXVW8hXT041B-y2LW7Djzs46n21WtW6d6Rfh3JFx2NW8PFR5t1srKnjW8qdBGj3f2xXyW88CDYV1WXcqvW5WQBNp20QQhWW63mbt92vnhnfW8bBSlG6XXrGPW2tFRgc3fjgf_f4gxMcg04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVTLSv31XXRlW8D27kY7vHRSvW65wf3Z5dxhLHN1MKFZH5jGjMW5BWr2F6lZ3l1W71l_KQ3vll-tW6Bgb6x2tXZVTW24-GX12zj8_6W4xn0NT5PfHjsW447BkZ14KlzjVQwn-65pVFfNW8Bk1kK4cBw91W4zMfv686LbRhM9ZZPpqLVT4N2R4P26bZ047N9b2Nr-QNv7KW7mxD4M2rTpMCVQkpgk3pRLz-W8Bnp5Y906-QLW3MjV6t8NvqjRW2kxrLC3PKgRMW1f92BX8v8B6wN72zXDFzlqr9VQ2hd58qnzPmVvm9T54b-wF0W4wfV5B4MVXzNW90dfSs67mWXPW6_YwVC59rskWW8Jj-8w6qvMXVW8hXT041B-y2LW7Djzs46n21WtW6d6Rfh3JFx2NW8PFR5t1srKnjW8qdBGj3f2xXyW88CDYV1WXcqvW5WQBNp20QQhWW63mbt92vnhnfW8bBSlG6XXrGPW2tFRgc3fjgf_f4gxMcg04 )

When paired with the IMX-5 you have a low cost tactical grade full Inertial Navigation System (INS). The solution can be further enhanced by enabling RTK positioning for cm level accuracy as well as enabling GNSS compassing for high accuracy heading. The IMX-5 is $275 with volume pricing available as well.

IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVTLSv31XXRlW8D27kY7vHRSvW65wf3Z5dxhLHN1MKFZn5jGjMW50kH_H6lZ3mmW8qP1n94DYtCNW8PRcB35lC0YZN7RhX26FjXjPW7mnZ9_5dPTw9W5l9n3y3wDcmFW6BrR078MXNDmVNV-sb8dkC8PW7Y2x845QkmbNW2hB5YM2lks0WN9hCKfgYSG3cW8D3hHl2plsdpMSbk62Pzv1FN6H1KnMrtl2DW8yM4j91XYc9mW3zBg9-5M2d02W3zkX1934pHZQW2DfbPD8r47LvW7tNsT823lByxN52Km2j2_7PjV3P_cB2bHNp_W14LdyH7K99qTW4SnK6L651HFLW7MNbXb7skkYTW9fFv0s1YjX8yN8z3rF78PrC5W7yJKWt38m-MnW6Sh8kr1qJ3xkW1lBFy42n2M1_W5tYd4w7WWC4TW8NBGG95FV7zZW9kq5jy4rPYlMW6M7CwV4HV9nGf2S9s_b04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVTLSv31XXRlW8D27kY7vHRSvW65wf3Z5dxhLHN1MKFZn5jGjMW50kH_H6lZ3mmW8qP1n94DYtCNW8PRcB35lC0YZN7RhX26FjXjPW7mnZ9_5dPTw9W5l9n3y3wDcmFW6BrR078MXNDmVNV-sb8dkC8PW7Y2x845QkmbNW2hB5YM2lks0WN9hCKfgYSG3cW8D3hHl2plsdpMSbk62Pzv1FN6H1KnMrtl2DW8yM4j91XYc9mW3zBg9-5M2d02W3zkX1934pHZQW2DfbPD8r47LvW7tNsT823lByxN52Km2j2_7PjV3P_cB2bHNp_W14LdyH7K99qTW4SnK6L651HFLW7MNbXb7skkYTW9fFv0s1YjX8yN8z3rF78PrC5W7yJKWt38m-MnW6Sh8kr1qJ3xkW1lBFy42n2M1_W5tYd4w7WWC4TW8NBGG95FV7zZW9kq5jy4rPYlMW6M7CwV4HV9nGf2S9s_b04 )

We also still offer our ublox F9P based products that are integrated in systems all over the world.

I'd love to set up a meeting for this week. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVTLSv31XXRlW8D27kY7vHRSvW65wf3Z5dxhLHN1MKF-T3l5QzW7Y8-PT6lZ3lDVtr4Hq3R6c8kW1fHbQw29Q370W4-g0yl2HT4ycW8-MwjX2Yws6cVnnD2D41KcXBW6Ghhzx6z7mRFVDryls8b6ZG9W5QhS882GFs84W54Yt7K8sRmmNW8cjh-l5JSgmmW1nq44r1tf8RtW1sQtJz2VtLgPN7XCGP2g19ZDW503yLN4H-vF_N7lMHKVy-p9mW6cXj9B1P4BRZW96bN9Y7Tb8cYW3vyzBb6kR0dJW1xlQR83DXkCjW1dlwDW3wMGTjW3bLZth7wnm1mW80JJdK71ZXHgW4HkXVF5srFj5W7tCXQ28DTjyYW4h84DW7P-zF_W12bCnb49j3Fxf8F2pR-04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-H6W3ghN4Y3z1H1cW3H0cwx3H89d8W2HrjD93NDDB3W3P2-md2r8Z2lW1VrJ6G2vyJfQW3_tnl42Kzxt5W2Yt67P2vwdZpW4cHcLX4tFmvXW4mmY5F1LCql1W2r41Fk3g7qmrW4kmNrf2p6g5XW2TKWMj23lwbXW49rXJH385yqNW2KDWn-45pSfyW30LhMT43PTNJW1Qnjbd4mzK9QW4kBpTf3VGZJtW2PWrtl3XPl51W3ZsJ_s4tFnDZW3QVbxQ2-d47WW2TNlSX4mGpmqW3_R61R4t8vcJW2x_BJb34B051W1_4RVH2MwVQ2W1-YvJm3LHjQPW1Lhl542zWqY6W3D_GMz2Pmhj3W3GVwPR3F4yz6W2CN-pd3dfXxFW3VSzh83-1TPpW22W0-x4hdwWVW2YFhXs1_r1VJW2FVDb12-rk2TW1Q0MHN38frMXW1LDdP31NhbKNW2qS-824ft4KKW3C8_Yq3d2Kv7W49S7Fz1X2g5HW2TQr391_ml8_W1_dcFN43qnw3W1Nly252FpmsJW4mnM0K3y-KkkW2sFqyf2Rj93ZW3gwYtd3_Z7hRW47LTdP4t591cW2xtGZ71V3h4gW4mjcC81NvPXNW2nGLZT3jv0CKW3LSsgw4pkFz-W2MCVwy2p1HDmW2YPHtR4rwvjQW1BqT-H4prT6Vf4fwbcQ04&_hsenc=p2ANqtz-8WF4wVrKtDfucg2No7oyc8KaprHHOzTZ7ry3WXss6P5p63BV_5S9pV2q6BUcp1PdGD_BLCwhRk0f-EF6KZBkUEMn6VyXTSeKtrWCN4Kagf7OM4jtur8Y4wGS7xkGZxKJw51Fc58yZ-TpiJbfRUoXHp7Y-Znw&_hsmi=305054157 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-H6W3ghN4Y3z1H1cW3H0cwx3H89d8W2HrjD93NDDB3W3P2-md2r8Z2lW1VrJ6G2vyJfQW3_tnl42Kzxt5W2Yt67P2vwdZpW4cHcLX4tFmvXW4mmY5F1LCql1W2r41Fk3g7qmrW4kmNrf2p6g5XW2TKWMj23lwbXW49rXJH385yqNW2KDWn-45pSfyW30LhMT43PTNJW1Qnjbd4mzK9QW4kBpTf3VGZJtW2PWrtl3XPl51W3ZsJ_s4tFnDZW3QVbxQ2-d47WW2TNlSX4mGpmqW3_R61R4t8vcJW2x_BJb34B051W1_4RVH2MwVQ2W1-YvJm3LHjQPW1Lhl542zWqY6W3D_GMz2Pmhj3W3GVwPR3F4yz6W2CN-pd3d [... cut ...] xFW3VSzh83-1TPpW22W0-x4hdwWVW2YFhXs1_r1VJW2FVDb12-rk2TW1Q0MHN38frMXW1LDdP31NhbKNW2qS-824ft4KKW3C8_Yq3d2Kv7W49S7Fz1X2g5HW2TQr391_ml8_W1_dcFN43qnw3W1Nly252FpmsJW4mnM0K3y-KkkW2sFqyf2Rj93ZW3gwYtd3_Z7hRW47LTdP4t591cW2xtGZ71V3h4gW4mjcC81NvPXNW2nGLZT3jv0CKW3LSsgw4pkFz-W2MCVwy2p1HDmW2YPHtR4rwvjQW1BqT-H4prT6Vf4fwbcQ04&_hsenc=p2ANqtz-8WF4wVrKtDfucg2No7oyc8KaprHHOzTZ7ry3WXss6P5p63BV_5S9pV2q6BUcp1PdGD_BLCwhRk0f-EF6KZBkUEMn6VyXTSeKtrWCN4Kagf7OM4jtur8Y4wGS7xkGZxKJw51Fc58yZ-TpiJbfRUoXHp7Y-Znw&_hsmi=305054157 )

andrewpriddis, 2025-01-02:
IK-1: For rapid evaluation and development.

 

We heard your feedback that evaluating and developing using an IMX-5 or GPX-1 module can be tedious. The parts are designed to be surface mounted for deep integration into a system. This makes accessing pins for evaluation and prototyping difficult. To address this concern we developed the IK-1. It has the following features to ease the initial interfacing:

- 0.1" pitch header for convenient interface.
- Interfaces with standard breadboard.
- Onboard Micro USB connector
- Onboard voltage regulation.
- Dual U.FL connectors for GPX GNSS antennas

There is a version for working with the IMX-5 as well as a version to work with the GPX-1

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsFl3l5QzW8wLKSR6lZ3n4W80hq-L6fMPGHW16xKNr7w5LdDW8-lRGD4qjWQRW7q3k3x2Wl3llW6NCSF33zRw63W2gjZ1S1lzYrfN3MZy4_VGyNcW6nVK8D9cbBLlVKXFxG4cGwDXW35NrSP3NJTmsW8WrLKK72m7kVW2HWVKs856C-4VY5cF11l60MXW6HyQsw1gDPDpW28k5v370TdVVW3l086r3tDNClN8dhKWfpKjs-W28d_S07ZbN0_N2TRwBvKFRWWW3k1MV177xKpFV94ZgV8K0cDqVGQGPd6FrfH-W6sQNGb4mwbxqW4kg6vD5xfVKPW6m_sZj28G7KgW8HmXYF7tC_4rW8fp8ST59xxJPVZPfmW7gHpnlf879lNb04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsFl3l5QzW8wLKSR6lZ3n4W80hq-L6fMPGHW16xKNr7w5LdDW8-lRGD4qjWQRW7q3k3x2Wl3llW6NCSF33zRw63W2gjZ1S1lzYrfN3MZy4_VGyNcW6nVK8D9cbBLlVKXFxG4cGwDXW35NrSP3NJTmsW8WrLKK72m7kVW2HWVKs856C-4VY5cF11l60MXW6HyQsw1gDPDpW28k5v370TdVVW3l086r3tDNClN8dhKWfpKjs-W28d_S07ZbN0_N2TRwBvKFRWWW3k1MV177xKpFV94ZgV8K0cDqVGQGPd6FrfH-W6sQNGb4mwbxqW4kg6vD5xfVKPW6m_sZj28G7KgW8HmXYF7tC_4rW8fp8ST59xxJPVZPfmW7gHpnlf879lNb04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsFl3l5QzW8wLKSR6lZ3nCW5rkHwY9jbLgwV19L9m1mrG0qW2WRjq61jyGwZW5H29w43gKbw1W6qNH9v3G_3_hN3DsjZ2SpKf_W5V1d8L8ghwvZW3GdFN-3sRNzKW6kBkTK1lsT5gW3mNddC7239p1W6j0rlT4NMyFFW201Fxr4X-0RvW6WdpK5111Rv1Vnh2_D139PccW5RBZPw7_jYWWW1hGcs84kWXrmW8rd7t94GZbY4W8ffByZ8PYdKMW39gpxL1V2dpJW7Jz2FL3P3wRVVPHDWf26KFGMW3TT7g87JD3tCW3CPWSx8dk8CDW8g9r6t1HHS5PVlZZn98fbZXwW4L8lkv8JSD6BVK1mgm2gJ6mcW8ZXd7S926jRQf6FGKjC04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsFl3l5QzW8wLKSR6lZ3nCW5rkHwY9jbLgwV19L9m1mrG0qW2WRjq61jyGwZW5H29w43gKbw1W6qNH9v3G_3_hN3DsjZ2SpKf_W5V1d8L8ghwvZW3GdFN-3sRNzKW6kBkTK1lsT5gW3mNddC7239p1W6j0rlT4NMyFFW201Fxr4X-0RvW6WdpK5111Rv1Vnh2_D139PccW5RBZPw7_jYWWW1hGcs84kWXrmW8rd7t94GZbY4W8ffByZ8PYdKMW39gpxL1V2dpJW7Jz2FL3P3wRVVPHDWf26KFGMW3TT7g87JD3tCW3CPWSx8dk8CDW8g9r6t1HHS5PVlZZn98fbZXwW4L8lkv8JSD6BVK1mgm2gJ6mcW8ZXd7S926jRQf6FGKjC04 )

Your work with the breakout boards is your first step to integrating  both units into a low cost INS solution. Be sure to review the datasheets:

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsCz5jGjMW50kH_H6lZ3nvW5J2TT_5bz0r7W1XxJCF5_MR-FVF6TXm3njLqWW8hVw-j4P7_1bN7xBBXF516x0Vphn561cNj4rVW8H5b21DnXgVQLLqT3dzNyjW9kFCpv2ZsNCcW7s_dJG1L302gW1bgxK75pD331N5-L4mGLs6tQW1sDNZ07SJSWFW7LC7Zt1FnQ5mW3xlcMY179K1nW9cyn9r3XS-1JW96KN718c9HczW2swSYr6YYqmgW65p2tz6WFKzQN3dKDsfptSPNW6HlKsd3fwQ5cW44p0cN2mqWgDW6NkVjj6y2wVMW6gV0WT7h_RQFW2bkrbV78mFzqW3zLsZs6crY2JW7BWc-B29CSWFN6x3HydMpTmyW5mf4hv26l8kkW25H34w4-qp3FW2mkfMp71wGJYW32DdDX7B71yTf5nmbmT04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsCz5jGjMW50kH_H6lZ3nvW5J2TT_5bz0r7W1XxJCF5_MR-FVF6TXm3njLqWW8hVw-j4P7_1bN7xBBXF516x0Vphn561cNj4rVW8H5b21DnXgVQLLqT3dzNyjW9kFCpv2ZsNCcW7s_dJG1L302gW1bgxK75pD331N5-L4mGLs6tQW1sDNZ07SJSWFW7LC7Zt1FnQ5mW3xlcMY179K1nW9cyn9r3XS-1JW96KN718c9HczW2swSYr6YYqmgW65p2tz6WFKzQN3dKDsfptSPNW6HlKsd3fwQ5cW44p0cN2mqWgDW6NkVjj6y2wVMW6gV0WT7h_RQFW2bkrbV78mFzqW3zLsZs6crY2JW7BWc-B29CSWFN6x3HydMpTmyW5mf4hv26l8kkW25H34w4-qp3FW2mkfMp71wGJYW32DdDX7B71yTf5nmbmT04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsFl3l5QzW8wLKSR6lZ3nCW5rkHwY9jbLgwV19L9m1mrG0qW2WRjq61jyGwZW5H29w43gKbw1W6qNH9v3G_3_hN3DsjZ2SpKf_W5V1d8L8ghwvZW3GdFN-3sRNzKW6kBkTK1lsT5gW3mNddC7239p1W6j0rlT4NMyFFW201Fxr4X-0RvW6WdpK5111Rv1Vnh2_D139PccW5RBZPw7_jYWWW1hGcs84kWXrmW8rd7t94GZbY4W8ffByZ8PYdKMW39gpxL1V2dpJW7Jz2FL3P3wRVVPHDWf26KFGMW3TT7g87JD3tCW3CPWSx8dk8CDW8g9r6t1HHS5PVlZZn98fbZXwW4L8lkv8JSD6BVK1mgm2gJ6mcW8ZXd7S926jRQf6FGKjC04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsCT5jGjMW5BWr2F6lZ3n_W6rzSpt3TFpP9W2WvMT34RNH30W2gzHQ82DqkMyN7Mld68q3W2KVKMdYb2qtBs7W2_wmVr4Zv3RTN9lXHQnrFt2CW3QZXVD7LN-w_N5CpQRbb8DQgVjRfkc6G1ksGW3s9k2J4tvzl5W1dh-6C60t6TVW8_vkjt2fT4DdW3-kLgs6G8_W8Vk-2SF7Rl4JjW7xzY2J27fG1rW28DNX_80s6JXW5nR3LB5b7kzgW9lfHVD6D-DD_W6Gcpx57vc13sW1Bq32s6S6x78W2dKmNm6pbkXvVW8XRZ8FfgZ3W6y0kLz5Rw99gW13hqF9752xXCW1X6CXw63JG-lW6Rcxqk9l_XJYW1qM47F7KcN6kVc-FGt3NmYwlW7ppY3V6jB2zzW50j2lc7Z939yW5YGsr51mjWT5W3P1BLH4b1Pb1W1gbbdJ3HXJGCf378FG604 )

I'd love to set up a meeting in January. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

***@***.***

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVK-8L5-t6tPW1bhNt32lRkBhW3mxgk35qkpYpN8JCsF23l5QzW7Y8-PT6lZ3nCW6LRJNF8DkfwDW7zxCHH3kwr0NW2ZVJfd5W8-gLW4DrGd04vtv8wW4_XgV52MPmxgVbnD_91F8-BtN4wkWnqTVZ-PW6yvWV53Rh4_dVGXZn_8bkLF7W6MVcbC45FxhMW6CYkH_8h-vVxW2_YsB35Zl1mHW3xs_cZ8GcwwZW58q9p32zQ4J9W6RnS3l4Jgn-2W6ZnKLz8NlHVHW2 [... cut ...] 3fW3JKc5p4ms8X8W2HsYSg4fG9xSW47PTRR3VQrn3W4cqrlM4msB6kW49Q4Tq3W40-0W2TK5b12YwdblW4cq3x11Qn0GrW1VgBjX2MBFryW2r8w8z3LBNqkW23fzzK2RQdyyW2CxC1Z41n3YPW41zYXp2CFxk4W1BxFQN23b9k_W30JF1S23n8GGW3QWqjy1ZpdxlW1Qg3qm2-CT_JW1Nyk-G2HDhcSW2KDthh2Ft7WBW3gpg3V41twPyW1VtQCY1_l1NfW2nXnJq3gmYKJW2p5tQr4fHSQHW2sSlNK2CKJ2tf3bsw3h04&_hsenc=p2ANqtz-9iKq84nhqbhceUNKGpZbKb2ccKLmCgFNIU1V-UQtc0CDokVwj3laIVbiBLTegVV5OEuB4N5W04gQjtd26A9tOCXXXdFvuTUK4VEs5KK_2m84lewKEWpRxu0c2ZRBCMPvzQdMNq7wKrSmcio3DvMyzJfdkpQA&_hsmi=340288392 )

andrewpriddis, 2025-06-27:
Order now to avoid tariff-induced price hikes in July!

Hello  ,

Hope you’ve been doing well! I wanted to give you a quick heads-up — especially if you're planning any new builds or procurement before Q3.

Due to rising tariffs and component costs, Inertial Sense pricing will increase in July. But if you order before June 30, you can lock in our current 2024 pricing on key products like the IMX-5 and GPX-1.

Whether you're sourcing for:

✔️ Defense UAVs, loitering munitions, or targeting systems

✔️ SATCOM, radar, or LEO antenna platforms

✔️ Or just need rugged, U.S.-made tactical-grade INS/IMU performance —

Now is a great time to stock up or move forward on your project!

📦 Explore Our Top Products:

Have Questions? I’m Happy to Help:

📞 Call me at 801-610–6359

💬 Chat with me (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVZ_hw7sQskJW16Dtlh3wbqYKW9ccHTJ5yj-cgN5bb0jT3prCCW95jsWP6lZ3njW3cmZ1w2WJN5TVghN7W5R6VQ4W7-WTy5659_tPW6fF_-z7DsBJtN6x1DrWgmTthW1bHChR27gHq9W3NjH586JqF9VW3rnbvn7HcLGVW3LmKZz9jWSTWM6SJZtmrDkWW3SWfH73s_HFVW73BYw93lFgH7W1P7lJ03-YDM2W5NDdfZ354d66W7c-5n73hH2xQW56D19r6xWynnW5QwDNP4lMg2kW8GH3-j7wwrqdW8pLzm96_h4kxW6SLKYY79XKH4W5D6Rnp29mqTmW2t_Clk3mkg3vMXX-hbvl8CSW7fdzpc13j9ntW3c8ptg5g2BtMW2lPSmp4qrWWpW7FBTph5NtxvFW8NCS5V5cJlvQW14YNJw1KdpKKW25S0gs5n2_byf6Y1hN604 ) live on our website

👉 Or place your order directly: inertialsense.com/shop (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVZ_hw7sQskJW16Dtlh3wbqYKW9ccHTJ5yj-cgN5bb0gP5m_5PW50kH_H6lZ3pbW95-Lx76qsKHnMNqLHTPxBcDW5Q9RmR1mWZqVW4z_2Nc1md-8jN8QC1tcvPYxpN80qdCR65zbFW8TY0qG2Wg4BkW6TVbZK6pCpKfW5WgLZ75hj6DRN3-QHkJMSL6pW7V_clb8wYgq9N51ccMcHtPz2W15RGXF614GMZW97nzHY78Q-lcW7KklZf697l_SW73hRTt1FdqnXW4Yg6LK8rTP49W5zCf5S7crxLFW6pp4YG2rG6ZzW57lCFk68NXGnW8fGYPM89TmvgW6NS5S01TvmlDW1MylGT2Lc0MvW1ds1TR89kkmLW746pr75Zbv_QW41qVTf9k8TFZW2yCm591R_YNDW7WjM9V1-6SP5W9fwvRG74n9m5W3dB8nd5JxbwqW8S8dT97cPXG0W8mtY1m4lNQsbf8vVNrv04 )

📧 Or just reply to this email — I’ll get back to you personally.

Thanks again for working with us — and if there's anything I can do to support your mission, don’t hesitate to reach out.

Best regards,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVZ_hw7sQskJW16Dtlh3wbqYKW9ccHTJ5yj-cgN5bb0jT3prCCW95jsWP6lZ3nDW5K0Clw1b25qfW6RMCrb8mdcYDW5J6fV85qC8rNW4gY5QS6BF4DPW7YdX4q7DNZKzW1k3Vpc4kSn6jW22v4Bz8qpt0RW1Ns0YX2DHmJ3W63_qRK4lMTG_W2Hn90M6zWHfYW2dtM883L4dRrW48sPQx8mFxdPW5wBcBY7L7xCbVktFqd1qSVb9N5xRbtWKvcWMW4jy2TR4dTQt5W1Jwlmp856SSTW4rbydT4NTX_3W6Dm2ly1wjjXhW2n2cdP36jMJVW1H1RJl99d41LW2D9qjS2vsYZyW5ZKbLz67fmPhW1NfBx33FwwBFW1LXWxn4PvBBzW52Dl3K5Gy6lBW8Ph8GX3SK0PbW7d56Qc5Vk800W4KrX0R1w8KzhW4kfJKv3NxCqZf31xJ3v04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-zMW3g974c4rhMpjW232FCC1X7jmcW41zzLx4pgMMpW47y79W3NDbMpW25fMFC4tmMwRW3W1rVQ2p2B4GW3XHDf62w1y2hW4kKv-R3LXcV6W3__cY2365QtSW1Sr1kq47RTbWW3bfKS64hq1Y5W1N7HqC2RmwL_W4tBRVV1SpQS8W3__fdZ38fVbrW2PM_MQ1BJP0BW25gbPm3BLgLfW4p5BPd4clfbCW4ph63V2Cwp8PW3NBY0j3d7jjNW238xkg3R3cP0W1Zfhfp1QtLqCW1Bg-GC49tZTQW47R1kL43Z9RBW1L724x4ttDs6W3dhB9y3BWptNW3SRswZ2CYmcrW4kpSyz2Htjr5W3B_lDZ3h-WNgW2Wv_1-2TSvbqW2MX0MK2Rs7dRW3M0gyV4mKf-SW1BNFlf3SRYlpW30GYXw45lTCvW36tpnW2sRvklW4tt-893W2PvZW30ryqR4mk4drW4tFkB63QQTH9W2p0WBp2WzlvwW4cPq7m2YtBllW4cN6wy2qPBySW3F6jL41N4lxTW3P01yb2YDxd5W2qDQKT3CjwJXW2HD9H12RP2JTW2MX5XY1BwVy0W217WVR3VztMWW3P1K8t41VxXxW3SyymY3ZVDTGW2t2X8K4htNkdW23jQy92w1x8bW49M73q3h-7MnW2-nS8Z1ZwrWgf4cJQ8t04&_hsenc=p2ANqtz-8BKi8MZHjNZiUO-OKE92B9UftTJuwMc2Wq6fP8Gmk5UQIFuzfCkblijn9oTDFoFhu4GlYyHQLfYc3Vt87RmTLRt8uP8CigN99YTAtCqXOHJv6uqBezXfUM0DVXtp9Vc740K2le3JwocNRcj5mIKDbnZuLqBA&_hsmi=368897815 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-zMW3g974c4rhMpjW232FCC1X7jmcW41zzLx4pgMMpW47y79W3NDbMpW25fMFC4tmMwRW3W1rVQ2p2B4GW3XHDf62w1y2hW4kKv-R3LXcV6W3__cY2365QtSW1Sr1kq47RTbWW3bfKS64hq1Y5W1N7HqC2RmwL_W4tBRVV1SpQS8W3__fdZ38fVbrW2PM_MQ1BJP0BW25gbPm3BLgLfW4p5BPd4clfbCW4ph63V2Cwp8PW3NBY0j3d7jjNW238xkg3R3cP0W1Zfhfp1QtLqCW1Bg-GC49tZTQW47R1kL43Z9RBW1L724x4ttDs6W3dhB9y3BWptNW3SRswZ2CYmcrW4kpSyz2Htjr5W3B_lDZ3h-WNgW2Wv_1-2TSvbqW2MX0MK2Rs7dRW3M0gyV4mKf-SW1BNFlf3SRYlpW30GYXw45lTCvW36tpnW2sRvklW4tt-893W2PvZW30ryqR4mk4drW4tFkB63QQTH9W2p0WBp2WzlvwW4cPq7m2YtBllW4cN6wy2qPBySW3F6jL41N4lxTW3P01yb2YDxd5W2qDQKT3CjwJXW2HD9H12RP2JTW2MX5XY1BwVy0W217WVR3VztMWW3P1K8t41VxXxW3SyymY3ZVDTGW2t2X8K4htNkdW23jQy92w1x8bW49M73q3h-7MnW2-nS8Z1ZwrWgf4cJQ8t04&_hsenc=p2ANqtz-8BKi8MZHjNZiUO-OKE92B9UftTJuwMc2Wq6fP8Gmk5UQIFuzfCkblijn9oTDFoFhu4GlYyHQLfYc3Vt87RmTLRt8uP8CigN99YTAtCqXOHJv6uqBezXfUM0DVXtp9Vc740K2le3JwocNRcj5mIKDbnZuLqBA&_hsmi=368897815 )

andrewpriddis, 2025-06-30:
Lock In 2024 Pricing - Buy Now & Save!

Hello  ,

We’ve seen a spike in orders this week as more teams move to lock in 2024 pricing. Inventory for popular models like the IMX-5 and GPX-1 is moving quickly — and inventory is expected to become limited.

If you're planning a build or restock before July, I’d recommend getting your order in today!

👉 Place your order now

(https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXdFZ53jpqG_W90t78-562VPjW6L2xF55yrw_ZN7Y8_383l5QzW7Y8-PT6lZ3pqW6frJ2z8MwmPhW6LqbF_3nPlx9MtPfxBRCDLxW7X4Kw46lHHjYW8_vrQ81VyMJTVGRzn_87D168W6RnwJB5ZWbTwW45R8xF7BflCvW4-kqpW1-9VtZW1nDhR-3_d5VZW1Tl-nF1M3ZvjW7X4rvk226q2pW4c9PP740VnjDW33P75q9295bZW7wP5z32vxkqMW5s_dbN4kzrrTW7f1-134jTGBFW5Jgfln86dRM9W8NztLc3bvz52W13L3845VKSLKW7CPHB36RqgdMW4-Jnwz5H4bXdW8_c8JT3ZbrpYV2Q5_m6fjN-4W57k2rZ2vhrvfW17y3zw4-4vPFf8KG2B604 ) 📞 Need help choosing a part? Call me at 801-610–6359

💬 Or chat with me live on the site

Let me know how I can support you!

Best,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXdFZ53jpqG_W90t78-562VPjW6L2xF55yrw_ZN7Y8_383l5QzW7Y8-PT6lZ3nCVQN8l065bwwDW44kyns8-zX42W2xGFMj70J-CwW5SQNMS5B6ZZlVcj4zx71gKKmW5LJ2XB5T7MdnW29M94312HjMxW6xqBpW4Zsn_-W2jwh5s9h9Zp3W8Gs1S34HWbt2W34B8Hp6T9VMCVxlmZ77y8HbQV1-ntq8b7PZ4N6DPgMvvXpZqN891L158cdSlW1vqYbb1qNypbW8YC9gg7HXdmBN5RKNxb7TzXbW1r7PmB1GZjHQW1j_Kg08CwpgpW8rrpj23-FJ5YW2nf83t4MJnKHW3Sb6sL5rvG49W3_l59646sGWzW5dmNXx4WKW-XW7vng0m8VkSswf5kg3MH04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-M2W3z1G3l3GYxMkW25hll449C_w1W258nw82xZ7R2W2HWDZT4kls8BW45PyW02Kq_1TW3_WC3D2PlRF9W1Vnp-j3SR3bCW1X9kyN2WL5JMW4kJgqP4cSnVLW41TkHJ30KvSHW1QlfYg3ZvpX3W1By4411L5LTFW4ms-F93QKWDFW4tvm1B3dnDYzW3Ccqk11W-VNHW2MmZ1v3QQ3GmW25hlN94fpSBbW2MByCw3N_bNGW4cs5xp3X_HsVW2vNgYk3yPk3GW2CFtpj43YfCzW3X-Ntp3XMmJ6W3dfYfb2HtMnHW41Gz4c1BFxTGW43yZxP2vY359W4cNZ6Z2sB6JdW3Ccvb43dptN_W3SZBGB3zd1fFW2RV0Fz3XzjrNW3NQHRY47zM9nW3bzlHJ45My6HW1Nzc0_2FYtQnW41JDpr30JgLJW2B35GM2FpkxbW4p9pH12FM-GzW4hnjNp2xyY85W2Mmy-l1Xmn6TW43RYfX2-qWknW253P2r1Z7WyCW4hd2xl2RBDD4W2nZ8cQ1_pVWgW1N6szs45w__vW3GY73g3VNMZcW3XVQm61-_j1MW1Bc7tj3Qzz4JW41RGVz41yhsDW2Wcc5h3dfs3wW4t7d-32t0vBTW3jmkSQ32xGNBW3ZW0Ml43YKKXW3F1HTP4p8VnWW2sV0C52Tjn61f2CKCzt04&_hsenc=p2ANqtz-9fL2-mYxe5phRCVuryRQ3kcEyT7qV2DuaBF43h6lEyPGQPVW17GZSlSDjLMl0jDvTZilYQ-aUQuANjCsCbBAM1W-UWnGMMoBNpU5CNwCrLbQzKRc9iyMP8imv7roMqCJb-GgMPgdsWL7gwmD1gSuVWDDEtvw&_hsmi=369255851 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-M2W3z1G3l3GYxMkW25hll449C_w1W258nw82xZ7R2W2HWDZT4kls8BW45PyW02Kq_1TW3_WC3D2PlRF9W1Vnp-j3SR3bCW1X9kyN2WL5JMW4kJgqP4cSnVLW41TkHJ30KvSHW1QlfYg3ZvpX3W1By4411L5LTFW4ms-F93QKWDFW4tvm1B3dnDYzW3Ccqk11W-VNHW2MmZ1v3QQ3GmW25hlN94fpSBbW2MByCw3N_bNGW4cs5xp3X_HsVW2vNgYk3yPk3GW2CFtpj43YfCzW3X-Ntp3XMmJ6W3dfYfb2HtMnHW41Gz4c1BFxTGW43yZxP2vY359W4cNZ6Z2sB6JdW3Ccvb43dptN_W3SZBGB3zd1fFW2RV0Fz3XzjrNW3NQHRY47zM9nW3bzlHJ45My6HW1Nzc0_2FYtQnW41JDpr30JgLJW2B35GM2FpkxbW4p9pH12FM-GzW4hnjNp2xyY85W2Mmy-l1Xmn6TW43RYfX2-qWknW253P2r1Z7WyCW4hd2xl2RBDD4W2nZ8cQ1_pVWgW1N6szs45w__vW3GY73g3VNMZcW3XVQm61-_j1MW1Bc7tj3Qzz4JW41RGVz41yhsDW2Wcc5h3dfs3wW4t7d-32t0vBTW3jmkSQ32xGNBW3ZW0Ml43YKKXW3F1HTP4p8VnWW2sV0C52Tjn61f2CKCzt04&_hsenc=p2ANqtz-9fL2-mYxe5phRCVuryRQ3kcEyT7qV2DuaBF43h6lEyPGQPVW17GZSlSDjLMl0jDvTZilYQ-aUQuANjCsCbBAM1W-UWnGMMoBNpU5CNwCrLbQzKRc9iyMP8imv7roMqCJb-GgMPgdsWL7gwmD1gSuVWDDEtvw&_hsmi=369255851 )