inertialsense/inertial-sense-sdk #466: ROS params for enabling RTK compassing with EVB2.1 dual F9P

jkaob, 2023-10-09:
Hi, 
I have been unsuccessfully trying to figure out how to use the ROS node to enable the RTK compassing for the EVB2.1, using two antennas connected to the F9Ps. 

I am using the `example_params.launch` as a template.
I am able to get readings from both, after flashing the board and enabling the gps1 and gps2 parameters in the launch params, but I am not able to get the `RTK_Rel_callback()` to trigger, or to get anything published on `/RTK_cmp/info` or `/RTK_cmp/rel` topics.

Some of the parameters that i have tried to configure (unsuccessfully):
```
topic: "inertialsense"
port: [/dev/ttyACM0, /dev/ttyACM1]
baudrate: 921600
ref_lla: [59.921078, 10.686089, 39.49]     # (this doesnt seem to change the NED odom outputs?)
platformConfig: 11                         # PLATFORM_CFG_TYPE_EVB2_G2
ioConfig: 0x26CA060                        # "inertialSenseDocs.pdf" suggests 0x...040

gps1:
  enabled: true
  type: 'F9P'
  antenna_offset: [0.1, 0, 0]                     # X,Y,Z
  gpsTimeUserDelay: 0.0
  messages:
    gps1:
      topic: "ins/gps1/gps1"
      enable: true               # no effect ?
    pos_vel:
      topic: "ins/gps1/pos_vel"
      enable: true
      period: 1
    info:
      topic: "ins/gps1/info"           
      enable: true
      period: 5                              
    ...
    rtk_pos_rel:                       # Relative measurement between RTK positioning base and rover
      topic: "ins/gps1/rtk_rel_pos"    # no output?
      enable: true                         
      period: 1
    rel:
      enable: true             
      topic: "ins/rel"        # no effect ?
    base 
      topic: "ins/base"
      rel:
        topic: "ins/rel"       # no effect ?
      info:
        topic: "ins/info"
    compass:
      topic: "ins/compass"    # no effect ?
      enable: true
      rel:
        topic: "ins/rel"
      info:
        topic: "ins/info"

gps2:                        # pretty much same as gps1
  enabled: true
  type: 'F9P'
  ...

evb:
  cb_preset: 3    

rtk_rover:
  enabled: true
  correction_destination:             # this defines which GPSs get RTK Corrections
    gps1: positioning                 # RTK precision positioning at GPS1 (RTK corrections are sent to GPS1)
    gps2: compassing                  # Enable RTK compassing (dual GNSS moving baseline RTK) at GPS2
    gps3: none                        # Disable RTK corrections at GPS3

  positioning_enable: false           # Enable RTK precision positioning at GPS1
  compassing_enable: true           # Enable RTK compassing (dual GNSS moving baseline RTK) at GPS2
  correction_input:
    select: xbee_radio               # Select correction input settings 
    xbee_radio:
      type: "evb"
      format: RTCM3
      port: "xbee"
      baud_rate: 115200

rtk_base:
  enable: false
```

A source of confusion to me is the configuration of rtk rover. I want the GPS2 to send corrections to GPS1, which i assume must be done through Xbee? Unless you can configure which serial port to use? 

There are also alot of parameters in the yaml file that seem obsolete/contradicting of other configurations, as the comments indicate. Can't see if they actually trigger anything, because the topics are not appearing.

Am I approaching this in the correct way? All I am trying to do, is to configure the dual antennas so that they can output the heading, without having to manually compute that from the 2 incoming GNSS signals. 
Thanks for any guidance / clarifications

andrewpriddis, 2023-10-09:
I'd suggest you use our other tools to understand how the setup process
works and that you are able to successfully stream data before moving onto
the ROS integration. Are you testing outside with a good GPS signal? If
there isn't GPS/RTK data generated/received then the topic will not be
published.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

andrewpriddis, 2023-10-10:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh8l5jGjMW7Y9pgv6lZ3n3N8KCQnbXNkQYVtdmpf2kzC0ZW2k1k2r4C1Bj-W1H2B7B5vf8VKW7lrS6X8g6QFbW4Vlcxs96VqkxW8Qjzk46bLnl5Vmh3xy9dwyn_W5V-S5N5KDk2ZW4zgypY6vqvx3N8fZzD2X0f_MW8t64zz1JLnjJW4rk94c1j_KSmW31Txyc6Z-r-wW3wC1Fm2N35MgW30yV_030Yp3QW7GsGxk92LlNbW77TVKd3rQqKwW887Bpt1cpJmDW79M43v4gXsfqMlc0MBGPDqjW5mjbsM1ffK29W9bppMl7w7k44W5CzSLN5vBwStW156Hr14vMKQFW631Gzx5pd0YdW3q7Fh920fgyjW7bdb4T8ZfZ4dW8jK4W179r2mRW7-K4cQ6lr0YCW964TTv7PYyBSW3m-6tt7dcN4DW3ktVzL5TM7KyW3CXTcq4y4vSPW7xcmlH2FFCB8MlGDVPM9hp9W67Yhfz2wGKCRW6W3Q-38dMWCNW8DBcQx6__-9BW153htc2fMxBdW84MjMJ1N7MgYW9b-nk06_tqfcf7vdTN204 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh8l3l5QzW7Y8-PT6lZ3n_W7TZktp2X7Gr6VRj6Rz2HJm3_W3LQ2X43tyZRWW60skQN4q6JrDN38h6LXQrX1VW2jSRWq7KBSz9W8mNXfj2ZZ0xvVWmyN21csntQW1KX7_85xzyCHW6jYdX48j75tKW4vnvt_2W6pjKW5bq8sR6M-5VSVH749N6pglZ-W8nlnjR7WWQX9V5qSLg7MJZfgV-LDDr2nqFZhVx33Tk4Qyg8DW8qrfQk4JWkhvW35VyRS4R0gghW4LvGnd9bXwtBW8r2Pvq7V4XpPW4BKSRW2Nh9VPW8f-PR_451HqFV-yT974fpcR7W8zy6rN252PSKW2-j6hd2QhtHrf7gnJ5z04 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh8F3l5QzW8wLKSR6lZ3mCW1p4pm699T_4dW2MNX0R8FB7jcW38-qhM7D4gC_W7h4VLB13jrbVVWqGYt4wT-fJW2j2Nvm83hYKgW2VFfch1LZ6kqW6mD5yW3RkcPcW3v3w6399FgWwW6jlzvs4npjWqVF9CHT5cTtHkVn5v8q3fXz9ZW3Yl0ct6s3WzHW57wzXG40qqd-W4QFNMM2TYm9mVG3k1R7f1Sm4W3v3fB289x7QJVvVxW54d9YffW17w_855cBQv9W2wl9Zn2Y1NqsN3mWQCZJK0QBW1HgZnF7-V1lNW8-C4S0550S5TW6l4qM-7LkDTvW6rf4xM1t460sW3rNB1B6Vc6-RW2rY15b5l9BNyW5MkBY792Stf6f5p-cdx04 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh787NSbdW5BWSxg6lZ3l5W95D9Wk4m_f4DW9gZfz82yr3mgVv6PSP858Vd2W19pcYv7T2S8wW6Hfz3P1YsSx5W4TNVGY6gZ1CDW1-M3Wj5_P5q5W1--23y8swFlqW4kKq7p7CNrW4W99cf6h9b0-ypW4sl4jd6hqWmmW52NZnC2d-J64W1TZkZt5sBZPYW2nTNbF7jvjgSW93Nnhv4kVgWzW58JC7R73zVn3W6HGv7f4cP6kWW921b2q2L2D9nW11ChFT4YYWPYW5X0p4-4lVnMXW475R4f5XkQwTN3JlWP0QrVJbV612Fp2n7YXqW8YVwJY6xlgsSW6GG5J87tjXGPW9kk28l9fnYNYW3kXKR67FDlmdW5D_CCF47_m68W4bvLPt39NQ87W91ThSr1Sp5K6W9gP8Zb7BS2KNW4QzL7X2KGhgCW1g3R2-1Bh0xKW8854P43bYB9LW8Pl03v5cLgxrW8GQxpP90KcC6W3GSH-Q54_rfkW8PTClz51MXmbW8ZBJlC17FPt1W4DLtPf4Pcy2SV4kW152YcFcmW7Z4TSW4xhmNGW5qd5P86sPvH3N4GqQ1nkCWHZW2CZwb44BWgnqW6szyq26j7303W3hyls96fB74yW2R_JSh2Xx8R0W1l2vkv4bxhsrN6JB650Jh4vZf59YjMW04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh6T5jGjMW50kH_H6lZ3kxN63LssxxTkslW2Qn_CN6l_SZdW90pYSr3lxw6yW2tDyPV26_d2SW7Y6n4P7Gx13zW84Qnpb3zxN1SW1RrYsr3FQ9-_W4P11SD1QnvLZW82d0nb45BrjQW7tS2HL5sNYv9VlJ2s677mCMfW4kZpSM5LL3-QVGX68c4sGWplW6F1yDX5N70W0W4C6xqP1RVVd-W3ScDmC4rg_GCW4G2G1G6lWlRPW4xTFCw2NQnf3VzrGnY5vdVzWW1RFNkV2hr3vTW6kzJwR44jjPNW5Z_wBq24c7lHW7YfF6d7yyq-1W3L9yxy9fZBvVW6cW5WR5VK9BlW4ZcyQ21fR7zvVND1DR7CCQD2VsX_lh5FQmS8N4lnYp7kqDkjW7qgP_27vfSFBW8ZJxNb3Qyf2RW3p6RZg7Kt1ytf6b_TWj04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh8F3l5QzW8wLKSR6lZ3plW943JBP12H-NfW1wxNx92DTTwfN27Hk_Q7fC91W1c2hn_1k_mDnW2TyRXl8nZCGDW4Ydy3F2MWKD_W42W4JN2nnR0FW8fzvyT1z2WqFV83R5T5W5DqZW6sXhj91X2RQFW58T-GC4j88RWW8dsTqs8V2p6wW21SYlG4WPnz5W7RwX894rhZDyW1llZtv5sdSY9N5Rs-k5ltvG-W4YWhKK7JqckbW2m55Bn4JhhStW87hRBf68b2WQW8gyky01cBtPMN51XrQwV3-v2W3mCcd11r-QBbW2XwygC80Sr4fW5msrCq5wctpyW4Q-nWM8swwC7W1Jc3vd2DhRZlW59zWn02dfwTGW7Dzznk1JRTQFf3QGzWj04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVCpn53xkcjGW1NKH--5HYRq_W56dvtZ54t6Z1N8Zqh8l3l5QzW7Y8-PT6lZ3kLW1JRnMY3PjQ9QW5vWvKc6qHWZKVVdBCm5D2DgYW3Nmc0C6VhlDhW3X6ks55PBCQ9W3Qwv5l11x2sHW1v7pH-31Fv11N1dy_C14JMpPW1BJndT19-0PfW6RpxP84zTfS2W1fc-DJ804yV-W7Mt5fk55sxKxW12QTV71wz7dvW89dcS88hDthzW8xqzlT7N9xkfW82_DGD7Q2x0FW1k8C_55D6QrRN6K7WnGYhQFZW23-7p-6DKQbwVPnlTk3mPs1yW6TbFsJ12S2nxN8tr-5S2c27vW4303368HjM75N4h0Dtk_BgWHW6ZPpJ98DKjpcW23s7sh7KKCzWf7XfgMW04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21jHl9W4rxkKR1_rybRW3_Rwtp3ChdF3W3M6rbR3KbsvFW3R0gVJ45Tp0JW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz57Dp91W6P8F-d4MwtnRW5cpbVS6jyFfDVXfm667SCH2KMXk_7Qbwc-_W8r-sbQ5LTxqMW2qvkvd6KZPzw31HN2&v=3&_hsenc=p2ANqtz-9rtpP6vvdKI07yC6i55373cHcEEV3mOphVxE30d0e08pi7pSk43Jhm_U9uSPGob_Vw2njorVn6ibNgCC_mLUpt8yzcSX1hnmTrsQtZ4JCvmSDWzOclyuqm8v1T9R7l7wvf8_5MJ9cFZBRiFMjCaiaJEXWwOg&_hsmi=86142041 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnzSyB6WzZmnW2p76LK3-1rCSW3X_fKX21jHl9W4rxkKR1_rybRW3_Rwtp3ChdF3W3M6rbR3KbsvFW3R0gVJ45Tp0JW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W15gBJz57Dp91W6P8F-d4MwtnRW5cpbVS6jyFfDVXfm667SCH2KMXk_7Qbwc-_W8r-sbQ5LTxqMW2qvkvd6KZPzw31HN2&v=3&_hsenc=p2ANqtz-9rtpP6vvdKI07yC6i55373cHcEEV3mOphVxE30d0e08pi7pSk43Jhm_U9uSPGob_Vw2njorVn6ibNgCC_mLUpt8yzcSX1hnmTrsQtZ4JCvmSDWzOclyuqm8v1T9R7l7wvf8_5MJ9cFZBRiFMjCaiaJEXWwOg&_hsmi=86142041 )

jkaob, 2023-10-13:
Yes, I am  able to stream data using the CLtool and EvalTool.

On inspection of the code, I can see that the master branch does not have any way of enabling the RTK compassing through the params only, but that the source code must be modified and functionality added

andrewpriddis, 2023-10-14:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412xbl3l5QzW7Y8-PT6lZ3kNW2RWcLR6fBzXTVH87br7X2xsgW8tDFbH279VH6W4-LqXs13rBWdN3mTnmZ4kdhZW1pKC4x926YTHW1ph67s26KN73W6Cp7JZ1jyTkXW5-vyC73JLPZcW8nQQrg66Ft5tW1VCDjN1L21QlW5cVzG48mNR-JW6YHrJ-2VBcdMW5fwlZw8_dfJlW5cCHFX65h3s6W6JQghB3V-6mJW3W8V2C5ztkFDW3v0dzT11Zt2RW6lJ3CH3KFZNTW2t0VQg9hr-NVW7l24q96klZgtW1V8-2p8QtyFhVFt9xX97LkQcW4XVGQs1CRCn9W8QxYt13X73tGW1LtkF-8Rzc38f4qS7bj04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412x8T5jGjMW50kH_H6lZ3q6W5PHGcG6WZz1MW4-TDFw5bM3rgW1xHZ8R6b6RWrW4HwtX499zKYgW881Fh52cThc2W57h1hw1Ggpf0W42PT824TXSCCW5QJCtR8mN8pRW5qXVP250y5_rW80C38V2XrJVZW5MxpBT543m9lW6-0jkn3r9xy0N3Njsns9JGJqW3x9mDT8S9GbQW4B1RdC2C4tQGW5h-hrj13znt4W1FdlR09bVzXwW6W_w4N7t7ydCW763bCs2VdcW8W53B9b21GnZNzW8NM4H63MBxlwW8Bq5H-4Lry1HW2fZDWM6Gx7XvN1NYFSrmtp6QW6r5VYq7_WlDZW8vJwmN2L_5-JW99tDpz1KHfgmW6p3Kt_7KgXWWW20YWZQ1_5WQKVSfL1Q5pcpGVW1XKJTQ35G5JnW6t1MNs7qDb-HdtKkhl04 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412x987NSbdW5BWSxg6lZ3ppW7lp8m67VDcgwW6WwVtb68W1tXVCwTYY4nhdFyW229QmV4MlLP8W1lN44m3f14WPW6fg4gg79Y3x9W46zSK25cDG57VyKwSW64N3ZGW6bXc-Y13SFCkN2czYPJMdS-cW58zQTK5CqXFsN8BQJTbz6_bHW5MvGLs3vM4qBW7bBZF98VJ2jVW46_4Hq8s_WzPW4hgHrd3TZ5StW4zXD6m5mrpdDW7_w6Hq3jhRY_VmRySq9hz7qvN2DyxRBfn1f6Vmwk0H7WLzbYW1pyVwW83pMBFW1xVBrB4jw9bQW5vfs3b3f44YkW4wlqXP8yr5HQW46DfR96r2ZyXW6Q6H504xHG3_W8htMKX6xlLJPF1rpcmcN7n2W8DdzYB25XtxYW98Vv_S5GVJBxW1z2PlY46m79MMSP88N5BVdHW1FMSL02XmJYcW8RwB3J7Zf2p2W3LVsN728n7nwN4RryJqCM4cmW3M333P2Chff-W1LMX081RSKRhW7WyYdb7mrBtYW44s54056mnp7W1DW74185SVf-W7xQN0v8p7w2fW1VVN__2pbcjDW6mYdzJ10lRr1VxtsCG8SS2-cW4gbPsj3-_jc2W4MVHwG2wfZrGW5mC8Xs89C-_yW1kQFj-2j1j0Jf1Z0zw604 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412x8T5jGjMW50kH_H6lZ3lHW5qfFc-9hQs09MV7PcKzt8qmW2v37wg7cyV_mW10yb0y2rKjtrF94bFxBdVMVN8VnCtnT7S50W4sKG_q65CwTYW3N22Ct1VTrHMW4bWlRd8n1RH3W5K_rj095ntt4N2fwWVYCyKv0W4qTkJp8gJ2LgW2ftnfh50_nkkW4_B9cF60JFwXW99jftb6l5xKnW5-lrXL1YtGm4Vp7x3d5JZl88W2FH5WN2X5wP8W8zMhQQ28T3XTW8Z8cQx9c6FgpW3kv0v_6T-DSvW7zs_g23g0cbtW3_QY_N23ZvV_W3ySF3L1NvXkLN7g3ZnD8fk5bW14g08H72CHkYN4r3nx3BbX6jW1vLcnz4myNfqW7CSql5197H9_W33W6jk5T9h25N8njbTh9njVKW82Vbqc6YpKbnf7m9d7T04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412xbF3l5QzW8wLKSR6lZ3mBW458zTn50MN96W2HZWx-54_lmzW1kH6Pp41X_m_W5JWSPF7DWjzMW7GLZtN8V64hFW6rpqZJ436BXnVnfzQ73YvtHKW5ZzqdW3xnyHXW8j9vNJ3rd_pmN1_-Y9SFylqTW4gJ_6-5G2pLgW10ttvF5zg4tlW5wF_9j5KF_ZbW50_MVr4V64dcW58h0yS6q5nk3W73yZny7x8j9bW2yvxxb3N4jzjW6HtHV553Vsm7W70cJJ15vmYlgW2dKhpJ80BS43W3njmsV5fSDXbW75M_WP2X232vW39jfH54w5HnzW8y8G4g2YwKD3VKgHY75QQfSNW86BfqY8Q9ngnVxChBd69bjnFW4CW4hg2s183qdF6X--04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VX4KRd2vFZ3HW4tQL8Q4yHCGbW7CpFKy54DJXbN412xbl3l5QzW7Y8-PT6lZ3mhW89x6Bs3DzBvWW96Fh8l4pFWpXVgSzx46Hdhf_W4q0wDg2HDsq-W2MRv3V61Mzn8W3y1g4z6M6LYTN5cctxQnDSjjVgBdQ35S0X4kW8DmRl47LSVBVVPw1dL8NVRc7W6wkhDy4C7TDbN7wmcYMK0fgFW4dprG142dbd2W8jSyZ79f8qZTW3Sn-BT49x0FvW289KFT4Mq_pRW5QFFwm4N_sqdW8XsdRr4B0L-BW7zmM3N2MfH9KVw_3cw1KVx-kV6cx9n73VkPvW5cNmrq94jT0TW6CYF_p7tmm_LW20zfjP2nyJDRW6ndPq34G1ddxW1NKh5731kBKGdwdQXg04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/manage-preferences/unsubscribe-all?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21jHl9W4rxkKR1_rybRW3_Rwtp3ChdF3W3M6rbR3KbsvFW3R0gVJ45Tp0JW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv7DRYyMW7lt41s58ctZdW4TcRbh4_FcddV28XSq8mQkz1MGKd_P20N7fW6ZWn_k7L8p0dW2xD0G98gnty151QDpq1&v=3&_hsenc=p2ANqtz-8Z4Z6rElRDCH12UVp7YZlzKSMG-5-YU6t-1lVoBJi5cvXjVU3zRTa5Z9TQY209f-1xlooI9tyc_LhU8M3NrXoPkzhPiBYGdpf0C3soBG3dBTFzfmA2Lhamncow4LO1BnHojHfz2aPIulMPOt7cvSZcZxaGIQ&_hsmi=86142139 )

Manage preferences (https://info.inertialsense.com/hs/manage-preferences/unsubscribe?languagePreference=en&d=VnBg1c6WzZmnW2p76LK3-1rCSW3X_fKX21jHl9W4rxkKR1_rybRW3_Rwtp3ChdF3W3M6rbR3KbsvFW3R0gVJ45Tp0JW3_znhQ45RlhdW3P28X24hCPvhW43T2P-7DvMj2W17f7hv7DRYyMW7lt41s58ctZdW4TcRbh4_FcddV28XSq8mQkz1MGKd_P20N7fW6ZWn_k7L8p0dW2xD0G98gnty151QDpq1&v=3&_hsenc=p2ANqtz-8Z4Z6rElRDCH12UVp7YZlzKSMG-5-YU6t-1lVoBJi5cvXjVU3zRTa5Z9TQY209f-1xlooI9tyc_LhU8M3NrXoPkzhPiBYGdpf0C3soBG3dBTFzfmA2Lhamncow4LO1BnHojHfz2aPIulMPOt7cvSZcZxaGIQ&_hsmi=86142139 )

andrewpriddis, 2024-04-30:
Upgrade to the GPX-1 for improved GPS accuracy and robust solutions!

Hello  

I'd love for you to take a second and check out the newly released GPX-1. It is a dual GNSS designed module for GPS positioning, GNSS compassing and RTK positioning in a single package. You will no longer need to fit two large receivers in your design. It is an L1/L5 multiband receiver to reject interference, improve fix times, and provide an overall robust solution. The introductory price of an OEM module is $325 with volume pricing available as well. The IG-2 is an integrated module with an IMX-5 and GPX-1 with a price of $788. The RUG4 is in final development and will be released later this year.

GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MW5SVXFVtDbW8xRDR-1XRYKCW58c1_75dxhN3N2XT73C5jGjMW5BWr2F6lZ3lzW1yRc-f22VybBW2n4RMc1DDbrzW3NcWc55bVsZ1W77zKJ56WNDqMW5jnFpg7LRZflW5yyPd797DpC6N2Z_vtqM6YsTMzN5JQ5Bq5PW4MgY_P4J_KQvW8gn_dr1C90yZW49NZ4m3kzLHgN2qBq6GVrLzhW6YxqYb3zZX29W5CCbrK44NyM_W2Pq7P552tpflW2f_Mpj48_6YcW2q4yYK2x3kN-W2mt8fF3PhC9gW2CXfKz6bG346W6SY8sJ2YYm_TW9kLygJ286ZX1W35GpB44G69N_W75shZM17MM8xW3Tnr9n2_tLNJW4WQWyM60Zh1wW8gv1yj4rBX4vW4_YpYP85zYwmVZsdLS9l_tLHVlc8mQ3FYkJ3W51Y9z22HyNh0W216CyC246WMdW4fdl6Z52fkbsW4r54cz8-TRfQW2C8V1Z5Bh3fnf65jdhq04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MW5SVXFVtDbW8xRDR-1XRYKCW58c1_75dxhN3N2XT73C5jGjMW5BWr2F6lZ3lzW1yRc-f22VybBW2n4RMc1DDbrzW3NcWc55bVsZ1W77zKJ56WNDqMW5jnFpg7LRZflW5yyPd797DpC6N2Z_vtqM6YsTMzN5JQ5Bq5PW4MgY_P4J_KQvW8gn_dr1C90yZW49NZ4m3kzLHgN2qBq6GVrLzhW6YxqYb3zZX29W5CCbrK44NyM_W2Pq7P552tpflW2f_Mpj48_6YcW2q4yYK2x3kN-W2mt8fF3PhC9gW2CXfKz6bG346W6SY8sJ2YYm_TW9kLygJ286ZX1W35GpB44G69N_W75shZM17MM8xW3Tnr9n2_tLNJW4WQWyM60Zh1wW8gv1yj4rBX4vW4_YpYP85zYwmVZsdLS9l_tLHVlc8mQ3FYkJ3W51Y9z22HyNh0W216CyC246WMdW4fdl6Z52fkbsW4r54cz8-TRfQW2C8V1Z5Bh3fnf65jdhq04 )

When paired with the IMX-5 you have a low cost tactical grade full Inertial Navigation System (INS). The solution can be further enhanced by enabling RTK positioning for cm level accuracy as well as enabling GNSS compassing for high accuracy heading. The IMX-5 is $275 with volume pricing available as well.

IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MW5SVXFVtDbW8xRDR-1XRYKCW58c1_75dxhN3N2XT73j5jGjMW50kH_H6lZ3kzW3k83kb8H91FLW8gq7DH7Tv2SpVGJqlF1DlW1mW4q6lBq3CJh8SW6PgC8b3K1LY2W9g8bZs2bQ4jNW9k7G_r4NTD8xW8ZYlRn18w8WTW99zpcN1gG6LdW3Y4ZgV8JCMpNW48tCd12QGZKTW7pDRh04JPBX_W5zJ-cg9dTP78N3zsr9YJHrkGW3RHhMP8s4VPHVm3S808FY3wvW4tfS5s5RJs4QW7-g5pr54jghTM8mSSkw_W0mW8DqQpx8fG7V7W5_SsZD4K_HsKW93LqLr6HKT4xW2MX-P63y6fZTW69jN1Y307zRkW4dHPdN2ds8bsW6RGq3Z5GYGMRW7BNlQP5fDPx5W5nCCxh65rR_8W6HDM7r99B5BsN1CVJsF3RgJmW5drKKx2lmBhmW6Rgcnq8XmX9LdTms-004 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MW5SVXFVtDbW8xRDR-1XRYKCW58c1_75dxhN3N2XT73j5jGjMW50kH_H6lZ3kzW3k83kb8H91FLW8gq7DH7Tv2SpVGJqlF1DlW1mW4q6lBq3CJh8SW6PgC8b3K1LY2W9g8bZs2bQ4jNW9k7G_r4NTD8xW8ZYlRn18w8WTW99zpcN1gG6LdW3Y4ZgV8JCMpNW48tCd12QGZKTW7pDRh04JPBX_W5zJ-cg9dTP78N3zsr9YJHrkGW3RHhMP8s4VPHVm3S808FY3wvW4tfS5s5RJs4QW7-g5pr54jghTM8mSSkw_W0mW8DqQpx8fG7V7W5_SsZD4K_HsKW93LqLr6HKT4xW2MX-P63y6fZTW69jN1Y307zRkW4dHPdN2ds8bsW6RGq3Z5GYGMRW7BNlQP5fDPx5W5nCCxh65rR_8W6HDM7r99B5BsN1CVJsF3RgJmW5drKKx2lmBhmW6Rgcnq8XmX9LdTms-004 )

We also still offer our ublox F9P based products that are integrated in systems all over the world.

I'd love to set up a meeting for this week. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/MW5SVXFVtDbW8xRDR-1XRYKCW58c1_75dxhN3N2XT74P3l5QzW7Y8-PT6lZ3nTW4NGktx8y2zShW1SLv776ns3s0W4t1sCy1nwN5DW6PMZ6_5LMdyhVv1DkG4LFjbLVpmLTq2HFjL7N2yvNYDRYtXYW2byVwS3sGG99W47dM-K6RDplkW8vTHqp4TjpmPW7ngNMt3SxQ6QW1m18pz7rt9H0W2l0zYs3zBYCWW20zSJs260MQlW6GhqbK4q7h9xW3LbZG92BLdzNW6HFL6P65bgCKW6bKZGn6t0p-xVGc1r37v-0X3W7jq3-795Y3p3W8S1GTf5dHYxwW1sM1mv4-HbXYW5SjXSY2CVhVmW8DHyhL2w3clzN8JvZ61_RZDcW11mDhf7Hq56ydYRKx004 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-F_W2YgkMT1_slFhW3y_d7Z4cyXj7W253P2v3_YKsPW2MznX33SLRjvW3M9kFt3P5WpbW2-8hkw3Y0rbgW1QhS6z3JJSJfW3BM3Gn3JSdf-W1_8VY136gvpbW3R66d63NQ_8tW2zxtFB2HDJ3pW3QK2gy2vKqr2W2WH8CX36yWhVW3QQ0Bs2zNn02W2CHBKP345zqsW2FFF0h4tp2HlW2-qWcq41FJl-W43PqVC4hwtNsW3VZJV_47BFt7W250VVf2RBd5TW3GLcJY43RX1PW20XRdC4tBq32W4pr3z043V2j9W34HPym3jf7FGW41SV0C3Y25C-W4pqGCw3zkWlYW2-F68Y3b1fY0W3M89zy2WNBBkW45H0mq2MWJ4KW2MHzrM3dqLmtW2CT8sd3K2xvyW2FHKmZ4mDfvjW34f3kC3b3MNcW32h0ST3NT-VYW41TR5k2RHyyvW3BJyX61LdT1vW4cHbqN2qPx5yW3NFSwL2FK2JQW3zfQp71BKhBwW2xLX4W218hZrW1QlJbN2WNcRrW3LWT_Y3FbVT7W2MNBWv2-CVPNW47p9K33_V1mXW1Y-mYz1Q4yvPW2MCpJ13g1D_JW3byYWj45Y59pW3NWF3B23h-3GW2x_DrY366cVmW4fnw2y25myWLW4pCCDm21kW14W2zKqkg3j5-XYf3LVy9c04&_hsenc=p2ANqtz-9uuJk-NtNcP867R-ipk-oLGM64rkLAQ_o2fDDZZKYzlwegPwakuPrJgP0lfGikKyLsLuMi7CsOUF0EbA01Sn_DRy9UOTsLrSJHym6AH8FG5jNB2ocS2YU5xlp6JI3yjMmRufYTFVmf1ehyINPTJe3Es996dQ&_hsmi=305054157 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-F_W2YgkMT1_slFhW3y_d7Z4cyXj7W253P2v3_YKsPW2MznX33SLRjvW3M9kFt3P5WpbW2-8hkw3Y0rbgW1QhS6z3JJSJfW3BM3Gn3JSdf-W1_8VY136gvpbW3R66d63NQ_8tW2zxtFB2HDJ3pW3QK2gy2vKqr2W2WH8CX36yWhVW3QQ0Bs2zNn02W2CHBKP345zqsW2FFF0h4tp2HlW2-qWcq41FJl-W43PqVC4hwtNsW3VZJV_47BFt7W250VVf2RBd5TW3GLcJY43RX1PW20XRdC4tBq32W4pr3z043V2j9W34HPym3jf7FGW41SV0C3Y25C-W4pqGCw3zkWlYW2-F68Y3b1fY0W3M89zy2WNBBkW45H0 [... cut ...] 4KW2MHzrM3dqLmtW2CT8sd3K2xvyW2FHKmZ4mDfvjW34f3kC3b3MNcW32h0ST3NT-VYW41TR5k2RHyyvW3BJyX61LdT1vW4cHbqN2qPx5yW3NFSwL2FK2JQW3zfQp71BKhBwW2xLX4W218hZrW1QlJbN2WNcRrW3LWT_Y3FbVT7W2MNBWv2-CVPNW47p9K33_V1mXW1Y-mYz1Q4yvPW2MCpJ13g1D_JW3byYWj45Y59pW3NWF3B23h-3GW2x_DrY366cVmW4fnw2y25myWLW4pCCDm21kW14W2zKqkg3j5-XYf3LVy9c04&_hsenc=p2ANqtz-9uuJk-NtNcP867R-ipk-oLGM64rkLAQ_o2fDDZZKYzlwegPwakuPrJgP0lfGikKyLsLuMi7CsOUF0EbA01Sn_DRy9UOTsLrSJHym6AH8FG5jNB2ocS2YU5xlp6JI3yjMmRufYTFVmf1ehyINPTJe3Es996dQ&_hsmi=305054157 )

closed 2024-12-11 by andrewpriddis

andrewpriddis, 2025-01-02:
IK-1: For rapid evaluation and development.

 

We heard your feedback that evaluating and developing using an IMX-5 or GPX-1 module can be tedious. The parts are designed to be surface mounted for deep integration into a system. This makes accessing pins for evaluation and prototyping difficult. To address this concern we developed the IK-1. It has the following features to ease the initial interfacing:

- 0.1" pitch header for convenient interface.
- Interfaces with standard breadboard.
- Onboard Micro USB connector
- Onboard voltage regulation.
- Dual U.FL connectors for GPX GNSS antennas

There is a version for working with the IMX-5 as well as a version to work with the GPX-1

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZH3l5QzW8wLKSR6lZ3lcW48Kkzh7vG43lN6GYsGsVps0KW2N-RRf68TPLVW2Yv24l5RF_XjW4l0dhP4YGzyXN8B26NwD3CRVW5qWWzD6WJKNhW4fJCvR5bWnSvW1T6xGN6CFPpcW2KJlwl3tt69rW8l75qY8cZ_4jW72xwth6QhSd2W18rSfw2kdY18W2YsFq51dm714N40dt2Vy91HyN1P-gH1FTG_HW91dGQp98Q4CQVpMWlT10-4ZKW8H3HN37LYDZ3W6Nz68q4s5yVPW7_cZyc6KH1p-W7658fy8dKP6YW8dcXY74TpCt_N8TtXHxcsB_kW2vl4v38_PqJZW6p4mF65z_7xfW20Ld8J10cvmFW8xZz3V5J7X27f2Lw_GY04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZH3l5QzW8wLKSR6lZ3lcW48Kkzh7vG43lN6GYsGsVps0KW2N-RRf68TPLVW2Yv24l5RF_XjW4l0dhP4YGzyXN8B26NwD3CRVW5qWWzD6WJKNhW4fJCvR5bWnSvW1T6xGN6CFPpcW2KJlwl3tt69rW8l75qY8cZ_4jW72xwth6QhSd2W18rSfw2kdY18W2YsFq51dm714N40dt2Vy91HyN1P-gH1FTG_HW91dGQp98Q4CQVpMWlT10-4ZKW8H3HN37LYDZ3W6Nz68q4s5yVPW7_cZyc6KH1p-W7658fy8dKP6YW8dcXY74TpCt_N8TtXHxcsB_kW2vl4v38_PqJZW6p4mF65z_7xfW20Ld8J10cvmFW8xZz3V5J7X27f2Lw_GY04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZH3l5QzW8wLKSR6lZ3lpW9lVjh65B9bdgW1x1WCY5mGXM0W1zKx-c5PPzDLW3DlRFv5lYQYgW4P3zvH3hFZbBV-Y8dt6PplnRW5MtzvN66WTS-W3WDBF47Yk10SW1DT3NB5Jl6ywW8CQT4X5nzt6jW895X4W1GsHvfW6cZ75M1433FwW7b_PF_1dfr7RN3fKhfM_TBCdW1NFpy18CjSW1W2TFvX-1CXVFMW6yv4m276Bxl7VF_LNM22nGp3VSVhzq45Wk2MW5ltFmZ3cTqV7W7x_q8V9f6c93W8kG8Xg7Hd-ZrW8_WKny5X6ysDN3gkSc5T9FyMW6tymzP2W_7RgW3SMMLQ98C5zFN5KbvVXrqNsTW1SgmhR66QFNHf5dGwSg04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZH3l5QzW8wLKSR6lZ3lpW9lVjh65B9bdgW1x1WCY5mGXM0W1zKx-c5PPzDLW3DlRFv5lYQYgW4P3zvH3hFZbBV-Y8dt6PplnRW5MtzvN66WTS-W3WDBF47Yk10SW1DT3NB5Jl6ywW8CQT4X5nzt6jW895X4W1GsHvfW6cZ75M1433FwW7b_PF_1dfr7RN3fKhfM_TBCdW1NFpy18CjSW1W2TFvX-1CXVFMW6yv4m276Bxl7VF_LNM22nGp3VSVhzq45Wk2MW5ltFmZ3cTqV7W7x_q8V9f6c93W8kG8Xg7Hd-ZrW8_WKny5X6ysDN3gkSc5T9FyMW6tymzP2W_7RgW3SMMLQ98C5zFN5KbvVXrqNsTW1SgmhR66QFNHf5dGwSg04 )

Your work with the breakout boards is your first step to integrating  both units into a low cost INS solution. Be sure to review the datasheets:

IK-1-IMX-5: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgXW5jGjMW50kH_H6lZ3lWW4y9PDG5kYX29W7yH7sx8GrP_yW3tMH1q4Z2yGDW54SvyJ6JKvtsN2tWynPtBcWcW1DVlZT8kkQwlW1j4Gdy2YH55SW4SnmHl5Hh4DbW2-PHgR4Vt-LBW1cPNS-7XxJV_W27YHX54cK5rZW96JyGV2bDxVnN2W61Q6g7ntKM2g0hZVzLJJW3ppqvT59YzQHF4lqgrlND66W1nr58Y8gBtQFW8lvvFx6QxRPFW6-WfGv2J4ZQ3W6G9GxZ4QzFp2W79SYdV90HQ8lW8RfCM1469Dz1W1NNjnf1WxpXrW80G5N64Rbs01VtmMLK655mvTVynF5b4Fyh91W27FFGl6b83-mW2M7rnL8sy4P_W56d0mF3zwT11W3bf4pG5DLxkDW7-CCbS16ZGg9W6BzPtZ8D4ZdFf8pC_9q04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgXW5jGjMW50kH_H6lZ3lWW4y9PDG5kYX29W7yH7sx8GrP_yW3tMH1q4Z2yGDW54SvyJ6JKvtsN2tWynPtBcWcW1DVlZT8kkQwlW1j4Gdy2YH55SW4SnmHl5Hh4DbW2-PHgR4Vt-LBW1cPNS-7XxJV_W27YHX54cK5rZW96JyGV2bDxVnN2W61Q6g7ntKM2g0hZVzLJJW3ppqvT59YzQHF4lqgrlND66W1nr58Y8gBtQFW8lvvFx6QxRPFW6-WfGv2J4ZQ3W6G9GxZ4QzFp2W79SYdV90HQ8lW8RfCM1469Dz1W1NNjnf1WxpXrW80G5N64Rbs01VtmMLK655mvTVynF5b4Fyh91W27FFGl6b83-mW2M7rnL8sy4P_W56d0mF3zwT11W3bf4pG5DLxkDW7-CCbS16ZGg9W6BzPtZ8D4ZdFf8pC_9q04 )

IK-1-GPX-1: https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZH3l5QzW8wLKSR6lZ3lpW9lVjh65B9bdgW1x1WCY5mGXM0W1zKx-c5PPzDLW3DlRFv5lYQYgW4P3zvH3hFZbBV-Y8dt6PplnRW5MtzvN66WTS-W3WDBF47Yk10SW1DT3NB5Jl6ywW8CQT4X5nzt6jW895X4W1GsHvfW6cZ75M1433FwW7b_PF_1dfr7RN3fKhfM_TBCdW1NFpy18CjSW1W2TFvX-1CXVFMW6yv4m276Bxl7VF_LNM22nGp3VSVhzq45Wk2MW5ltFmZ3cTqV7W7x_q8V9f6c93W8kG8Xg7Hd-ZrW8_WKny5X6ysDN3gkSc5T9FyMW6tymzP2W_7RgW3SMMLQ98C5zFN5KbvVXrqNsTW1SgmhR66QFNHf5dGwSg04 (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgYb5jGjMW5BWr2F6lZ3mWN8__8TQn5xLVN7sQG8YZ88grW7r7V8Y6CpgJ6W6zR2Dx2sJn7dW514zyJ83HVS4W5L4fT988CZJ0W2R61wN2ncDbvV6kqh2614rLHW8yYd_48rmz67VjGs7M2rrCqBW2T2_Ph9kK2sLW3fFT0L5FCqQFVJpNJK7fyX32N6pZ59vzjQhjW2H0MnM6PT1qWW5BpPwT3fhGHcW1gpWrf1wx0-JW60Xq_H7rn_MsN6-0n356PcnTW3cD1R-2NBsCVW7xFgD519L8gDW3sZ9pF4JDwhHVstp1q4R8ZgfW7XJhyZ1p8_XLW2J_-Fx1YR_nnW2dQJZK5PCQ5RW7LWPHn57jYP6W158dNN7YkRDcW7_NKSJ6ppxSbVP4rP21d0Pw3W1t99hk4_tRJxW6_0hQy2D9V0zVPVJMJ4Q6s8VW19yG5S3hzbvhf2CgkbW04 )

I'd love to set up a meeting in January. Please shoot me back an email and let me know what times work.

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

***@***.***

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWKspg2xjVlzVRK44c42LWpvW4PtzZh5qkpZ-N1sYgZn3l5QzW7Y8-PT6lZ3nNW5_8vsH81Xc66W5czmXK5db9DmW2jL5Jb5F7DlCW8WdYNW8cz7KDW6BjzwX2QPK0VW3Lr4WH6PtknBW7kD2jp8c7GC9W4bQrDh4tDY_1W7z0_vr3MVkSrN3WW__mSs7VVW69qQRj4PwTwsW930T7d5D36n7W9d7QFW4GSbd5W5MV60m49nR7lW4yFxx_6ZMSB8W68Z4LK4KmbqGW4Wl3sp [... cut ...] CgW1Sb_nM4fvS0vW4tjpBB2Rw2RPW1BfkgF4cQhFKW383V9F2vxQpXW34pNVb3VMsy8W43RZSC32sWbvW23mQPd45WXm4W3Q_NN12nSM2pW1Lxfqz3SMlC2W2MwTd81X0zgxW3ZNnGB1_skZnW1Bnlsw25pbyHW3K5RxN3d0zdjW43WgkV3B_QRJW2r63G32RJRxwW49Bl272-d3DpW3bkWW91LwtgGW3XVQPd2Wrg4yW36m6Cs2Hst6SW3-0h-v1_f4rKW49Tn0_4hwWknW3XxHRP34n8-zW3X__p33T4b_3f3-0Fdf04&_hsenc=p2ANqtz-_ju7b1J3VkgPFN5tcLPiqlCPF2IasqGwBQkXgdM8Qd8l8fMP6Oj_wvwSzyLdN6GSeF-U_DVTcBt8L3yeLZY8oziNg0CUsYJiZDqk8mtf0W8fanENpWFDU9JjljXoQsPY4KI-RIhWqxQZiryByCKQS_QnQVDw&_hsmi=340288392 )