inertialsense/inertial-sense-sdk #489: CAN bus is not yet supported. This will be available in a future release.

matheuskbc, 2023-11-02:
Hello,

I am trying to use the Evaltool and the cltool to configure a baudrate and enable CAN messages for my RUG-3-IMX-5-G0. Every time I try to change the variable DID_CAN_CONFIG, the previous values are repopulated in the interface overwriting my changes.

I checked on the change log of the latest firmware release the following: "(IMX) CAN bus is not yet supported. This will be available in a future release."

Does that means that for the IMX-5, the CAN interface is not yet available?

Best regards

Matheus

andrewpriddis, 2023-11-02:
@matheuskbc - That is correct. We are releasing new firmware in the next week or two that adds CAN support. Watch the repository so you see it when it is released.

andrewpriddis, 2023-11-06:
CAN support was released Friday.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

matheuskbc, 2023-11-07:
Thank zou so much for the updates Andrew, I will be testing the new functionality. Thanks

matheuskbc, 2023-11-08:
Hello Andrew,

Thanks for the update. I am executing some tests with the CAN bus now. I am building a ROS driver to read from CAN bus and publish s I would like to read IMU data plus the INS orientation solution. However, I am able to send only 3 CIDs at the same time via the CAN bus. Is this a limit?

My config:

1. 250 kbps
2. RUG-3-IMX.G0
3. Messages that I want to receive:
4. CID_DUAL_PX, CID_DUAL_QY, CID_DUAL_PX, CID_DUAL_RZ and CID_INS_EULER
5. 0x5A, 0x5B, 0x5C and 0x5D as ids of the CAN messages above

Best regards 

Matheus

closed 2023-11-16 by matheuskbc

andrewpriddis, 2023-11-17:
Were you able to configure everything to get the data you need?

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

matheuskbc, 2023-11-24:
Hello Andrew,

I could configure most part of what I wanted. Out of 4 messages that I need to send via CAN bus, I am able just to send 3. Whenever I enable more than 3 CIDs (e.g.: 5) I am able to read via the CAN bus interface only the latest 3 activated messages. I need to read linear accelerations, angular speeds and the absolute heading orientation out of a IMX-5-RUG-3.G0, the can protocol allows me to read each axis acceleration and angular speed in one message each, that means if I wanted all axis I already reach the limit of 3 messages. I would need to send at the leasst 4 messages, 1 containing the absolute heading and the other 3 the axis acc and angular speed.

Do you know whether I am missing some parameter configuration so I can send more than 3 messages at the same time?

Best regards

Matheus

andrewpriddis, 2023-11-27:
We're working on increasing the buffer size to support more CIDs, but
currently you are limited by the base data rates, CAN baud rate, and CID
period. Increase the base rates or CID periods and you'll be able to
receive more data. The future behaviour using larger buffers will ensure
faster message generation.

[image: Inertial Sense] <http://www.inertialsense.com/>
*Andrew Priddis*
*Sr Systems Engineer*
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

matheuskbc, 2023-11-29:
Hello Andrew,

Thanks for the response. I will check that and come back to you.

Best regards

Matheus