inertialsense/inertial-sense-sdk #495: navUpdatePeriodMs is not persistent

matheuskbc, 2023-11-09:
Hello,

I am trying to set the **navUpdatePeriodMs** in DID_SYS_PARAMS via EvalTool and check the effects reading the frequency which I receive CAN messages (CID_DUAL_PX has its multiplier set to 1 at a CAN baudrate of 250 kpbs).

The default configuration is 16 ms, when I set to 8 ms I receive the messages at every 8ms. However, when I power cycle the IMX-5, this configuration is lost and it becomes again 16 ms. I tried to use the **Save persistent** in the **Data Logs** tab option but without success.

So basically it works at expected but only during run time, when I cycle the power in the unit the **navUpdatePeriodMs** is not persistent.

My configuration:
1. IMU: **IMX-5-RUG-3.G0**
1. DID_SYS_PARAMS: **navOutputPeriodMs**: 8, **navUpdatePeriodMs**: 16
2. DID_FLASH_CONFIG: **startupNavDtMs**: 8
3. DID_CAN_CONFIG: **can_period_mult[CID_DUAL_PX]**: 1, **can_period_mult[CID_DUAL_QY]**: 1, **can_period_mult[CID_DUAL_RZ]**: 1, **can_baudrate_kbps**: 250

My goal is to read the CAN messages at 125 Hz, but I am only able to read at 62.5 Hz. I manage to reduce the **navOutputPeriodMs** to a minimum of 7 ms and the **navUpdatePeriodMs** to 14, but the interface seems to now allows values lesser than these.

Could you help me understand that behavior? 

PS: When I use the ROS driver via USB I am able to read the messages at the desired rate of 125 Hz, but I think the ROS driver changes the variables in run time, whereas using a CAN only solution I can not really change during run-time.

Best regards

Matheus

waltjohnson, 2023-11-09:
Use the DID_FLASH_CONFIG.startupNavMs parameter for that.  It is persistent and is what sets the navUpdatePeriodMs on boot up.

matheuskbc, 2023-11-10:
Hello,

thank you so much for the support. The default value of **DID_FLASH_CONFIG.startupNavMs** is 8ms (at least when I checked the Evaltool for the first time), and with that configuration **navUpdatePeriodMs** automatically sets iteself to 16 ms when I power the unit.

I tried to reduce the **DID_FLASH_CONFIG.startupNavMs** to 4 ms, so the **navUpdatePeriodMs** would be 8 ms (I notice that **navUpdatePeriodMs** is twice the **DID_FLASH_CONFIG.startupNavMs**). But the minimum that I could set is 7 ms. The interface does not allow me to set the **DID_FLASH_CONFIG.startupNavMs** to a value lesser than 7 ms. 

Is that a expected behaviour? Is there a hard minimum limit of 7 ms in DID_FLASH_CONFIG.startupNavMs ?

[image]

[image]

waltjohnson, 2023-11-14:
Yes, that is correct.  On the IMX-5 the minimum EKF output period set via DID_FLASH_CONFIG.startupNavMs is limited based on which features are enabled:  - INS (w/ GPS): 142Hz (7ms)  - AHRS (no GPS): 200Hz (5ms)  - VRS (no GPS or magnetometer): 250Hz (4ms)The minimum EKF update period is 14ms on the IMX-5.IMU INS GNSS Configdocs.inertialsense.comWalt JohnsonCTO & Founder3000 S Sierra Vista Way Suite 2, Provo, UT ***@***.***: This transmittal is a confidential client communication or may otherwise be privileged or confidential. If you are not the intended recipient, you are hereby notified that you have received this transmittal in error and that any review, dissemination, distribution or copying of this transmittal is strictly prohibited. If you have received this communication in error, please notify me immediately at 1.801.609.3697 and immediately delete this message and all its attachments, if any.

matheuskbc, 2023-11-16:
Thank you. I could achieve 100 Hz in the output, that is enough for our application. I am only able to achieve the 125 Hz if I remove the magnetometer, but then we would break our solution.

closed 2023-11-24 by matheuskbc