inertialsense/inertial-sense-sdk #603: [IMX5] Loosing data stream in case of reconnection in linux

divinebird, 2024-04-09:
I use NVidia Orin board with Ubuntu, linux kernel is 5.10.104-tegra, aarch64 architecture.
Using default USB cable that coming with IMX5. When you connect any application from simple terminal to example application or my own, at first connection there is data stream and everything ok. After closing connection and opening it again, it is rising up without any problems but without data stream anymore... In case of sending data to IMX5 there is no reaction at all. I can resolve problem by physical reconnection of USB slot.

tmcd249, 2024-04-24:
We are looking into this issue. I have a couple of questions to help me reproduce this issue. What Inertial Sense platform are you using with the IMX (ie IG, Rugged)? What messages are you streaming and at what period? How long do you stream them before disconnecting? How long do you leave it disconnected? Are there any other steps to reproduce that you are aware of?

andrewpriddis, 2024-05-08:
> I use RUG-3-IMX-5(that coming in aluminum body and without GPS). I had
> idea to collect maximum information with using different platforms and
> finally found that problem happens only when you use USB hub. In case of
> using direct connection there are no problems at all. In case of using USB
> hub at the middle first connection rising good. And no meter how long you
> will keep this connection, no meter how long you will be disconnected - it
> is enough close serial communication and connect again. You will have
> resized connection but without communication at all. Situation is the same
> in case of using Jetson and x86.
> I have other devices on STM32 uC with the same behavior. So, may be it is
> a problem of the hub... But not only that kind of hub that I using for test
> - I see such behavior on different kind of hubs.

Hello,
We have indeed seen issues with certain hubs. I can't really give you more
direction than that. We are able to enumerate 40 devices at a time on a
variety of systems reliably. What types of hubs did you use?

[image: Inertial Sense]
<https://cq5Rp04.na1.hs-sales-engage.com/Ctc/RF+23284/cq5Rp04/Jks2-6qcW69sMD-6lZ3kRW2cJKSk7V97_3N6jLd-5jvzlFW8WyCXW5SWddFVkQx_c80rTyKV8MR4j8qwvhlVL_dct6xZGwcW2cCzdX5TkXYPW27KvHh6_D_4CW65p6m-4Lvb2CW73vRd98Skfx0W5mSRbX7CVmBcW7BdvFd3X86bDW1Wz8jB7fh0HvW5gTJxB8bZZW1W7jrfnB1nkNqsW7vqsPP7gLp6DN1YLCJ5DgWJ4W3HDTwx1bRtYxW6m0FXk3_N8jSN4mq19CGFsd1f37ybfl04>
*Andrew Priddis*
*Senior Systems Engineer*
*Director of Sales*
801.610.6359 (Mobile)
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

divinebird, 2024-05-09:
As I understand the main chip of the hub is Genesys Logic GL3523.

<details>
<summary>Here is lsusb output</summary>
<pre>
Bus 002 Device 003: ID 05e3:0620 Genesys Logic, Inc. USB3.2 Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0620
  bcdDevice           93.07
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0

Bus 001 Device 003: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0610 4-port hub
  bcdDevice           93.07
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Single TT
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
</pre>
</details>

andrewpriddis, 2024-05-09:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBBC5jGjMW7Y9pgv6lZ3psW415Bjr86mB6nW1dT7lW4HGHdQW3vbPvq293KPhW7dRy052bpJJbW6VGmv91d5BXjW3rHvk-4hzd_FW1vWvBb5Kc-HHW8TQBrh8KWT83W3kZdg86MxRbrW8FQ8Pt3fbydSW3cTNYm674ll6W7YF4fv6vGlMGW9l8TMh4yXXThW62rNVr92Rr5vN7kBrLq2rhlXW9g7MJH5x4CvcW6m_1m_2-JfYQW5fhLTf7c0-13W1tCqqr5cLQvGW504zDw3LmHtjW4vZxtV6Jz9SdW8_nsHz7BBLHFVSmJT88t5NcbW9k-0Tn2G5nblW7Lln_537CHw0W6TRVnv2LWlqDW26H5dp3gKZJyVNnNnj4mynSgW54d0MY107yc5W5CvSqZ3R1ds0W43422j22LKrNW2qbVxv3xT328W17sr4r4-F3_fW7pQDvC6_HP58N7TQMLC6rdSlW1jmFX56PM3RVW4wfl_D9fDd2FN8z5CG130y2mW3PYsN44kgDFyW1N9-xr7RKmNhW5Mr5zc7JWYKsVq0tRk1K6rgSf8rsY3K04 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBBC3l5QzW7Y8-PT6lZ3lVW6TCWYz8vF4sSW88PpSt8LFbLrN93StXFJ-PDVN1xp3c5yXXHFW2CTj3c4y163_N2hq9nd4TzS2W4tZqCn6tgmVqW6PY-QY5V5GczW3xdLLM6MyKQHW8lHsJm2LGZ_FW55xH344C4c4NN8bJNg2jp7KPVmly4484G3p2W41YGlr6hxYGGW952Pdk7pl9HWW40SkDL1xRsx2W7bMyWK5cmKwFVmBFNf3JSq9sV8Xy8N69r7bHW13pScn8-cdjdW5GS6WD314WXYW6Q1vVt6yS07PW4Rk8kT4V14n8W40WxfT1Z-dZrW3TQ0Fq9fd-LzVPWtKb5bR06Jf7QFXFW04 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBBW3l5QzW8wLKSR6lZ3llN8fsTP4fHJvMW69Fv4M3_gY3TN9dGnfWJjsYfW8DYtd_1pN2nQN5FZs1qBcvywW2b49Wr1L8vvwW2l7v6X7VcCD6VmMgXK9gn0GJW7dybsr6XdBWtN6PQMhl1w9HQW5_BkGM56b4cKW8BPRfC7DPR_2W991-lR5tVzxYW1NLnp_79zcQSW6w803n8rjgv9Vh1wTL4hWM2zN3Nm6ks6G9bSN6Rtk_DNk6hZW3l9rRy7RwkWPW64-s95779QPSVRVMRv89f2mzW33xDmH70h0vnW8Kpvmr8mWrc0W3VdbY_407h-JW78h4j21qSG3vW4wxWXQ7RhpHrW5cvGzK8j6nHnW2DDdxR3VQM1hf6cl6xC04 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBzq7NSbdW5BWSxg6lZ3kVN2zcJHBpv_BNN1lWbHmMtPc5W4c7d163LdcQsW2XtGgL9hFGfyN9h5blD1WTRrW41kwmB8sqS_rW5FCP1V4Bp6xcW8qGLDW7XwrM3W6R1Xrd43nJl1N8W27bGJ86D1VKtNV63Dh_2zN4vpV9Vlq1-rW8r34LQ6PqHJpW2nJYNQ7-nJy-W7qKgKH7xdTJfW8Z0T9W7GLqTYW4ShvcL7mr_R3W6hjnYs34sn6CW6_QdFx13_5pNW8ksf-691QFh1W5xbJDX3nP8SKW7VT2sR467wpHW2xDvNM1pY0ShW8RtKvq8_94Z4W8SsyPf4cf-MNF2gq7QFGs-2W391tJH4m8dbbN3BCQbhN69J3W2N4ZNb2p7np3W6hP1qP37PfGTW4x0xXD6SFpqyVksTDD64_CYBW3wQ44H68nhQsW2PrkVv6WXnhnVBfr-T8BRQ8fW4PrVFj5K-9tdW5bdN6W2r9QM_MNz7GSYns3cW169Nd97PFFqhW3y4Lqq2RN_TyW1Hrxyb3KPcPgW86-q0W6RKnrtW58NgMT8w5f96W43hc9z8BxWTlW5NN6gQ2BdhDCN3lb4v5S9m7ZW89FXs72GN06qW4SWfDd4J84HRW7Td-z83BcwbcN4VTjTJs15tlf3vYs0W04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBz65jGjMW50kH_H6lZ3ksN89kXhfrj7khW4hm_l34lPjMXN815RLkRs5rpN14Yvjh1LtGjW71SsjM8h9rnRW1_DrfC2Y6cgsW2WcFp82n-jPQW6chzdN3NDgDXN5nTT3lXSh95M_FzKXMZHR3W5RCC41791hlxW7yp9dq8wCBD5W3ggn_N481ybpW19ppMS2SrSk3W2KYdC78p9VDnW3B5jSv1TjThTW9jNTSR991fy1W5HG5CN2qC9hlW7t8x062p2THDW603R6n5MY2R6W6_kgQz6c87HQV_x3zX1yt_nMW6vqcSt9cqZqFW3-H4LZ5cfjqRW1X7q4V6qBFYwW6khH4B4sB42NW3c8ClM11_bLfW7brYS15_7bZzW8T_zhz6cqhN0W7w69sF7lYDLtW84zWQQ5QR1LWW4MPRjf7Vffwvf4qbcwY04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBBW3l5QzW8wLKSR6lZ3mBW1yV5dY1NbM68W88fmMp7GlNqJW1p8XSz1Xhb1bW2wmcZG5Ls18DVQHPdC8RZ3jNW7vm3wp21ZyTQN805pTFJ2nQnW4sS9H42vVDHjVyFNFn1SYKTdW55sVlY54M-lpW4dtj8m5bJfRbW7v95Wf4vSnKRW2dCf8L3wrbkTN1fmy2J5b5wXW4rCvpy81G6SFN12S1nT7prr0W8_vZGK1mJDRMVRR_D57ck_3FW2cSMfQ4ktWq5W4G42Ll1B_KtsVR-hfn41s8DbW4S3djS5T63KzW99L-0T2SCRS-W1mfQYy1YmstjW904q-x7bMJ1nW7pPryC6JnS6xMf7NDXYjNlZW2QnZrq7QdMrrf7f547g04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVwRK_7SjSNCN8p2sf1qzDGmW50jS-p5dVz7nN2VVBBC3l5QzW7Y8-PT6lZ3nlW8MjSr43m7NQmN3k7CF_WyMMnW71sp6G43c772W348Z035zSKtfW2W9BRc562DP4W8lPwNs5D9y4HW4YP8w77gyt-2W6CC5t64BkNNlW86-FZP45tN4lW9gdy4V3H2NwbW6TkC6K3wlz24W3Zq8gQ4tXNcwVkz7gk1yyRFvW5Mt_wP8jSS-ZW8KL9Qf8kTFbPW2C23SM1sS_zcMgfdV6XMXC1W4lGgVX6gcSFTF5r801GylctVTYzTM3BcVklV1dt9v7SnX_NVNxl1w4KNFzPW4hqJww6tN2N3W30WgMP7qvxgjW703n6W4_MfhKN4SnXCgBL3_Lf6QnB9T04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-M1W45LLvD2YMZM-W3QBQww4rrf-gW4fKx2m3F1gYjW43Nc771VpdL_W2-J0JD3h_N5WW3H9Scn2FS7pVW1Z9wGg2Tx2LqW2MPMpN3jcrFyW45n_CL2YwBfSW4fwByt2-zMfsW47NB2B3djqp7W2sTXNg3DZ_jRW2WtK1m41q_llW3jx3jx2FV_2jW3P6JGd2Ysg8zW1_rvQK2Tl0gDW4hCngB1_kHgCW2sDb993VtYc6W2sF0g62zLhnzW2WPR1X21bqW2W3NLsNd3QVzMsW2Yv1wg3d8wQMW3j9M8l3NT69NW214x0n2y1krRW3VJdsN1SvrqMW230DzT2CVmw8W3_RV2B1Vyz7xW45FdC14hL1JvW4tmkJV1VvzqvW2MnhcN3_XXFbW3dyjM03SJnxtW2vLf0J4mLznVW21m9_t3g6HfnW3ZTm1V32hqlGW41LkBj25k66jW3T1LgH1XcSXhW41r28l3XQgpSW1Q5PRy3T3l9mW2KVd9H4mp8qgW4hL7Yf3Y2wRgW3ZZRh02HZZNlW2HtmBQ3F7VyhW21jJ7k3zd3-fW2p9G8h1Sw9-FW32hN352WhRKcW3JH6Rb2MPrnqW3B-y-l3QKqWVW3C6txk2zxth2W3Vz_vK47GnKGW1BCRw12F_s0NW3ZTm1l4frXBKW2FTlsW1VcfJjf3FfKJB04&_hsenc=p2ANqtz-_dV2qAtEW52DsOoYRaXe4hVr8flb6IEii5zRpjsVTzvC5s6t7Zzx4p8VcYMlWSf8mFk6TVuYx65gcYEi607HYUpZrIzYDmH2xYmKkps12SmeZ4n7TOdtKdR8Vt4vQ6UIY3No41COVn9HbDzXPF79fNFO0HkQ&_hsmi=86142041 )
[... cut ...] zqvW2MnhcN3_XXFbW3dyjM03SJnxtW2vLf0J4mLznVW21m9_t3g6HfnW3ZTm1V32hqlGW41LkBj25k66jW3T1LgH1XcSXhW41r28l3XQgpSW1Q5PRy3T3l9mW2KVd9H4mp8qgW4hL7Yf3Y2wRgW3ZZRh02HZZNlW2HtmBQ3F7VyhW21jJ7k3zd3-fW2p9G8h1Sw9-FW32hN352WhRKcW3JH6Rb2MPrnqW3B-y-l3QKqWVW3C6txk2zxth2W3Vz_vK47GnKGW1BCRw12F_s0NW3ZTm1l4frXBKW2FTlsW1VcfJjf3FfKJB04&_hsenc=p2ANqtz-_dV2qAtEW52DsOoYRaXe4hVr8flb6IEii5zRpjsVTzvC5s6t7Zzx4p8VcYMlWSf8mFk6TVuYx65gcYEi607HYUpZrIzYDmH2xYmKkps12SmeZ4n7TOdtKdR8Vt4vQ6UIY3No41COVn9HbDzXPF79fNFO0HkQ&_hsmi=86142041 )

andrewpriddis, 2024-05-09:
There isn't really much I can do. Our USB testing passes and yours seems to
in most cases.

[image: Inertial Sense]
<https://cq5Rp04.na1.hs-sales-engage.com/Ctc/RF+23284/cq5Rp04/Jks2-6qcW69sMD-6lZ3mbW5xz5YS3BQ6xmVWY-_q2KbCV4W5SmZnL6XFR9RVTYZx37xpTv7W3Sw8Sn6bLH_LVsqzCm7dNG8cW18SlGc7TSkldW3ggqzk5XT5_BW8_m7YR6d_s1wW31XPY96krhZ-W77KRRZ3_6KcGW1b4xdV3vhVCJW4Jd9QM2s1KtnW5R7j4k22pJBlW4H1Mgk52ntdYW8YVM0z5mPsY5W7qyGDc3PyCNyW1LNVhV5R4JP8W5zYbxf8gk9WVN1dxY5hwJs-zf7t12Pn04>
*Andrew Priddis*
*Senior Systems Engineer*
*Director of Sales*
801.610.6359 (Mobile)
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

andrewpriddis, 2024-05-13:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4Cz3l5QzW7Y8-PT6lZ3q3VCty3w75KdvCW1Y8TPQ6P6qw1VZGygr29Y-G0W8Txs-H3Ng72NW8Dt7w35YL8Q8VchV8J8_mlx0W2HQnVp51wwjRMBMhTZrV_CLW8f2jq65GvwltVr2PQt3-y_WKW6ksjdL10DYc7W64MXR34QsCTtW7pQspq79lN1nW3l2v1S8jNJdXW286LmZ4zj502W9jJwMz7WHLScW7n9jJf771N_0W1shgBD8BFn8FW4M9Mrc7vrMBsW89l6pN2Hpt3KN96Py0M92F-CW5zhD_B6wQZMZW7dQK_X4wqSG5W8qsBQz3HfqznW8VTdLD2mPjFpW5zC8Yx76SMM9f1TJ4qn04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4B45jGjMW50kH_H6lZ3nwW99Z3Bd7Z6DRrW8PVzny3dJ3S-W6YzX7w2fMLbSVxp9tp281tt3W8W-Y7z3Tscw1W4y0gLw9jbc74W3cq90H1JdVJNW28x9Yp4CVFkDW11kX137GY0RZW973vWn33bxQrN6Q4xxcR511wW4dZV2L7jycT2W7Hc6kj5-Zc4jW934QcZ2j_vvvW8Pkc0t4J-S4kW82Z6Hm8PcQvfW2VJljL6s1YWVW4fr0Zq2ms2b0W4rwt--4jWKBFVWpTcz4zTD9XW86M4cB6x4jWcW16GFtn2bXqjBW5kVgLZ769BlMW2h4q1v3V_QPlW5gY-G71HKTwSW5RVtmQ5pKLX3W5t98p44kkVg7W209gpJ4zGkmPW96DqnD1stG5FW12dJbs7VNHqqW5p_Kx27LydG-W192FD-7f7MWJdrY-B604 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4Bn7NSbdW5BWSxg6lZ3n-W1SRMYk7NdN6nW5PyGm78ClzCjW8CMzxQ29TbY7W52VyCD8S21YdW7GtDnC7WbmjlW6t_tfm6_6dfPW8By_lx85rDMRW4h1QMP4C5ppKW4tDp0L2b9_qxW5dGhgd8xzc_pW14B2qs8tqwzxW6tmK8y70ml3FVxbvjh7bD0myW22PfhY3b14WjW5htHWn4mMhCpW1v-3HZ6h4sb_N1hTrwfChtpvW74mT722hTqWbW64hP0B95Wx70W4dS3004ss67WW6CLChx2320KmN27tyqV6CHJQW3yYQH06hgtSQW4KZLRY8pCPVDW7RL2284vwHCQW9dfRG43dzf9BW75VbNM5KmllMW4_kqDK5ksxcfVgmqlh6LtWNzW8B3_Nn4z8sPTW1VXxhw2yZHhgW1Bzvrm3Mb92wW1lbXC01KVmQbW8CBQMp3ZRp8RW4DNzbZ3x4_MbW5hSStX4ZN5sVW6MqDkr6pVNwqW7r5jxd5cJMmpW7DzwPk5Mmw06W9d8lRQ6kxwW_W69lPtx8HGgmRW3nsly683tVwMW1k8_zy5VVrlwW5kP9jc3RSFsWW7vJhc-1gMZ4gW61v-jp4v9-WBW6z9VnV1T-ZmPW7bNswD8CttYSW3hz4-F3jLJ0zW3Z9RVg4rzxM2f66g4jC04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4B45jGjMW50kH_H6lZ3nZW71BsB43zlH_jW3lh-HT4N5xZlW3CPk--2YT9NRW7Xgsc-2pgHQBW2VmVXD1VB3z_W1DQHXY7XTZ_GW8cRlgg58Jdw_W4ynSvj6NPv7tW90XqNd8_FbJ_W5N13Ds6h3b7TN6xdV-hDfM6DW7bVg1j4k9PZVW5s7tWS8tM5WrW8c-Ttx3jTBwGW5MCZBX1CXPjjN6VvSSWHZD6lVwM8DJ4dTDR9W7QhxpQ48-B2HW69p6lt3ZM0xWW64bMZ11Rk8ZrN1Xp_Cwgl1pPW2xb3l51c9QpnW4sNZ1j741bzhW4B7V9G38XqfmN7pLrLkKG0N9W2nKCMX79R-j5W5cZmr81wRcJkW6Wh2Zy9lmmGRW4tJq0R8ZNfrHW8xlHdQ2q3Rx6W1fPkWV288f86W8q9Mfm2F7PyXf8SfwGb04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4CT3l5QzW8wLKSR6lZ3pkW3cYK__3d1v2BV4khpC3g1GgHVcvFnR54YWDkW4d_scM3fh3gZW3VQl8z58568HW8-w3j016XmNZW3RX_2t2Nkq5WW85qkFC334CM0N7gPcdKXH1zPW4-lMsm81Sxk5W6RJMpm5dtzFGW47ZV3D34FDVyW2rF7Kd3zrRNWW6xMXLs8swB0CW793hyR7FqqqdV17Ft68l0Y-ZW4Fqx0L66CD6wW6QXlPh4vw9VmW1lX49N3SB33TW5ltSnb5BY_bYW1B7X896Bk7SgW63zhVH27R1ftW4BNKgB3_nzdWW8TGFc-1JgjPqN1VrH_z__w0zW4pCmdL1rcZFcW6BkdfN6Wvq_gW3cGdBk8NvNK7f1Rk9Nz04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VV_2Zc7Y9mFBW3sYp5R56mw4vW8rPchQ5f275zN8pR4Cz3l5QzW7Y8-PT6lZ3nsW1b8sRm5f2HmlW5qNy1825v3BMV8TMPn24HPv-W7Bt9HH2bB7pYW1pJptY4t8b_5VJD3vF40nJFvW6mLl0l2Q_XbQN8sG_9swzjP2W7TxV_880XLrsW92gdYv3G9y2jN1bYpqT9-mgsW8ySyxD7f8KZDVn9fkV5mMTGPW7lM6zl4BTNV0W2QRb9p369yDJW5LKB406VxW77VK6Xb38z0cb8W6whH_C7hn2KdW94j8Sz8DY_r8W12kGyc92HJfSW7CPbB97thszMW7p1jgy94pvvQVzN_8m8cq7Y9W2L29TY8yQpx7W7-Vhr_7x-YkFW7jZ7Dv8wH0bWf3JPhqC04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-GCW3yWGDZ3LDWCvW4tgmYq3Vw_0KW49HNVn3bvFZmW1N81-_2YcpVJW2y0w1y2PvnMNW3FfNZ72PCflnW3QS6rt2WH8lMW3GSTf434vRG3W3yPpRb3dc35RW3HcTxK2Kx1bbW2MMjm_2Rm6pPW24RRmx3jkFj3W2My48D2RMPYgW4cQjfd4pymzzW1ZqVyy2MWbH-W2p6GWc24WZ8hW4tt7hC30Jbw4W47pbdr2PwBnxW3M7fPN3J-l01W2xRrf83jwHHtW3V-bH32nRs3kW3QFfFc3j4RYqW34xsDm2xJMvCW1R_l_v3NR3yzW1_b90p3Fg7DmW4fvnCB3VMvtQW3gg8473P9d9xW1W-88B1QDfbBW2RtLL41VcdXqW3XX4mx1Z7Z4ZW1V3hwf2nXPQXW3F4-jz2WhmTGW49zv5S23135KW3jf-kY32CnnfW253knr1V7YfnW1Q0SvM3NT6mhW38l4vv41RMb6W2Tz_513btlPVW32hPW647PTSFW2FYyc03bBcWGW2PlXP_3gfmNfW3ZCDsS2sW8bVW4pyM9Q1BH65RW43Gj-42G0gJgW4cv_xs1Lk4tdW3BJz_13K1MsNW2Ydg8Y3NRrGrW1_qH2m327YCtW43VMSm3j9N6cW4rzpbD3jx37gW1QxfRK2Ry565W4cxfnp34BRfff2MCPBc04&_hsenc=p2ANqtz-9GbPDGWfC3Elr7HpXhzARDhOBPI6comIrdsqygy-hDW8LKMZ4K6NixLR7L7xNzbCrW1Q1-TRvGIf1jZU1ifFh1GNetkvG7cOZdkquOM_GLWropqU14vxZ_c7UILceTzJf8EfJbqzbOvLnpkhVJyOfl8HoWAA&_hsmi=86142139 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-GCW3yWGDZ3LDWCvW4tgmYq3Vw_0KW49HNVn3bvFZmW1N81-_2YcpVJW2y0w1y2PvnMNW3FfNZ72PCflnW3QS6rt2WH8lMW3GSTf434vRG3W3yPpRb3dc35RW3HcTxK2Kx1bbW2MMjm_2Rm6pPW24RRmx3jkFj3W2My48D2RMPYgW4cQjfd4pymzzW1ZqVyy2MWbH-W2p6GWc24WZ8hW4tt7hC30Jbw4W47pbdr2PwBnxW3M7fPN3J-l01W2xRrf83jwHHtW3V-bH32nRs3kW3QFfFc3j4RYqW34xsDm2xJMvCW1R_l_v3NR3yzW1_b90p3Fg7DmW4fvnCB3VMvtQW3gg8473P9d9xW1W-88B1QDfbBW2RtLL41VcdXqW3XX4mx1Z7Z4ZW1V3hwf2nXPQXW3F4-jz2WhmTGW49zv5S23135KW3jf-kY32CnnfW253knr1V7YfnW1Q0SvM3NT6mhW38l4vv41RMb6W2Tz_513btlPVW32hPW647PTSFW2FYyc03bBcWGW2PlXP_3gfmNfW3ZCDsS2sW8bVW4pyM9Q1BH65RW43Gj-42G0gJgW4cv_xs1Lk4tdW3BJz_13K1MsNW2Ydg8Y3NRrGrW1_qH2m327YCtW43VMSm3j9N6cW4rzpbD3jx37gW1QxfRK2Ry565W4cxfnp34BRfff2MCPBc04&_hsenc=p2ANqtz-9GbPDGWfC3Elr7HpXhzARDhOBPI6comIrdsqygy-hDW8LKMZ4K6NixLR7L7xNzbCrW1Q1-TRvGIf1jZU1ifFh1GNetkvG7cOZdkquOM_GLWropqU14vxZ_c7UILceTzJf8EfJbqzbOvLnpkhVJyOfl8HoWAA&_hsmi=86142139 )

closed 2024-05-14 by andrewpriddis

andrewpriddis, 2024-05-14:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvBd3l5QzW7Y8-PT6lZ3lBW4dqT8_5YhBSHW8-pXr67D6CZfN3rpfKP1mz3DW58tr_P22ff7DW3js0jg8rxP3dW6g_QJB4V0YFLW6bs-Gd96BnTRW5gv-BY1fZLz7N63CGcJqRJd8W8fXzxm2FXFbgW74hH8Q5RV7jZN2vtxBG41sjHW3NbFCq5qMLQgN5Dk01My6Sl6W3wp2K85MqVPQN7ZpX-x3XlzLW5q39qW5PstnFW7Zqx4t5rrHBNW1-C-8r5bGdChW7QT8BR8nZLQ-W7-xT2P1W_tLnW7xYgqN4Rs8hzW17QCzD4W4fxnW2gygsp1GXqzfW3GNj5M8XJXF-N3cf35W5yBm-f6CTsmg04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvyM5jGjMW50kH_H6lZ3n-W6yHm5g3J0R4gW8p0MqF3hXlW3W7K3VBv2MR0MZW495H1-3ZRVzSN97FVDmhztBxVcdlMr3jypmQW3rW2Fx2dXjmtW8hZJ2733_dvnV7NhPr3nBcpMN8KVW8gzT1xvW94s5c421lfbWN1H6F-_5mbtlW4FDjyh8h7c0DW9jKpyQ1N67jWW71kmbD38wXGNW9khPDP4j5tdNW1NKXWL5-JYQ1N47WG-gttH2FN2ZK_DLwRZSfN7zwf3F1sgGyW2JCq0M6MnsRDW6QdC4020vrVQW5RdS6h7gNfx5Vh9Nc_63bT9jW5DpLV88JcFxhW6N429j1FpWBMN6pKXmDQh1FhVRvYF_6KFWlsW88QBwV2l3JBVW6-__Q83hDQP3W79P6WP2mhyZdW3hV6Db80X2PBf4K68Gg04 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvz27NSbdW5BWSxg6lZ3nKW8TPnj9148JbBW3ZppSm1tF0RgW7Tq_ZJ2zqDkcW8kXBK58MjjYCW2_z9K8373FsKN6q3Yt61N4w3W2cgYfT4Jx02_N5sDKV7LFhTGW8VXt7r8PTbP_W3m1wCK7hXrsRW14Tb531MwySDVCnNk28SdXQ9W2WxSpF3PTnymW3ddpWg3K9k-xW4b_g2d408F3gW7TxDLv6wWnlJW4vLqL_8fYFGrW1rCK123flwfjW1QxnfR7cLdyCW3TB9V27srNY4W8MyCXz7nJhfnW86W60Z2vxVzCW2x_zzQ7p_ndtW3J9QmP6K0d63W3qSN1s6RrhcfN6yw2VGDC5fHW68f8zY5LqrMJW2rwYM-8183ZWN1MysvXyJF1wW5-jL8K6v5mfXW878XwZ2JwdFvVNGq5297v0KnW7P63Kd80J6__W96wZsz5vzxKCN6Tn-5_zF2dxW7rzPd11rSvbbW7ysnX169M-BcW3Kj49c9bSXDqN1dP9k52PHf4W2k5YSj3swJxHW5zgKKL18Rm1fN6wvqF6KMRdJW6VJtlm6b5YTjW1rg9gT8HCTJ0W8R23pb8XjPjZW49PPwz11VZwtW2lcM957Q3W7fW5_czlh6tCdPlW5ScCv61Q50kHW7ynwNw5QgYJCf4yX_Rx04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvyM5jGjMW50kH_H6lZ3kGW8qJrHY31xNnJW2w82LV3k1NZCW3pfn5q3lQnVnW7DQgXW3VvqNnW8CT2Kp1_6qcFW8gm7tB91_X-RW8b2_nX8QJpCbW4_hRjX7Nfl4XW8jjPVs1m3MxzW3FWKBF3MsN62VYRSdx40bb9nW8d_dlk6YY0rPVGcR853Z-PJQW1nYddq4PQ1qrV3S01K9csyD3W5BC-xD1kz9srVmKYS03z_N1nW58bJZG2jP-MKW1FtbtP71qrRHVLTB9r9022PcVJcH3t3kgWT4N5RT6yj3WMGdW8Grc-l81rd6CN5ZX2BWDlW3YW4ykSmz2YL8xFW51jc4F5CFC3QW8q0mMw86g429W4XtcJg1XTkr7Vv13yQ53QB8JV5-Kls4gFY2mW1HMnWt1GKMBVW5HMMSY3JM5R_f7vtFW004 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvBx3l5QzW8wLKSR6lZ3k_W87qMqX4kNzdZW2n5PgX2PnRDFW2ykpd36C9YpmW8YN9SF52gh6WW1_HDbf8SkVPLW6-h23r336J39W8l3hKr8qDpT2W7MPVgk4k654mW3P4zZC7K042rW8jxQTt4B4zFbV8mGTf8FPkt6Vm7vF394B-55N35-r58Ww39yN5jmN4HflTzYW2xFbmK6ygt5ZW75dsHG6p32cjN42FfRHfRlj3Vs0DJ523sNZBW5R-5Qd5vwKx8W1phx7G7tJd53W7D9FjG7tGP7gW2_BrJQ7sZFbNW1LmJNf9jcNBjW939H2Q15k7MNW94Vktj7mtDtwW8ZKN8m661mk-W9jf9DS4Pr8B4W3QDx2f7KnSpxf5ml7GF04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VXj1GG237KJ8W8rbqDR4_JmvxW6rg8CD5f4txCN2QYvBd3l5QzW7Y8-PT6lZ3psN2QHJ071_pvKN4jRy90CnnvbW5xhHJr44W5dYW8fVnnV5fcTPjW3HmyXL6Sk6DZW3VJ4qt5lZ4RMW5hTT625VPChFVsZWQd2n2MPlW8RVFBB5LCcjqW7kGT-Q4hjZXBW2ZMvxW1x-T1dW5yB-DK10Rs04W6QFT0y2qsSFbW4nB3Dt8HYQDwW269V164jGsP9W7VBDyz9gBl6pW4770sD1mC_JhW3b3B2C2XKTcHW6_4M6r7CXKdNW1cNTgj2pBr9NW2x9hFc5QrlKWW1cyWz91Gyg3NW1_qd0h6VdZVNW3D-7_Z4tTyH0W356FgB3nNCtZW2-_bQP3D4y4sf1wfYmT04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-H8W2TBQgJ4fLfndW43PPWg1S8Zd7W2p2xb71_b_lKW237H7R3_KKp3W36hk8P2sQck7W22Wq1L25h1RQW34Cd6z2p53R-W327sJB3P2Wv4W3ZVBk81_cRnJW43p_VY2PLjVVW2zMYbW2Wp6KZW1LmzJl1L6HwDW2q--Dx2YJCjFW1_4lnv3D-QhTW1V1cv94cMnzSW41S3jh3JYK5XW4fyPf_4rq3W2W45mKfJ34pMWxW2xGKNV2FpjsKW3Kb0Cx2HQhcvW4mCrMS1_9jvyW2H-kxf211J-GW3gtkQK3_T-chW3z1HxY3H7KhSW2sTws44hF2N_W3H0c7f3D-h_WW3gcJC638l-ffW3K49Q03FbRftW3SPNXL4fvnTqW2zZPMb4cj7-0W364ZSF2HSjlXW1Y-Nlp32xgKdW2nJngN4hmtv0W32zLb647N_99W38lZ_c3J-jkQW30pph1217SFwW3gnr7n1Q0QGVW43q2qK1SvlNJW3VY7CL3P2X_KW4hJpmw4kCbCRW4mpCQv3gwRkvW3bcdqK1ZtPrfW4ff3pm2HDGp_W3F5m_j4fv0N7W20T6fy1L8d_9W3Hbd_549Dwk1W32ldTl3df-rDW2-nnBN45V-vSW1Z721r4mmCcKW22RGpk234N0nW349LlG4fgh8cW4fsn0j3H3-n_f2nVkym04&_hsenc=p2ANqtz--T5NaxgqD7KrVJQ9sJkmdIzRR610EcPyVRHh9Xus0VZhP5FrNopV6xGkXy8pQ9dpdWjb3sERNmYevJgsF1KYWryg6ame6lKJMf4W3OB187BrGnJXj5zwaxIvrquE2-PiSHH2uctCUiYsGf970QUwc8RUn0rQ&_hsmi=86142139 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-H8W2TBQgJ4fLfndW43PPWg1S8Zd7W2p2xb71_b_lKW237H7R3_KKp3W36hk8P2sQck7W22Wq1L25h1RQW34Cd6z2p53R-W327sJB3P2Wv4W3ZVBk81_cRnJW43p_VY2PLjVVW2zMYbW2Wp6KZW1LmzJl1L6HwDW2q--Dx2YJCjFW1_4lnv3D-QhTW1V1cv94cMnzSW41S3jh3JYK5XW4fyPf_4rq3W2W45mKfJ34pMWxW2xGKNV2FpjsKW3Kb0Cx2HQhcvW4mCrMS1_9jvyW2H-kxf211J-GW3gtkQK3_T-chW3z1HxY3H7KhSW2sTws44hF2N_W3H0c7f3D-h_WW3gcJC638l-ffW3K49Q03FbRftW3SPNXL4fvnTqW2zZPMb4cj7-0W364ZSF2HSjlXW1Y-Nlp32xgKdW2nJngN4hmtv0W32zLb647N_99W38lZ_c3J-jkQW30pph1217SFwW3gnr7n1Q0QGVW43q2qK1SvlNJW3VY7CL3P2X_KW4hJpmw4kCbCRW4mpCQv3gwRkvW3bcdqK1ZtPrfW4ff3pm2HDGp_W3F5m_j4fv0N7W20T6fy1L8d_9W3Hbd_549Dwk1W32ldTl3df-rDW2-nnBN45V-vSW1Z721r4mmCcKW22RGpk234N0nW349LlG4fgh8cW4fsn0j3H3-n_f2nVkym04&_hsenc=p2ANqtz--T5NaxgqD7KrVJQ9sJkmdIzRR610EcPyVRHh9Xus0VZhP5FrNopV6xGkXy8pQ9dpdWjb3sERNmYevJgsF1KYWryg6ame6lKJMf4W3OB187BrGnJXj5zwaxIvrquE2-PiSHH2uctCUiYsGf970QUwc8RUn0rQ&_hsmi=86142139 )

andrewpriddis, 2025-06-30:
Lock In 2024 Pricing - Buy Now & Save!

Hello  ,

We’ve seen a spike in orders this week as more teams move to lock in 2024 pricing. Inventory for popular models like the IMX-5 and GPX-1 is moving quickly — and inventory is expected to become limited.

If you're planning a build or restock before July, I’d recommend getting your order in today!

👉 Place your order now

(https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWGlD8690GfhW422c1_3Xvsm_W3QbRmh5yrx0NN880g4g3l5QzW7Y8-PT6lZ3lXW84dmW014gl4mW70nlKP5Dm3lfVWx1n34rtXTWW8dcKJm2_Hq0mV9MMpL2d_B7GW6N7NR54XFnbTW7P19Nn4xcjq6W1CP5d96TkV7bW7Tq2Vp1xkz4xW3TCfXF1cdCTMW4xj1pg2-53YXW7ryyNw4y9MlbW2lRJwl3c4sysW15QF9W6hDp-nW64Fv208b1Mt2W5WqnJ37HXjdRW8rtw-C4FgzzsW5z4wlY5jSzF0N92VNzZ1W7lKVz1vyh3ykcy3VDS2Qh1SMqTrW3V1LF21KSs6DW3yrFwY7MfP1vW78DkMz6Kfw8JW41979D6tw3KhW6T7Cf17F6nKSf5Dvzrg04 ) 📞 Need help choosing a part? Call me at 801-610–6359

💬 Or chat with me live on the site

Let me know how I can support you!

Best,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWGlD8690GfhW422c1_3Xvsm_W3QbRmh5yrx0NN880g4g3l5QzW7Y8-PT6lZ3nSW4S1mhf7g0gD4W5SnV2z4c-3pRW2Wzwrs2NHDftW4F5Qsv6WDh7mVS23S02qWM0PVvF_Ns9gplSCW5JnDDx7j2cRsW5cD25G4H4mk5VG0TX84CZtVBV2VrtK1CmCXpW95YVnx1xLms6W8R458G8rR32pN5DzWGF-36W0W5P0h7w8qT9JDW8DBRl94TbPwPN3fQmzX7KjX9N5MbMx8SX02jW1wP5vV2r4YycN4K6kmrmxzNsW6n96dc98LRBMW3Kzww_230GrqW8md3rp4Lr4Q6VCvHBp1_Zq73W6CYVSj6P9m8sW4DTjXL3N8yfkW2Q6xws7FvNqHf5zdD1604 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-M1W2PQs2V3g0ljgW2CHby53M90wxW2HsxrK2t5s_1W2sW9sw3DKDS_W3zdxhS2PsKCtW1NvPSd382HtGW2t6fFT3dz7lqW4pnxCk1VrjNcW1V1ckt2y3lnDW45C--C25fMXwW3d9p2b3NQDf0W3CgStK3Y26XVW2CZ91M3LNNK9W2D0XyH45nZT2W2CL4XH2KHmn7W45SWBM3_BYWXW3BQ1ky4cPV8wW1XrTdQ4kfrH5W2RvC4v1Qy4fxW2p9GfJ21rJr1W2r168G25dxmVW4hnpJt1L8J1wW3XX0cX49LLLpW41Fk4x1_5BK7W1X6p-Q258PWlW3XL4r-2xR1hmW1Zvf5f2TvrXpW36Fwbt4th_5-W36CsjG2qWQwXW38Bs7r3XN9vQW2RLWjc3S_HtjW47tnWr24TxK1W4kx6C02qRhVPW41V9953ZHm1qW327YlL2RCVtJW2CVmH63Hc2L1W2FSzk04tlY-HW3H6Z1Z3zk3vTW2PQQWp3QBj1ZW1XqJQZ2MR2KsW2TBn-F1_jWjGW253L8824W8BfW219XjH30sm6NW3gcH-J2MvFGxW1NnXLM1_jqBjW3VRhX81Zww7XW2HVr9w4kt7HxW4kppjy1ZsYs6W214yJD1L5Sy_W3FfQJt23n8L8W3DT33K366g3PW2zP6yR3XHXYRf1SzZ1w04&_hsenc=p2ANqtz-_MSViCNQXy2GE7x1YkszXjf5p3LZo6WPrl5f5DEO8lUbtngx6l_qu2FYkcasmwcyDV2_eP4bZ9YRdklGccH-kglJMdLVk5vscP0WKaysdxPlBLgoajSnNhdU2ehaJrGhd1E3r6d4YogQkxrhyZ5dya4UYO-Q&_hsmi=369255851 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-M1W2PQs2V3g0ljgW2CHby53M90wxW2HsxrK2t5s_1W2sW9sw3DKDS_W3zdxhS2PsKCtW1NvPSd382HtGW2t6fFT3dz7lqW4pnxCk1VrjNcW1V1ckt2y3lnDW45C--C25fMXwW3d9p2b3NQDf0W3CgStK3Y26XVW2CZ91M3LNNK9W2D0XyH45nZT2W2CL4XH2KHmn7W45SWBM3_BYWXW3BQ1ky4cPV8wW1XrTdQ4kfrH5W2RvC4v1Qy4fxW2p9GfJ21rJr1W2r168G25dxmVW4hnpJt1L8J1wW3XX0cX49LLLpW41Fk4x1_5BK7W1X6p-Q258PWlW3XL4r-2xR1hmW1Zvf5f2TvrXpW36Fwbt4th_5-W36CsjG2qWQwXW38Bs7r3XN9vQW2RLWjc3S_HtjW47tnWr24TxK1W4kx6C02qRhVPW41V9953ZHm1qW327YlL2RCVtJW2CVmH63Hc2L1W2FSzk04tlY-HW3H6Z1Z3zk3vTW2PQQWp3QBj1ZW1XqJQZ2MR2KsW2TBn-F1_jWjGW253L8824W8BfW219XjH30sm6NW3gcH-J2MvFGxW1NnXLM1_jqBjW3VRhX81Zww7XW2HVr9w4kt7HxW4kppjy1ZsYs6W214yJD1L5Sy_W3FfQJt23n8L8W3DT33K366g3PW2zP6yR3XHXYRf1SzZ1w04&_hsenc=p2ANqtz-_MSViCNQXy2GE7x1YkszXjf5p3LZo6WPrl5f5DEO8lUbtngx6l_qu2FYkcasmwcyDV2_eP4bZ9YRdklGccH-kglJMdLVk5vscP0WKaysdxPlBLgoajSnNhdU2ehaJrGhd1E3r6d4YogQkxrhyZ5dya4UYO-Q&_hsmi=369255851 )

andrewpriddis, 2025-06-30:
Lock In 2024 Pricing - Buy Now & Save!

Hello  ,

We’ve seen a spike in orders this week as more teams move to lock in 2024 pricing. Inventory for popular models like the IMX-5 and GPX-1 is moving quickly — and inventory is expected to become limited.

If you're planning a build or restock before July, I’d recommend getting your order in today!

👉 Place your order now

(https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVqkh92X4XrDW5mtYW28bVDMTW99f7mw5yrx0TMDyKn63l5QzW7Y8-PT6lZ3ksW69jljj7Nb-wpW4Tcb5T15G3P8W32gql31sLMxDW6HllN-7k8H-4N8_lBVZbnbJKW6K8dvD6DdgZ0N6svJqy2WXhfW7Nc5YQ15pgKMW59dsF01RpvwmW6sb2Fd3Pk6ZFW22K3KJ3ytmt-W4ZYk6K8tFDJTW20T_7F3DJ7gNW7RsQmK6xqZQwW4Klm5N9jR8plW6qTkmX8Fr6ryN7P4JjmfmzxHW6kNjSj6LL-XkN3D5N3j2Hb69VzQ5X41KkdphW1D8Sdy96FKSTW5_Kbdp19SGzmW6f00pQ6x46BXW222hJL1ZxqBFW5k-zdC8FwPMTW4BnnHj8JW62jf2bf-VT04 ) 📞 Need help choosing a part? Call me at 801-610–6359

💬 Or chat with me live on the site

Let me know how I can support you!

Best,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VVqkh92X4XrDW5mtYW28bVDMTW99f7mw5yrx0TMDyKn63l5QzW7Y8-PT6lZ3pPW1KWhLn7m-dHsW2d2dZ93WHXszW4Tgg1m5ph_GvVxSVCm8YfJ4jW82B0p85r8mBwW67p3Lm6mGj7VW2jq6w36HdSBKW5XQ-9G70FpyTW22f_yb8BGKYnW5hKhHZ5jSSGXW3bvnqS5BXKt1W1-ZQ1G8YMSXMW3JDZBS31LkbTW8cQqxl1r32C-Vr80z12hyRycN3DY78-m2pYKW6xx0j22vkcw0W8fp5tD5nFhHkW2G-5q34PFp_VW7ttHd489vdJvW7BnVGc2KbT18W3R1Vh-4QN8nXVDnZG_5PB-QbW28j8D96w3TNPN1746MX3sFYNV6WmHq4vrmPPdTw9Kj04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-BhW2385B149rXV2W2PxVDj32GM_BW4rlMVy38f81GW2PvrHn43YhbhW1QCRW449S8lHW2Yyh-93bt0t4W3GHJ_m1LCtr3W2zxVTl3d1XbqW20YJ1Y45WVYhW3P4bzq3QD3_lW23ncDZ2sPqBxW1Nhzyn2FDXtnW2Rlj3-1_4hMsW2qRhQ72HWbl4W3gnPT-4hywqpW2TdCg-2KTnQDW1Np10J1QxLc-W38D5C04rH-MwW2sRvHQ4r5P7WW2WywQp1N6W4wW3-0bJY36pcv-W1W-z_C3bwsTwW2HSHsk2PS8SZW30dGzh4rHH59W3DSh9C1NDY3KW1V6_Tc1NcmPPW43QdVt4fFJFYW4kHTLP1QhPnBW2PM7Xl1Lg4SLW2Rj7pf47npM2W4t4kt04mpbBrW3R5Kn143wPMMW2FWW1q49D1kZW2Wk1ql2p8m0gW1VfJyN3GHhq2W34h-Xb2WFv1mW47zn3W2vCcBXW4hdppp4mKc46W4kl0Ws2TQ3MNW30kFFy3ghsdyW3F4zTQ21lg_yW2Cw2yC2CLtVGW3HdJXQ36s_x8W3R6XSZ2PzZnlW2-rLPl4tl3ndW1Ndhs_2xJmJnW3N-QN_2RrDrhW3NQ8p62-n0cqW38vQGk41sl25W36sWvZ3LXKzGW36pCWm3JTvjcW1XqjjR3bgBmcf1Z45X304&_hsenc=p2ANqtz-8HxuXdYPb_Ar0rZftXp11s6EA8-hOVGLkeMJc1VcN11FuI5WGzbLjSMGbqZDFUC5aczE1GdH_iHrDs2DDg-RMgB01_qnwALcf3VwS7LFLQlPYpNyi-WEKFAr5m87O3hexVhaHfdH4zi28wGvLcpQTGDKEMHg&_hsmi=369255851 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-BhW2385B149rXV2W2PxVDj32GM_BW4rlMVy38f81GW2PvrHn43YhbhW1QCRW449S8lHW2Yyh-93bt0t4W3GHJ_m1LCtr3W2zxVTl3d1XbqW20YJ1Y45WVYhW3P4bzq3QD3_lW23ncDZ2sPqBxW1Nhzyn2FDXtnW2Rlj3-1_4hMsW2qRhQ72HWbl4W3gnPT-4hywqpW2TdCg-2KTnQDW1Np10J1QxLc-W38D5C04rH-MwW2sRvHQ4r5P7WW2WywQp1N6W4wW3-0bJY36pcv-W1W-z_C3bwsTwW2HSHsk2PS8SZW30dGzh4rHH59W3DSh9C1NDY3KW1V6_Tc1NcmPPW43QdVt4fFJFYW4kHTLP1QhPnBW2PM7Xl1Lg4SLW2Rj7pf47npM2W4t4kt04mpbBrW3R5Kn143wPMMW2FWW1q49D1kZW2Wk1ql2p8m0gW1VfJyN3GHhq2W34h-Xb2WFv1mW47zn3W2vCcBXW4hdppp4mKc46W4kl0Ws2TQ3MNW30kFFy3ghsdyW3F4zTQ21lg_yW2Cw2yC2CLtVGW3HdJXQ36s_x8W3R6XSZ2PzZnlW2-rLPl4tl3ndW1Ndhs_2xJmJnW3N-QN_2RrDrhW3NQ8p62-n0cqW38vQGk41sl25W36sWvZ3LXKzGW36pCWm3JTvjcW1XqjjR3bgBmcf1Z45X304&_hsenc=p2ANqtz-8HxuXdYPb_Ar0rZftXp11s6EA8-hOVGLkeMJc1VcN11FuI5WGzbLjSMGbqZDFUC5aczE1GdH_iHrDs2DDg-RMgB01_qnwALcf3VwS7LFLQlPYpNyi-WEKFAr5m87O3hexVhaHfdH4zi28wGvLcpQTGDKEMHg&_hsmi=369255851 )