inertialsense/inertial-sense-sdk #615: Building issue with catkin_make

andreucm, 2024-04-30:
Hi , 
I'm in Ubuntu 20.04, ROS noetic. 
I follow the steps in the [ros/README](https://github.com/inertialsense/inertial-sense-sdk/tree/main/ros#setup) to build the ROS package. 
It seems that the SDK part is built ok, but an error arises `error: ‘struct p_data_t’ has no member named ‘buf’` when ROS compilation starts:

```
...
 95%] Building C object inertial-sense-sdk/ros/devel/CMakeFiles/InertialSenseSDK.dir/src/libusb/libusb/strerror.c.o
[ 95%] Building C object inertial-sense-sdk/ros/devel/CMakeFiles/InertialSenseSDK.dir/src/libusb/libusb/sync.c.o
[ 96%] Linking CXX static library /home/andreu/dev/workspaces/inertialsense_ws/devel/lib/libInertialSenseSDK.a
[ 96%] Built target InertialSenseSDK
Scanning dependencies of target inertial_sense_ros
[ 97%] Building CXX object inertial-sense-sdk/ros/CMakeFiles/inertial_sense_ros.dir/src/RtkRover.cpp.o
[ 98%] Building CXX object inertial-sense-sdk/ros/CMakeFiles/inertial_sense_ros.dir/src/RtkBase.cpp.o
[ 97%] Building CXX object inertial-sense-sdk/ros/CMakeFiles/inertial_sense_ros.dir/src/ParamHelper.cpp.o
[ 98%] Building CXX object inertial-sense-sdk/ros/CMakeFiles/inertial_sense_ros.dir/src/inertial_sense_ros.cpp.o
In file included from /home/andreu/dev/workspaces/inertialsense_ws/src/inertial-sense-sdk/ros/src/inertial_sense_ros.cpp:19:
/home/andreu/dev/workspaces/inertialsense_ws/src/inertial-sense-sdk/ros/src/inertial_sense_ros.cpp: In lambda function:
/home/andreu/dev/workspaces/inertialsense_ws/src/inertial-sense-sdk/ros/include/inertial_sense_ros.h:79:86: error: ‘struct p_data_t’ has no member named ‘buf’
   79 |                                 this->__cb_fun(DID, reinterpret_cast<__type *>(data->buf)); \
      |                                                                                      ^~~
/home/andreu/dev/workspaces/inertialsense_ws/src/inertial-sense-sdk/ros/include/inertial_sense_ros.h:79:86: note: in definition of macro ‘SET_CALLBACK’
   79 |                                 this->__cb_fun(DID, reinterpret_cast<__type *>(data->buf)); \
...
```

Any idea ?

Thanks!

kylemallory, 2024-12-11:
@andreucm We have recently updated the ROS packages, including adding examples for ROS2.  Can you confirm if this is still an issue?  If not, I'd like to close this issue.

andreucm, 2025-05-13:
Sorry for the big delay on the response.

At commit https://github.com/inertialsense/inertial-sense-sdk/commit/ac70afb5813f1748f32bbef357f8b62cb9b37776, building on Ubuntu 20.04, ROS noetic, the issue no longer appears,  with the following instructions:

git clone https://github.com/inertialsense/inertial-sense-sdk
cd inertial-sense-sdk
git submodule update --init --recursive
cd ../..
rm -rf src/inertial-sense-sdk/ROS/ros2
catkin_make

Hephaestus-Maximus, 2026-10-06:
Thanks for following up in May. Your retest at ac70afb (first included in release **2.4.1**) on Ubuntu 20.04 / ROS Noetic shows the original `struct p_data_t has no member named 'buf'` error no longer occurs. The ROS package was reworked after you reported this, and ROS2 examples were added.

Since this is resolved on current releases, I'm closing it. If you hit the error again on 2.4.1 or later, please reopen it and include the build log.

closed 2026-10-06 by Hephaestus-Maximus