inertialsense/inertial-sense-sdk #618: Zero Motion Command with SDK 2.0.3

Gokru, 2024-05-07:
We are trying to do zero motion when our system is motionless. We have to do that with SDK. When we send the Zero Motion Command and its inverse command as the datasheet says, we can not see the INS_STATUS_STATIONARY_MODE  status bit clear in DID_INS_x.insStatus. Value returns as if no command has been sent. We can do this process with EvalTool but not with SDK. 

Do you have any advice or sample code piece related to this topic?

Thanks.

andrewpriddis, 2024-05-09:
What do you mean when you say you don't see it clear? Do you mean that when you send the command that the insStatus bit INS_STATUS_STATIONARY_MODE is true, but never returns to a false state?

Gokru, 2024-05-10:
There is a flag in "data_sets.h" header file in "eInsStatusFlags" part. It is INS_STATUS_STATIONARY_MODE and its value is (int)0x00002000. When we read datasheet, it says you should see this flag value until the zero motion progress is completed in DID_INS_x.insStatus, then this bit should be clear. We are sending command but can not see any change about this flag. We are able to use other reading and writing operations without any problems, but after sending the zero motion command, we can not figure out whether this operation can be performed.

Gokru, 2024-05-10:
> What do you mean when you say you don't see it clear? Do you mean that when you send the command that the insStatus bit INS_STATUS_STATIONARY_MODE is true, but never returns to a false state?

There is a flag in "data_sets.h" header file in "eInsStatusFlags" part. It is INS_STATUS_STATIONARY_MODE and its value is (int)0x00002000. When we read datasheet, it says you should see this flag value until the zero motion progress is completed in DID_INS_x.insStatus, then this bit should be clear. We are sending command but can not see any change about this flag. We are able to use other reading and writing operations without any problems, but after sending the zero motion command, we can not figure out whether this operation can be performed.

Hephaestus-Maximus, 2026-10-06:
Thanks for the report, and apologies that it went unanswered. We reviewed the current SDK: it sends the zero-motion command and its inverse as documented (via `DID_SYS_CMD`), so we could not identify an SDK-side defect. Whether the `INS_STATUS_STATIONARY_MODE` bit sets or clears is determined by the device firmware, and we have no reproduction of it staying stuck.

This issue has been inactive for a long time, and we have no way to reproduce or confirm a defect from what is here, so we're closing it as stale. If this is still a priority for you, please open a new issue with your SDK and firmware versions, the exact command or code you're using, and any error output or logs, and we'll take a fresh look.

closed 2026-10-06 by Hephaestus-Maximus