inertialsense/inertial-sense-sdk #622: libusb error when buidling cltool and Examples

andrewjlock, 2024-05-17:
I am trying to run some IMX-5 units (and hopefully soon some IG-2-IMX-5s) on Raspberry Pis running Ubuntu 24.04 server.

I can build the parent SDK library successfully. But when I try and build the `cltool` or `ExampleProjects` libraries I get errors relating to `libusb`  such as: 
``` 
undefined reference to `libusb_init'
undefined reference to `libusb_close'
undefined reference to `libusb_exit'
```
As far as I can tell, I have followed the Ubuntu-latest continuous integration workflow exactly, and installed the required dependencies. I plan to run the sensors with ROS , but wanted to check integration with the CL tool first. I have tried in vein to include the `libusb` package in the CL tool `CMakeList.txt` - but I admit I am not particularly experienced in building complex C++ projects.

Do you have any advice how to get this running? Thanks.

andrewpriddis, 2024-05-17:
You need to pull the submodules. libusb is one of them. git submodule update --init --recursive

andrewjlock, 2024-05-17:
Ah, thank you very much for the prompt response. This has resolved it.

I had overlooked submodules, thinking the issue was with the system libusb installation.

closed 2024-05-17 by andrewjlock