inertialsense/inertial-sense-sdk #654: Missing instruction to build ros package

andreucm, 2024-06-17:
In the building instructions at https://github.com/inertialsense/inertial-sense-sdk/tree/main/ros#setup, there is a missing line, just right after the clone:

cd inertial-sense-sdk; git submodule update --init --recursive

otherwise required libs like libusb are not found,  and a linking error appears.

kylemallory, 2024-12-11:
@andreucm Can you follow up on this issue as well, regarding any issues with ROS instructions?

andreucm, 2025-05-13:
Sorry for the big delay on the response. 

At commit ac70afb5813f1748f32bbef357f8b62cb9b37776, to build on Ubuntu 20.04, ROS noetic, the instructions are:
 
```
git clone https://github.com/inertialsense/inertial-sense-sdk
cd inertial-sense-sdk
git submodule update --init --recursive
cd ../..
rm -rf src/inertial-sense-sdk/ROS/ros2
catkin_make
```

Hephaestus-Maximus, 2026-10-06:
Thank you for the report and for the updated steps in your May 2025 retest. The `ros/README` setup instructions were revised in the ROS rework, and at ac70afb (included from release **2.4.1**) you confirmed the build works with the current instructions.

Closing as resolved. If you find a step that is still missing or unclear in the current README, please open a new issue or send a PR against `ros/README.md`.

closed 2026-10-06 by Hephaestus-Maximus