inertialsense/inertial-sense-sdk #655: CMake fix to build cltool on Ubuntu 20.04

andreucm, 2024-06-17:
On Ubuntu 20.04, with 

```
$ uname -a
Linux maupas 5.15.0-107-generic #117~20.04.1-Ubuntu SMP Tue Apr 30 10:35:57 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux
```

I have to fix the [this line](https://github.com/inertialsense/inertial-sense-sdk/blob/255701ee1c44d0427b7588ab415732437230f177/cltool/CMakeLists.txt#L12) of the CMakeLists file, replacin the line by: 

```
link_directories(${IS_SDK_DIR}/build)
```

Doing so , the cltool utility can be built soccessfully, otherwise the following linking error appears: 

```
100%] Linking CXX executable cltool
/usr/bin/ld: cannot find -lInertialSenseSDK
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/cltool.dir/build.make:99: cltool] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/cltool.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

kylemallory, 2024-12-11:
@andreucm Can you confirm if this is still an issue with the latest 2.2.0 release in your build environment?  We are also working on a new 2.3.0 release (`develop` branch) which has significant refactor around CMake and our build scripts, etc.

If this is no longer an issue, please let me know so we can close this issue.

andreucm, 2025-05-13:
Sorry for the big delay on the response.

At commit https://github.com/inertialsense/inertial-sense-sdk/commit/ac70afb5813f1748f32bbef357f8b62cb9b37776, cltool builds ok without the manual fix I did before. 

thanks!

Hephaestus-Maximus, 2026-10-06:
Thanks for the fix suggestion, and for retesting. You confirmed that at ac70afb (included from release **2.4.1**) cltool builds on Ubuntu 20.04 without the manual change to `cltool/CMakeLists.txt`.

Since the build now works as-is, I'm closing this issue. If you run into trouble on that platform again, please open a new issue with your compiler and CMake versions.

closed 2026-10-06 by Hephaestus-Maximus