inertialsense/inertial-sense-sdk #667: Timeout after reboot

k97evin, 2024-06-28:
RUG-3-IMX-5 hardware version 5.0.5
Tested with both ubuntu 20.04 and 22.04 on different machines

After changing firmware and software version of inertial-sense-sdk from version 1.11.2 to 2.1.0:
Connecting to the inertialsense works fine, but after rebooting my pc if I try to connect to it I get a timeout:

sudo ./cltool -c /dev/serial/by-id/usb-STMicroelectronics_STM32_Virtual_ComPort_... -did DID_INS_2

Timeout waiting for response from ports: [/dev/serial/by-id/usb-STMicroelectronics_STM32_Virtual_ComPort_]
ERROR: One or more connected devices are using an incompatible protocol version (requires 2.x.x.x).
 SN0 Hw?-0.0.0 Fw-0.0.0.0 0  2000-00-00 00:00:00 Proto-0.0.0.0 () 

This also sometimes occurs when connecting it through a usb hub without rebooting. 

If unplug and plug the usb then it works again, or if I use the reset button in EvalTool.

This issue does not occur on version 1.11.2.

honzov, 2025-01-27:
Hello @k97evin, have you managed to resolve this issue? I am asking because I have the same issue.

andrewpriddis, 2025-01-27:
@honzov - Can you describe your setup and configuration? If you'd like, you can also send an email to support@inertialsense.com

Hephaestus-Maximus, 2026-10-06:
Thanks for the report. The connection timeout after a host reboot (RUG-3 IMX-5, SDK 2.1.0, cleared by replugging the device) could not be reproduced or traced to a defect in the current SDK, which has changed substantially since 2.1.0.

This issue has been inactive for a long time, and we have no way to reproduce or confirm a defect from what is here, so we're closing it as stale. If this is still a priority for you, please open a new issue with your SDK and firmware versions, the exact command or code you're using, and any error output or logs, and we'll take a fresh look.

closed 2026-10-06 by Hephaestus-Maximus