inertialsense/inertial-sense-sdk #745: RUG 3 IMX 5 DUAL RTK - Ubuntu 20.04 CLTool Firmware Update Issue

C4Consulting, 2024-10-28:
Basically title.

The device is recognized in the EvalTool on windows but firmware updates give an error. I did this just to test the device but ultimately it needs to be on a jetson device running linux.

[image]

[image]

[image]

On  arm64 jetson device running Ubuntu 20,04, CLTool updater does not work either. CLTool has been compiled, etc. but the uploader doesn't work.

[image]

Current device version is 1.8.6

[image]

bboyack, 2024-11-20:
@C4Consulting Where did you go to get the older version of CLTool? I'm having the same issue as you

C4Consulting, 2024-11-20:
Hello

I found it best to test the hardware is working and update firmware on Windows, and then do what you have to do on Linux or other platforms.

To find the latest firmware, use this link: https://github.com/inertialsense/inertial-sense-sdk/releases

In my case, the issue is I have an older version of the IMX5 Dual, so I actually had to use the uINS firmware and not the IMX firmware.

If you know you have the older version, this means, like me you can go up to firmware 1.12 as of me writing this. If you're using the newer version, you're at firmware 2.2. To do the updates, you will have to install the eval tool that comes with the firmware update, and then use the update firmware button. If you have any issues, likely it's easier you upgrade a few firmware versions at a time and not just try and go straight to the latest.

Once you confirm all is working with the latest firmware, then I would install the correct CL Tool on your linux device and try to see if everything is working.

Good luck!

bboyack, 2024-11-20:
I will give that a try. Thank you so much!

Hephaestus-Maximus, 2026-10-06:
Thanks for the report. The firmware update failure on the RUG-3 IMX-5 Dual under Ubuntu 20.04 was only documented in screenshots, and a commenter noted the working path was updating from Windows. We could not trace a defect in the current update code from the information provided.

This issue has been inactive for a long time, and we have no way to reproduce or confirm a defect from what is here, so we're closing it as stale. If this is still a priority for you, please open a new issue with your SDK and firmware versions, the exact command or code you're using, and any error output or logs, and we'll take a fresh look.

closed 2026-10-06 by Hephaestus-Maximus

C4Consulting, 2026-10-06:
> <https://github.com/inertialsense/inertial-sense-sdk/issues/745> as not
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Hello

All good to close nice since got everything working and even created a
driver so that I can use the rug3 on ros2 if you guys happen to want to add
it to your git. Otherwise feel free to close this.

Cheers

*Christopher Allen*
Owner | *  <http://instagram.com/consultc4>
<https://www.linkedin.com/company/consultc4/>*
301.254.5556 | ***@***.***
*Creative Coordinated Consulting Company LLC *

*[image: consultc4.com] <http://consultc4.com>*