inertialsense/inertial-sense-sdk #769: CL Tool not building on Ubuntu 24.04

bboyack, 2024-11-22:
I just cloned the repo and tried to build the CL Tool both using the build_cltool.sh script and manually using cmake. However, when building, I got the following error...

``` 
~/inertial-sense-sdk/scripts $ ./build_cltool.sh 
==========================================
 BUILD:  cltool
==========================================

=== Running make... (Release) ===
/home/yride/inertial-sense-sdk/cltool/build
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yride/inertial-sense-sdk/cltool/build
[  3%] Building CXX object InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderAPP.cpp.o
[  3%] Building CXX object InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderISB.cpp.o
[  4%] Building CXX object InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderDFU.cpp.o
[  3%] Building CXX object InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderBase.cpp.o
In file included from /home/yride/inertial-sense-sdk/src/ISBootloaderDFU.h:24,
                 from /home/yride/inertial-sense-sdk/src/ISBootloaderDFU.cpp:20:
/home/yride/inertial-sense-sdk/src/ISBootloaderBase.h:18:10: fatal error: libusb.h: No such file or directory
   18 | #include "libusb.h"
      |          ^~~~~~~~~~
In file included from /home/yride/inertial-sense-sdk/src/ISBootloaderBase.cpp:13:
/home/yride/inertial-sense-sdk/src/ISBootloaderBase.h:18:10: fatal error: libusb.h: No such file or directory
   18 | #include "libusb.h"
      |          ^~~~~~~~~~
In file included from /home/yride/inertial-sense-sdk/src/ISBootloaderISB.h:24,
                 from /home/yride/inertial-sense-sdk/src/ISBootloaderISB.cpp:21:
/home/yride/inertial-sense-sdk/src/ISBootloaderBase.h:18:10: fatal error: libusb.h: No such file or directory
   18 | #include "libusb.h"
      |          ^~~~~~~~~~
compilation terminated.
compilation terminated.
compilation terminated.
In file included from /home/yride/inertial-sense-sdk/src/ISBootloaderAPP.h:16,
                 from /home/yride/inertial-sense-sdk/src/ISBootloaderAPP.cpp:13:
/home/yride/inertial-sense-sdk/src/ISBootloaderBase.h:18:10: fatal error: libusb.h: No such file or directory
   18 | #include "libusb.h"
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/build.make:258: InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderDFU.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/build.make:244: InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderBase.cpp.o] Error 1
make[2]: *** [InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/build.make:272: InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderISB.cpp.o] Error 1
make[2]: *** [InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/build.make:230: InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/src/ISBootloaderAPP.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:126: InertialSenseSDK/CMakeFiles/InertialSenseSDK.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
[***** BUILD: cltool - FAILED *****]
```

Do I need to install something else to get this to work correctly? 
I built it just fine about a week ago but then switched computers and now it's not building.
Thank you!

andrewpriddis, 2024-11-22:
@bboyack In the SDK there is a directory called scripts. Within that directory there is a bash file called install_sdk_dependencies.sh. Run that. Also you need to pull the git submodules. On the command line use 'git submodule update --init --recursive'.

bboyack, 2024-11-22:
Ok. Thank you. I must have done that before and then forgot that I needed to this time. Thanks again!

closed 2024-11-22 by bboyack