inertialsense/inertial-sense-sdk #782: sdk 2.2.0 is not compatible with OpenCV due to leaking EPS define

honzov, 2024-12-09:
```c++
#include <opencv2/opencv.hpp>
#include <inertial-sense-sdk/data_sets.h>
// #undef EPS

int main() {
    cv::TermCriteria criteria(cv::TermCriteria::MAX_ITER + cv::TermCriteria::EPS, 5000, 0.000001);
    return 0;
}
```

```cmake
cmake_minimum_required(VERSION 3.16)
project(InertialSenseOpenCVExample)

# Set the C++ standard
set(CMAKE_CXX_STANDARD 23)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# Find OpenCV
find_package(OpenCV REQUIRED)

# Include directories for the Inertial Sense SDK
include_directories(/usr/include/inertial-sense-sdk)

# Add the executable
add_executable(main main.cpp)

# Link OpenCV libraries
target_link_libraries(main PRIVATE ${OpenCV_LIBS})
```
Above you can see example where the compilation fails on 
```
 error: expected unqualified-id before ‘(’ token
[build]     6 |     cv::TermCriteria criteria(cv::TermCriteria::MAX_ITER + cv::TermCriteria::EPS, 5000, 0.000001);
```

You can resolve this by including `#undef EPS` right after the sdk include.

kylemallory, 2024-12-10:
@honzov, can you elaborate on this a little more?  We do not define an EPS macro, that I can find (or am aware of).

In the resulting error that you included, the entirety of the statement shows an explicit reference into the cv namespace; If, by chance, we did define an EPS macro, I believe the cv::TermCriteria name scope would address that conflict.

Can you provide additional information about the version of OpenCV, as well as how OpenCV was built and installed, and on which platform you are building for?

honzov, 2024-12-10:
Hi @kylemallory, thanks for getting back to me so quickly! I've found some EPS macro in header file `ISConstants.h`. If I am correct this file is included through `#include <inertial-sense-sdk/data_sets.h>`.
```
ISConstants.h
368:#ifndef EPS
369:#define EPS (1.0E-10f)
```

I use Arch Linux. My environment looks like this.
```
~ uname -a 
Linux host 6.11.2-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 04 Oct 2024 21:51:11 +0000 x86_64 GNU/Linux
~ g++ --version
g++ (GCC) 14.2.1 20240910
~ cmake --version
cmake version 3.30.5
~ yay -Qs opencv
local/opencv 4.10.0-10
```
Hoping this will help you!

kylemallory, 2024-12-10:
Not sure why my search for "EPS" turned up no results!  Sure enough.  Thank you!  I'll let you know what I find.

kylemallory, 2024-12-10:
@honzov Now that I know where I'm looking (thanks again!), it looks like this macro is only used in ISEarth.c to calculate ECEF to LLA (and even then, its not really used by default).  I think we can easily refactor this and update the existing references.  For now, you should be able to safely remove the definition from your local source in ISConstants.h until we get it released.  Let me know if that works for you, of if you run into other issues.

honzov, 2024-12-11:
@kylemallory Thanks! I am currently just calling the undef and that works for me quite well. Till next release I can definitely have it like this. Looking forward to the release!

kylemallory, 2024-12-11:
Perfect.  This has been [resolved in this PR](https://github.com/inertialsense/inertial-sense-sdk/pull/784), and and will be included in the upcoming 2.2.1 release.  Let me know if you have any issues.

closed 2024-12-11 by kylemallory