inertialsense/inertial-sense-sdk #820: inertial-sense on raspberry pi

andermi, 2025-01-30:
My team is having an issue getting the inertial sense to work when connecting to a rpi4 using Raspbian OS. The issue seems to be that there is no driver available in Raspbian OS for the usb to serial device used by the inertial sense. Do you know of a driver for the rpi? Or, is there a way to configure the board to output raw serial somewhere?

MorganMasters, 2025-03-14:
can you give more context to this issue?  I had no troubles from a fresh out-of-the-box RPi 5 with USB communications, using standard installs per these instructions: https://ubuntu.com/download/raspberry-pi

andermi, 2025-04-01:
I was able to solve it... but it took quite a bit of doing on an armv7l. You should have more luck on a 64bit RPi. The main thing did to get it to work was to use the TTL mode of the RUG-3-IMX-5-Dual instead of the built in Serial to USB. I then used a TTL to USB device that used FT232R USB UART driver

Hephaestus-Maximus, 2026-10-06:
Thanks for the update, and for posting how you got this working. Per your last comment, the issue was resolved by using the TTL serial mode of the RUG-3-IMX-5-Dual on the armv7l board; a 64-bit Raspberry Pi OS (as MorganMasters noted) should work with the standard USB driver setup.

Closing as resolved. If you hit further problems, please open a new issue with your OS and hardware details.

closed 2026-10-06 by Hephaestus-Maximus