inertialsense/inertial-sense-sdk #826: inertial_sense_ros.h REPO_VERSION is out of date

RCoder01, 2025-02-05:
The values for `REPO_VERSION_MAJOR`, `REPO_VERSION_MINOR` and `REPO_VERSION_REVIS` still say `2.2.2` when the most recent version is 2.3.2. This causes a false "Protocol version mismatch" error to be logged when the ros node is started up since the node believes its versions is older than it truly is.

andrewpriddis, 2025-02-12:
Unfortunately currently those values are manually updated. Sometimes when a
new release happens, it appears, they get overlooked.
Modify them here:
ROS/catkin_ws/src/inertial-sense-sdk/ROS/ros1/shared/include/inertial_sense_ros.h

[image: Inertial Sense]
<https://cq5Rp04.na1.hs-sales-engage.com/Ctc/RF+23284/cq5Rp04/Jks2-6qcW69sMD-6lZ3n3W7r70xH3lnRqDW2PS81m68BN6DW5jsktF6DH22_W8Kph2P5HC93qW8xXZmZ7fG1g8W1bTY_W3WcLWhVYNkfY7JQTtXN5TY9dMDNWKRV-yG5G5rkBhWW5mPHz37HdCFNW5mTPKM3_FYtxVYZx8W1FPfKZW21n9_08bdqkSW10lmlQ1lsP55W3B7YDG2VMnqzW5VWYlC7JVPsrW87b-zQ4rXk42W5LlvlY5-SCMJN8gMFvd73z1wW1GbKqm5K7Sbcf33zY_P04>
*Andrew Priddis*
*Senior Systems Engineer*
*Director of Sales*
801.610.6359 (Mobile)
***@***.***
*inertialsense.com* <http://www.inertialsense.com/>

andrewpriddis, 2025-02-13:
What does mems precision and calibration look like in 2020?

View in browser (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs505jGjMW7Y9pgv6lZ3kPW1gxc171qSs_WW7ykfL39bjDFwVlT92q832-0MW1Lby267kP36nW5v6kYy4PgjZxVYx6PB8Ct7czW760XN97XYPrMW2NHBd-32sG8NW4fxCWB1_0-rFW55pgh63fjpgfVCYMz51GLfcQW36-FZj2ShbgcW3b1kbC4ypLg-W2t1Gqr4bcQjMW7P8C7214YgNzW2MbQml2Ct_3jW60GHBd8ydXpJW9fmlx11pzVdjW6tJ5BD8gXWxHW2CxxG84f4mGQVS99HD8z53hyW7TfYqm5j-_RyW6jJkX57VV87tW1P512j7ShGPJW1yKZnm3J37pRW5M9Twk31vdVyN3NC_-51VsPMN1znVjl-DWJzW5P44Wj7D9dJ6W1Hh1R25B1J-wDXhL8M_VM7W38Zl0F3DXjp4N36jNvZ6-Yb6W7V3Y7S3x50ysVYsc_X8GRxYJW5Cr5LP2mSfLdW7315D122h5BDW6mPNM-6b1_7XW8nxbCB3vSg2nW94V8NT4jb-2kW2gnlJV4BbVGmW6Rgv5t82TSjGf4mtCtY04 )

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs503l5QzW7Y8-PT6lZ3lfW4VrPst5sbWMSW4mMF7D7pbmZHVvXSRN6m_j33W6zZ-d16B_ghTW9kcBm94k8BhWW5RLtqW4Zn6h0W5kpsvv1FKk-VN3wDVLHlVTnPVS3Qk81Cd0XbW5Xt7FL3nCjgsW7kNW601hGWD5W2z5FY639pqtjW20jBFt2kM2TDW2G9Mdn97Z0CDW5lj4b75T4QwrW4mdsgf4LPL79W7lD6jQ3LsQxbW1N1n_t4sWjtbW5W2LgH5FTHXpW4mP3cT80ZxCmW4w8rxy1Lx1CkW2MkXjz7Y16-6W3hkzwn4cvYLsV6-Gpw1TYpq6VZS-Qd2xNGxXVXc_N28SRgNQf1Z19qW04 )

The Tiny Sensor with Giant Capabilities

What could your device do with this kind of power?

IS Sensor Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs5j3l5QzW8wLKSR6lZ3q0VvR2HS95-2dkW2ymZ-K7bdGd0W7pfwsN7V14d1W4f9nRy7MvRJdW8Snnl349zYsXW1KzgX33tjyG8W3mjWQn8rWbFvN5WC4vLst_yQW39bsQN5ghYjXW3YDLDK57-VJKW4hLd1w7tMz-wW5xwggV1NFVrkW1kX3cP46bY_DW4vQk0P2vqV3hVpzqLQ1pWYZTVx-5H33gzjtVVmQkvj7PbtKVW2d_NY95Ncbb-W975t0M6dN89qW3ny0Z-7bTmWjW27cXtL3-zRWqVwmHYs91jBsTW2jcbgF1p-5DfW223wbZ9f8v6zW2hNXfk6SmTHsW2nGg4f2g7lp9W7dSFMB2BlK_TW26vJ8k4XXphsf2mYR0W04 )

IMU Datasheet

Your device deserves high precision and quick telemetry without being weighed down. Check out our Datasheet and see if our sensors meet your specifications.

Download Datasheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs3R7NSbdW5BWSxg6lZ3mMW4sY-rY21d2hXW5VTwYd2r5Kg1W18krmf6FW6fSVZm0r52Plj2_W7D64wn5tLYmJW7q02W11t_l0VN61J0sH7dZHdW8Q1nmF4y82r8W2-hWlp86xv_XW8hLrl18HNX35W1_5Ncc67BJWHW9hHddR5hcCGNW3LTKYG1wXrpRW7Q1h277VTB1xW4SvLft2dm9-hN4RbgnJ9B88RW1GdVMx8yFmltW33q_Qf4FwPJJW5jTZgw76NRtWN3VPCPfZgT1jW7_m9PN1FTHNXW2sN9FK4DXxVsN64S9RKpWLGkW7Td6b83QxpwhW13VwKK3jgDsDW8PpDPX8mYnzHN2CxYrcfpPqdW6xkmGq1sky7bW7cDjB-93m4xHVQY66g4jyXWsW8j03Nf4j7FH4W5xvd8g5ghz6JN8rLdFrvDlD6W7sGSpD4YsgvtN2ZcZcnL4CLwW84mVR286dbQbW3mnRMw8XSPv3W8vd1-q11SHj5W2JGZb434kpbxW7kq_gJ3hLJCFN8bGTpC8gRwnW3cnfls4Jt1VgW5f3lx71Sv1BSW575-pM4MdV0zW5HFQ0C3K-YHmW5W6hM053l9h-W79MBp-56PvPbW7RrFd750Zh2cN5r8VW6_WD_3W6RS_7c8w1RVHf7ngZrb04 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs3x5jGjMW50kH_H6lZ3mdW3DqVp66Ml0ZHN4Vt9HyvRYcvW3S_f9W8Rr3lKW5lPr2m1JRVY-W2mNjRF3j2W9MV-KMxt84n7SWW4wQcG93DYVFtVTvmdM266HzHW1M1X8c4gXWddN8Mbghj30shsW51s9kr4tZnnJW768xvx20bQSfW5nvzfT645-nFW3qs-W-6jPx7JV6Z-mP6zvyCdW5TyrnR7pqDVmW7FWQ_59609PhW7jLckl8DNDVTW93_-Zl4gyxRDV-6LPv8ky_VhW2LZPC03Dn4f5W66YJYG8mYPQCW3Y7snV3MtqhwW4Z22f08HhbJyW7sFh1j3S11PgW8PYhRG7gYdp3W3Lcy473y3c80W7RHk_37vCy9sW1ppgx75y-Tz-W254xRg13Nj65N1WMq0zFzSKLW3hk__K6tZm9ldsxgRC04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs5j3l5QzW8wLKSR6lZ3p-W4rQWzh2Bs6JJW8yHz9b37N_9bW7kgtj64rL84TW7NGP1G1RNyHLW3mXg7c40pQxPW67rTMW1711QQW3XnV0W17f_4YW1qCSMc7x-vGhW6BTVVz9knRC1W6v2BKr8FKNmCW1tXr6t3sQzsVW8443M28BFV6yW3Vhmnl5ZFq4CW1N6Xnl4w7n6bW5S6tff1T9165W8JV-YK5LwTWzV53mmZ7YHqHxVkHzjm5C9m5HW7F4LGf5gcVxtW8gxZMt5zgSV4W6pzvR26vsNnnW4HfHrt8gRst4W2cYt0r8pdL_2W395D6m8sXT96W6Y1G9f3_9RSLVRk5hF4Ph2XyW4GtmxL7W--RSW3Z4Px73WpMgKf1Lz8n804 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWgF9R2mMKkHW6MspWS7mYJjCW7xsyK95r_ZW_N8hWs503l5QzW7Y8-PT6lZ3mPW7SKk6c2l3-ThW8qpLwv30x1cTW99pYGV7g3ZHMW3GG_Wy13St2mN3_rcSGYHhssW2kxykd6_XzRvW7Q7BD019BSxxW2hYzbQ4YKKy5W4TYFSG8zhlN3W6L39wk8vCXPnW8hJMGH4G3hxTW5-P-gS6Vq-K6W6w4GZ770H0t0VdR4l93DDDjRW3zb91C8RTgsgW3Cydmg2HpvmHVvKnSZ8dqDTzVJdJ-w3YJqM6VJ2DB77hl5f-W7VMg-D45s99DW4DyXbZ8VtQN3N5zrCSyk4c6PW3CY4jt6whqF3W61-gcl3NXFwtW5CsDdx6_RrvJN3rkMnlr_Bbwf7W3XPs04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-zFW1Np1h22126fGW348slC3yL2VjW2vLHsP45qcW3W1LHcZ73GQQDqW2RsVtp2r5DwrW30pnbT23kjGbW3H8-4l3B-bRcW2Wv84t3bhmTNW1VtlcS4kx0DJW3jmJVJ4tx15TW1LrjKJ1XwhX9W2WrcQ12sVmLFW3dbGqs1L5SYcW2MMlFv3jmm0BW4rk1tV3dnBGrW2nVpdM2vZJf9W2s_bR336zmT5W3ST4dm2TgH-ZW2KsgM8234QWLW2C_Hdz4kHSP1W3FgDZ01_9pSzW2RxjJ62KykTqW2Kz7wc43FYrWW2PxX4-2vBp4zW1Y-lxr32J07BW3SLxrp2RzmMtW3P5rKW43V19mW3brLWd3yWb58W2-pBhr2sP0JZW25lNHz49lVRdW4kDXGD1VkrM1W3W3wFP3M3Ck-W4fP8Jz1BnR57W2RjCql2PzzLGW4r9Df24rHcClW385dXV23f7LwW2PW3yt216GDNW2TK42K3_N6vyW4tkLy-3M1wVtW3ZVZYM2vBnbfW4hFvYJ1Nz8XqW1SbB-z2PmkMXW2w3ZBs3SShyGW3N_fdf4pJgS7W3LWs0343lLqQW2YNpz93SzmVQW2Whjl_4mmzskW2WPrkl3bCnT7W2HBYVm2Wnk-dW2MNCJd1Lnl7wW1VtQ-53NQ8cHW4tzF9-3M8yYcf3dr2zt04&_hsenc=p2ANqtz-9bPR5Y9NEx0HVUGu5vkMnNA2jqwZsEOjIF-1s5hWrevjUff-MpKn1ws8Y5zeMJMXmD9WMzZI30RA1xP-LOmhaLcqDiFYed0qkHBaWBz8-Sfn5xW0ClA2ARZ_M5NjFdhisOhkak5wKG4QA3DOANNtNznKko0Q&_hsmi=86142041 ) [... cut ...] 0JZW25lNHz49lVRdW4kDXGD1VkrM1W3W3wFP3M3Ck-W4fP8Jz1BnR57W2RjCql2PzzLGW4r9Df24rHcClW385dXV23f7LwW2PW3yt216GDNW2TK42K3_N6vyW4tkLy-3M1wVtW3ZVZYM2vBnbfW4hFvYJ1Nz8XqW1SbB-z2PmkMXW2w3ZBs3SShyGW3N_fdf4pJgS7W3LWs0343lLqQW2YNpz93SzmVQW2Whjl_4mmzskW2WPrkl3bCnT7W2HBYVm2Wnk-dW2MNCJd1Lnl7wW1VtQ-53NQ8cHW4tzF9-3M8yYcf3dr2zt04&_hsenc=p2ANqtz-9bPR5Y9NEx0HVUGu5vkMnNA2jqwZsEOjIF-1s5hWrevjUff-MpKn1ws8Y5zeMJMXmD9WMzZI30RA1xP-LOmhaLcqDiFYed0qkHBaWBz8-Sfn5xW0ClA2ARZ_M5NjFdhisOhkak5wKG4QA3DOANNtNznKko0Q&_hsmi=86142041 )

andrewpriddis, 2025-02-17:
What does mems precision and calibration look like in 2020?

Inertial Sense (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCGH3l5QzW7Y8-PT6lZ3lRMmzM09D3hbkW8D0NMy7MLnRHW8HNzgD63CgN6W37SM8d7vpTRSN4BcXhLZsZGcW8M__Nm5JkmvnW2MYN6v74dM7kW3fy8bt6kVRJFN3CS129J23fXW2kYb0l68syS7W5lDRk84_8v_SV4LPlr1jj961W3810cs4C3pH4W8Vgmbm3nJWKRV_bywl2P8PhLW30K39_7KZ2gSW6Nx4j38cDJqrN6dbMg5JYQ2WW3knCkn41qKqlW6ydnzl4cglz6W7rqMjk5Q7xwBVnfF3Y200w84N5DVmMpNmqpmW1jdR7G6bcCpnW1dt7rX24DXTrW5zGb1w3ntV8Wf5qX5sH04 )

The Guide to Choosing A Sensor

Which sensor make sense for your project?

Sensor Buying Cheat Sheet (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCFb5jGjMW50kH_H6lZ3kqW45qK-v6vM3pVW8_8MS32MchXmW9j3rxJ49bcBlW6HH1Yq7j7fPYW5pY0WG5YqgxSVw6jcQ3_50Q7W2Ff1pT1YZSbTW6CWVbp3P35wXW6HFmFg2BJx2CW6mMrcs3PTsBSW8JWNYm1t_wzjW98rrGw40G7_pW7H84Cl2Wk8VSW59Bjfq6vHV6PW7bdstB3_jhYNW5GXTzr2tgShMW9lFDf649006MM71yghH-jlKW8H9HbX9fvLlTW59DW2B7lYsVfVQ5Hnl3q_NZ8W54_-VR8qF-svW6L-ZV31n5-BhW5HqTN94RCsC7W3qGSds7xTSz2W8VPTTh455WTsW6KB7Zs3_PH3FW3sT4vR13ddzvW2nXFht26n2yjW71kP9q1Xml_TW7kMJN08rQSNFW21DV753nlH0rf5VBVx404 )

Sensor Buying Cheat Sheet

At this point, you might be wondering where to start when shopping for a sensor for your project. Deciding between an IMU, AHRS, or GNNS & RTK can be difficult. We've made this handy guide to help answer some basic questions quickly.

Read Blog Post (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCFv7NSbdW5BWSxg6lZ3nQW93-Xk34KqWkDW7fSNgs3KGC4-W47LmyC25WdqJVR19zR45mp23W6BXx4B5FFBPLW1z-nW73gNbgdW6mtl9S5N8hcDW2ygbcG9gtN6KW8cTRCL70HR0hW81k0_-4VPfgWW7LWl-P86XVMyW5S351868Vrb4VTH_5g8kRS_GN3rLGS29-LY8W8S981v3Rxrp6W987FFZ1bH4kdW5_PpJl8CpKSqW85g7697TrLkVW1LSCMn7FtMq1W7dGwv45jfq74W7156w337YJNDW2bgm_D7wM2hHW7bK7p64-KM1MW2llmb48rJL60W4zLSp-4Y0d2lW5WRPwg3CnH4kVwYVnt7fqZPrW8DL2b16RZXr1W7V87kY53q_TGN7RqxD2ztH3hW90XQ2_2dsw-fW309TSn6XvDdPW5sCSxd7tNTgHN1W-ZxJm8Mr9W81JJMB50NM6_W1BZ3pr9hdqKMN4zSKy_ScZ5NW1qV3Kv4sCNk5W1QFyTm7Mrw9SW6125Cm4RBW2mW2x9V6_18P6XYVDFvvg97zdKhW7q2gY077JmS2W38g8Zz1k2CSsW7V_T9B2PFdw-N5bL78GTP4yvW5xGmrd3tLvhtW1F53JM7qB3f_N71PGYM-XfW9W490WSG107CPbf1bFkh204 )

"Inertial Sense has developed a disruptive technology. It is going to change everything about our industry."

- AUVSI Judges, 2019

View Products (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCFb5jGjMW50kH_H6lZ3ldN5T1KkQlB7s2W4Bv63971F71PW6FBnjZ6FY20gW7kscqk8b_kj-N6ctGmg8Z0L5W8Rh_hc5JjzKSW5qns2J9hPMfVW5_bHqw6B1N-sVSyntb4JBXY2N3ptwy8lFt8RW4d2sTW77XBZMW6DKRr18JzMjrW6QbKWf5k5F5kW52xxTl6CddDVW5lY_8g6swLDTW1dMd4-4S8BGFW5Qj7bD89FWPRW3Th3T36-GVM1W2RNSVj7zkHjyW8gFKgt4X8l-zW6rjMkm6qC2k2W92fxNT6ggy2bW20KmRb7n39PXW5Sj9-93ZPWvRW2z36-f1pxCR7VJQ56K89F5N3W1TJ0wq6nn3WlW97cPMS2SpML4W9lFMLl19b_RpW8vVH-x2tpKltW6hw1lw8M6RtkN21p5lm1Ng2Yf4zk-HC04 )

Schedule Meeting (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCG-3l5QzW8wLKSR6lZ3m3VnHFJy2t0wD1W6lxjVL96Y9vNW41BmTd6lQxmNW626zN890B04RW904GNQ84MFRfW3wVp919bkJ6XN61FwHlB_WpKW79gKCS7Qmf2MW3w6Cz17G8l9VW4PxNqg36z9xfW16c9pP877zJVW5MHty_3fjpzGW5XD8Pn6TyC06W16B7P01l518DW3SNtGk5xlxcjW484H5p2GGQm_W24lytp99TSLHW3WyL6K5F5JW2W4h9RBC5QzfYsW2v2pxc3rlVCnN8NwdKD92XPqW11Sbsp85lKRpW4ZyGrF4sc9MrW8h-cP55Mdkz7W2RQC7j6HfX49W5gz6rt4nmpW4TT7CX61Tkd_W2QcW_j6wH6Tjf8ZfVFj04 )

Read Blog Posts (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWN5rN69_K7MW57hR4m2-ms-dW8VsNl15s8xVCN3JhCGH3l5QzW7Y8-PT6lZ3lFW23HKF134jSmdN69nZC0tmj5kVCK1FN4ltTMyW44qhV531ClswW2-9N606CYjNVN3CBPlcZYZTpW9dJ_0d5S9p5WN5D9WMYyxk-YW75qyKv3GNqt-V16Kw626r1rVW3YND1B4q_4NZW3LQ1GY2dKnlNW4_YP7z7_9QxlW3T92Kb8Sqp5rN3kKQJmbn0M1VWsq6Q34Qm8dW6F9QnC1d5GdPW9jYWrf660b44W5jZ5rJ5hZdL4W41BmrW3_fW7_W7-tmC38wx2yzW5pXXc25pGff0W50JDzr7-LlJMW8DH0t97DYzzzN8qR__Bw3nQ8W2gVjdb3jCQtdf2f3WXl04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-B8W3bcDdV4tBRDTW2z_rwP36lMkbW21qX_K1_jT0jW1Lg8VH4csZNZW32wSNX1Zq5BnW3VHpYJ2RKdX_W3JGLWz3DWDqdW49mltf3gd2nPW2-rFPS43DJKlW3Y1JcY30lZBXW4r5Q-T3LWrnCW2H-j_61LCpF6W2FXcpY3j8XF5W2r5bTl2vJ9xnW2YfrnL4kf78wW1Vcbrp4kFmVtW2TtbjV3F0rxzW2r1W6w3g94KdW3GQTQg4cwnJhW3yN8vm36wTRtW2MzL0J4mqQ8nW2w2Q0743pzmcW2D01gd3yWCWJW3djPGw36CnQlW2TkJHN43Sk49W3JRYh645DTXXW1QsBlp1Vqpf4W2KqzM02w66QgW386P7K4myBkzW3yZ6LK2TStv-W3z1jgC3JJq3SW43qRtl2PNMY3W2H-Pxm34qHTnW30HMtf2YmKC4W38b_b23DWBzTW1VsCbq3QVZyCW3XPlL21Bq0rgW41Sw003ZvlXPW4kfrLZ2vJ_8YW2-GQQQ2Yn-1zW3gtnmc1StB83W4kCfqX3hZlhDW1Y-PSP49DX-_W4cN-xG3R1xRkW4hPS113ZwzYpW2MCR3C2YNPWXW4rxHBL2TdHzmW2WNd5q2MPMphW3SKdh93byy29W3M91p12-p_Y-W2MV0N63jrTgJW3LJ9dx252ZrJf41yfCD04&_hsenc=p2ANqtz-9grvbUGGwyv9p7l__JcUTU5t8vnIrR08NZ6hlgXQWz8q6TaFEyt1B9e4b3aPkBdr6KGkBZOpGy6nsGH5TJ70MOEENPNk36oKPdCORPdRPs2dhNgKubS0pQgYhUXKYBPEYbdxGcKkk__V_v0FlBVdGs8uXBXw&_hsmi=86142139 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-B8W3bcDdV4tBRDTW2z_rwP36lMkbW21qX_K1_jT0jW1Lg8VH4csZNZW32wSNX1Zq5BnW3VHpYJ2RKdX_W3JGLWz3DWDqdW49mltf3gd2nPW2-rFPS43DJKlW3Y1JcY30lZBXW4r5Q-T3LWrnCW2H-j_61LCpF6W2FXcpY3j8XF5W2r5bTl2vJ9xnW2YfrnL4kf78wW1Vcbrp4kFmVtW2TtbjV3F0rxzW2r1W6w3g94KdW3GQTQg4cwnJhW3yN8vm36wTRtW2MzL0J4mqQ8nW2w2Q0743pzmcW2D01gd3yWCWJW3djPGw36CnQlW2TkJHN43Sk49W3JRYh645DTXXW1QsBlp1Vqpf4W2KqzM02w66QgW386P7K4myBkzW3yZ6LK2TStv-W3z1jgC3JJq3SW43qRtl2PNMY3W2H-Pxm34qHTnW30HMtf2YmKC4W38b_b23DWBzTW1VsCbq3QVZyCW3XPlL21Bq0rgW41Sw003ZvlXPW4kfrLZ2vJ_8YW2-GQQQ2Yn-1zW3gtnmc1StB83W4kCfqX3hZlhDW1Y-PSP49DX-_W4cN-xG3R1xRkW4hPS113ZwzYpW2MCR3C2YNPWXW4rxHBL2TdHzmW2WNd5q2MPMphW3SKdh93byy29W3M91p12-p_Y-W2MV0N63jrTgJW3LJ9dx252ZrJf41yfCD04&_hsenc=p2ANqtz-9grvbUGGwyv9p7l__JcUTU5t8vnIrR08NZ6hlgXQWz8q6TaFEyt1B9e4b3aPkBdr6KGkBZOpGy6nsGH5TJ70MOEENPNk36oKPdCORPdRPs2dhNgKubS0pQgYhUXKYBPEYbdxGcKkk__V_v0FlBVdGs8uXBXw&_hsmi=86142139 )

andrewpriddis, 2025-06-27:
Order now to avoid tariff-induced price hikes in July!

Hello  ,

Hope you’ve been doing well! I wanted to give you a quick heads-up — especially if you're planning any new builds or procurement before Q3.

Due to rising tariffs and component costs, Inertial Sense pricing will increase in July. But if you order before June 30, you can lock in our current 2024 pricing on key products like the IMX-5 and GPX-1.

Whether you're sourcing for:

✔️ Defense UAVs, loitering munitions, or targeting systems

✔️ SATCOM, radar, or LEO antenna platforms

✔️ Or just need rugged, U.S.-made tactical-grade INS/IMU performance —

Now is a great time to stock up or move forward on your project!

📦 Explore Our Top Products:

Have Questions? I’m Happy to Help:

📞 Call me at 801-610–6359

💬 Chat with me (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW8TKK5Pj7d2W8mbq-M4CnncwW3mxgk35yj-cSN2XT75n3prCCW95jsWP6lZ3nyW8gltc_9bHQLWW86_26N5pTHsQV7kZKz4t6RzXW8bq2BK2c9CLcW5TT-7L559YjhW1MKBWf5Z2zLcW2wYgpL7scwp1W1fy1hJ3gkqyfW21kfSM62hvLfW5wKgBy1X4sf8W5YCbF84h99T8W3StB275YRGmzMxwXcY-Rr75VWH-Ky2t31VMVs-DYX47rpR7W79K3L27bwMvpW8vMbLw2SHGgNN3mXnnGvH_dzW1tF6Y37CnzlBW7KYXsD27nzKyVVW0NH7mFW5cW8Rbbms52y04RW8ps4PB7wbLWXW8h8Trl5F2bVQV4VB105VNtWfMbDtx_nGrs7W5ldd1r3RCH1qW4czkgm83j8BqW4WPdvm3kGtNNN7BDttbZ-KKyf3XWjmg04 ) live on our website

👉 Or place your order directly: inertialsense.com/shop (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW8TKK5Pj7d2W8mbq-M4CnncwW3mxgk35yj-cSN2XT73j5m_5PW50kH_H6lZ3lvW5l-sm_5lvQH6W5C31Vz7Z4Xy1VfW1pM46QfzjW3VJLl67DD0vzW1jX60m2KTkN5W1rYmL522M0hgW8lxyhC53tYRnN8-tvqmv22-DW7V3CbF6GyM2lW21TDYb3pBbhbVmhRTY6sFwxqW7csz3v7t7ktYW8_Ffnq28tw4pW4p-7P38ptqPnW52rvBz3d6pXxW8Twq5T7XSV4JW3WcMYH1lcqdxW4M7nbt4tDysqW1snJQb13y38jW6BD5rp7KKhNwW3KVyPt8cs73TVJNtH625T1McW5FCj8m69KQcfW4X1q9X3HfD_9W5s7N-53Y0W-RVZMK9w90lM4rW51M9Ds3xvCJnVrJ5yr7Hwz_KV6n_fs4b-z9FW589_8V2blVGZVZr6Zr3LbJgyW1T0v2J6f2c98f8Y15l804 )

📧 Or just reply to this email — I’ll get back to you personally.

Thanks again for working with us — and if there's anything I can do to support your mission, don’t hesitate to reach out.

Best regards,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VW8TKK5Pj7d2W8mbq-M4CnncwW3mxgk35yj-cSN2XT75n3prCCW95jsWP6lZ3nTW1Pn9fp2395W4W1l0P486_00C6W1tpjBg8zP9CCW7kvCH66xq2zmW1px5QT1qCW7zW906Hmd8JfglVW1M4Rh51s1xpXV7_THX1Fx1WFW6QDbd25pdbzWV8bqGb1KmfWgVyPmM36XHVRwW3MHcNn6BBMvtW7Yrbbd3xTrDHW6R-KSW1HCrwJN7j0L6XBTL51VkyyD423BLfJW35nfZV8k9kMLW2h4FLg4nqzWCN1nsw5kN547kW9hBWQN3Bm00mW66X_MP4Fvm7cW31zzp74KDv7zW3jYBLk1VGFdsW3ZGxvp27XTJ6N6WNkH3mRH_VW8g26zs539cRkW2c3__T7QDRFKVngxHG4XjNVrW26xVLY33656fW68MRHQ8mgzPDdQV-Yd04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-GdW3Cj3k41VwslYW3QK1Ms4hPt0HW3-1-3049J8YZW2KG7bt4m73YNW2TM9vq2zMXwZW43VQX82sW9rQW1Xv2NK2vzX54W3QzWTz1XgNdNW2MpzyF3NFtk_W3W17j02WMmXTW43QCKt2nJlCJW3LP__72nDzj6W3GWrcn3GLgTTW49v0qc2zDX6PW34h66m32gtPnW3XDCpY4tgLkxW1Qj9nN21cJPSW1SBlsH3JYJf5W45DvF-2qC93MW3DKbXZ1L7W-LW3GY2N347Lwc2W4rFtXM3F8Nh7W2Ws3K94tFkMfW34wCJ21Ngn1SW2sxHRh1Lfg8yW1Q6GKs47PTL-W4rH7Fs4cKhSNW2Tr4D44kBpMvW3S-qln4p84s7W3LTkCN1QhnD_W1BKjhW49HPRnW2xWFVv4ff7m3W3BYWwr1Nx5lPW1SkfSr2p0nS3W43VSdH1NxY6rW3dkGWd2MSNJPW1Bc8Z8254Dg5W41Vz8v22WkRJW2CZ-jB4pnzfgW4fpp9f3_tmrFW4rC3h-2q-Yr7W2RmVdx2nVlwRW3jl5824p5BJ-W2CYflp4fvPhGW30l5XN1V7wHyW3bfGNG2RPT50W1YZrJF43VMY5W3gwwsp4czkH-W2CqTBs3_v6hmW4frqWq1NlxwsW3SYdb03BM2P4W2CMN8l1ZcDvZf49LfGh04&_hsenc=p2ANqtz-_8CSfuQ05NncNI6WZPsvraA79afAmj75VmqAHX8DaHD2EEpZpqGiot1qdXnTcZ_24RndNXl0AEsqwmTukDPQw2awBv7FxTBS7C5QYU24YMaI3SB-Cb2S4DNvLq0R5cLiYbh4ykHpm2yPxQpWrYyzBGJEBaYA&_hsmi=368897815 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-GdW3Cj3k41VwslYW3QK1Ms4hPt0HW3-1-3049J8YZW2KG7bt4m73YNW2TM9vq2zMXwZW43VQX82sW9rQW1Xv2NK2vzX54W3QzWTz1XgNdNW2MpzyF3NFtk_W3W17j02WMmXTW43QCKt2nJlCJW3LP__72nDzj6W3GWrcn3GLgTTW49v0qc2zDX6PW34h66m32gtPnW3XDCpY4tgLkxW1Qj9nN21cJPSW1SBlsH3JYJf5W45DvF-2qC93MW3DKbXZ1L7W-LW3GY2N347Lwc2W4rFtXM3F8Nh7W2Ws3K94tFkMfW34wCJ21Ngn1SW2sxHRh1Lfg8yW1Q6GKs47PTL-W4rH7Fs4cKhSNW2Tr4D44kBpMvW3S-qln4p84s7W3LTkCN1QhnD_W1BKjhW49HPRnW2xWFVv4ff7m3W3BYWwr1Nx5lPW1SkfSr2p0nS3W43VSdH1NxY6rW3dkGWd2MSNJPW1Bc8Z8254Dg5W41Vz8v22WkRJW2CZ-jB4pnzfgW4fpp9f3_tmrFW4rC3h-2q-Yr7W2RmVdx2nVlwRW3jl5824p5BJ-W2CYflp4fvPhGW30l5XN1V7wHyW3bfGNG2RPT50W1YZrJF43VMY5W3gwwsp4czkH-W2CqTBs3_v6hmW4frqWq1NlxwsW3SYdb03BM2P4W2CMN8l1ZcDvZf49LfGh04&_hsenc=p2ANqtz-_8CSfuQ05NncNI6WZPsvraA79afAmj75VmqAHX8DaHD2EEpZpqGiot1qdXnTcZ_24RndNXl0AEsqwmTukDPQw2awBv7FxTBS7C5QYU24YMaI3SB-Cb2S4DNvLq0R5cLiYbh4ykHpm2yPxQpWrYyzBGJEBaYA&_hsmi=368897815 )

andrewpriddis, 2025-06-30:
Lock In 2024 Pricing - Buy Now & Save!

Hello  ,

We’ve seen a spike in orders this week as more teams move to lock in 2024 pricing. Inventory for popular models like the IMX-5 and GPX-1 is moving quickly — and inventory is expected to become limited.

If you're planning a build or restock before July, I’d recommend getting your order in today!

👉 Place your order now

(https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWfkK_5H1FZ0W59dMbD1yqfQ-VfQRxk5yrx0sN65wGVW3l5QzW7Y8-PT6lZ3lKW66ZlkT1CxCGkW90jssy1LJrb2W8S5TQl1p8870VgD00M5NF0JjW8zMwr71ZC5TVW99cgmd297DC0W7TNY164v4x4LV1Z6PG8ky0J2W4bX1Ff3hxTbCV-bQ3b1kB-ysW6DXPLN3qJPSSW6TL8Rs7MHypdW43xL_P6QK0qXW40S5z91vKM0GN1LWX8xWq3FbW533xn97bWTdPW1W_4wP5Zzqk7VhlbTy97pVbgW4mq6h9353P-qW6WKYy52MwWR1W7y2RfH4GbwgNW5SYSsY2cs_5-W1Q347k40mtBsN8bYw5k6FBnnN82fSVcgf9bvN7rqjwp3W8KXf5s3pgW04 ) 📞 Need help choosing a part? Call me at 801-610–6359

💬 Or chat with me live on the site

Let me know how I can support you!

Best,

Andrew Priddis | Sr Systems Engineer | Inertial Sense, INC.

Mobile: (801) 610-6359 | InertialSense.com (https://info.inertialsense.com/e3t/Ctc/RF+113/cq5Rp04/VWfkK_5H1FZ0W59dMbD1yqfQ-VfQRxk5yrx0sN65wGVW3l5QzW7Y8-PT6lZ3mQW9fJ-gx6m4y4hW69jBtS4J_JVBVnTp_n4dY-dmW96ltF94HqRYZW4v90fs8Hp4qTW57l75x5mQ_XXW89Q1s-59yjW0W4T2s2s8_-f3VV5574D3k0-4WW8t3c1p5PjCGxW3R-RHW38YpK8W50Fxqv4K3xK6W3dLdXG3f2bH9W3Y-G8h1b35WYN4KpMF7VL-ZfW4-5xW41dsmBkW7n_lDJ59lW-tW451p-N1nbcmHVDc-C-1S8K6VW8wMHT81YH7JmW78Y1BD4gy5cxW6YztMC62pndkW4Q8BsN3t_9g6W6fNHnK31d_KKW4WrNXn2DXf_QW2L6yq74hZmhsf3LygyF04 )

Inertial Sense, 3000 Sierra Vista Way, Provo, UT 84606, USA, 801-610-6771

Unsubscribe (https://info.inertialsense.com/hs/preferences-center/en/direct?data=W2nXS-N30h-S4W4mqptM1_5dcgW4r9b2r2Fr27JW45FLKH3j615XW4plWKG3d88SdW2MWgxY1N2hpfW2CSJDl4knBvSW25000Z2WfFPLW3W0Gjc2CxcStW1Vg_x-3H5Bp0W36vByv34f6MnW3z7ssr3XTZlbW49LhZB3D-Lb9W1L6FY43_XVlDW47SKQC2YNmWpW3GJFh-4rvDgQW3T2ykW1VzPwSW36jBc42qXfdtW3M84b_3P4GzcW2Yfxyf2xBzmSW1LkVYC4kqj6BW1St8GP1V7S3rW4fHmnq328kNBW1QdVCG3M9T9rW1Sxt3-4clMlrW1VqSts2xyV8mW1Skjcp2t78bFW3R2kYl43QKx2W47m2V44pD6KzW1LxkLx3H5bvZW2KrWtp38fZScW3JL1BQ3SJPXQW3ZHmtc23n8p4W4kbFcw3b9zlVW3C7MZ54ckrXvW41RjRD3BMZSnW36FzJm2PSW9WW3SRwlb30HTCRW1L7r8W4kJcxwW2xW_vP4hNYGyW1N55J62Tz7sRW3-1wyV4rlJtfW4cNDcZ3gmZkjW2p5Tlz49wfq3W2CxdLH1NmgsfW41ztW643QDBhW1Bwssy41J9x0W45Y8wL20WzXzW2MBbLv4hpb-lW2p7-h52xy349W23gQyW3VyklFW2KSzmF43zhxGW1Z7t_R43yZDVf3Q_PDr04&_hsenc=p2ANqtz--9sOFLqZHbnikuNPxuP15mNfNQw5frj1VCY8tK-Cm8P6Wji2YNKqB_rt_0lS53Lj3Zrolx7ixEMBPgppuBeoQ77Psok2S9dfjeznT8AQ_ebuQz68e2fFL7eZwinTv4glIBwyjrKTtOJdZzfeJoCHjJcFLp-w&_hsmi=369255851 )
Manage preferences (https://info.inertialsense.com/hs/preferences-center/en/page?data=W2nXS-N30h-S4W4mqptM1_5dcgW4r9b2r2Fr27JW45FLKH3j615XW4plWKG3d88SdW2MWgxY1N2hpfW2CSJDl4knBvSW25000Z2WfFPLW3W0Gjc2CxcStW1Vg_x-3H5Bp0W36vByv34f6MnW3z7ssr3XTZlbW49LhZB3D-Lb9W1L6FY43_XVlDW47SKQC2YNmWpW3GJFh-4rvDgQW3T2ykW1VzPwSW36jBc42qXfdtW3M84b_3P4GzcW2Yfxyf2xBzmSW1LkVYC4kqj6BW1St8GP1V7S3rW4fHmnq328kNBW1QdVCG3M9T9rW1Sxt3-4clMlrW1VqSts2xyV8mW1Skjcp2t78bFW3R2kYl43QKx2W47m2V44pD6KzW1LxkLx3H5bvZW2KrWtp38fZScW3JL1BQ3SJPXQW3ZHmtc23n8p4W4kbFcw3b9zlVW3C7MZ54ckrXvW41RjRD3BMZSnW36FzJm2PSW9WW3SRwlb30HTCRW1L7r8W4kJcxwW2xW_vP4hNYGyW1N55J62Tz7sRW3-1wyV4rlJtfW4cNDcZ3gmZkjW2p5Tlz49wfq3W2CxdLH1NmgsfW41ztW643QDBhW1Bwssy41J9x0W45Y8wL20WzXzW2MBbLv4hpb-lW2p7-h52xy349W23gQyW3VyklFW2KSzmF43zhxGW1Z7t_R43yZDVf3Q_PDr04&_hsenc=p2ANqtz--9sOFLqZHbnikuNPxuP15mNfNQw5frj1VCY8tK-Cm8P6Wji2YNKqB_rt_0lS53Lj3Zrolx7ixEMBPgppuBeoQ77Psok2S9dfjeznT8AQ_ebuQz68e2fFL7eZwinTv4glIBwyjrKTtOJdZzfeJoCHjJcFLp-w&_hsmi=369255851 )