inertialsense/inertial-sense-sdk #835: Ros2 topic /imu does not publish DID_IMU_RAW data as stated in README

bboyack, 2025-03-03:
In the ros2 README it says 
```
- imu(sensor_msgs/msg/Imu)
       -  Raw Imu measurements from IMU1 (NED frame)
 ```

I'm not sure what the difference between IMU1, 2, and 3 is but it makes it sound like it should be the data from DID_IMU_RAW running at a default of 1000 Hz. After running `ros2 topic hz /imu` I found out it was only publishing at 71 Hz, and then after further digging in the code found that it's actually publishing the DID_IMU data, which is after the integrals and derivatives, not the DID_IMU_RAW data.
For reference, I'm using an imx 5.0.4 which is why the DID_IMU is publishing at 71Hz  and not 250Hz like the 5.1 should do. 

If anyone is reading this that doesn't know the difference between DID_IMU and DID_IMU_RAW, I found this section of the inertial sense docs to be very helpfull [https://docs.inertialsense.com/user-manual/application-config/imu_ins_gnss_configuration/#imu-sample-and-navigation-periods](https://docs.inertialsense.com/user-manual/application-config/imu_ins_gnss_configuration/#imu-sample-and-navigation-periods)

Dannykcw, 2025-05-01:
Can you share how your param.yaml and your ros2 run command look? My yaml doesn't seem to affect the parameters that are published.

bboyack, 2025-05-01:
For sure! However, you may need to tweek things to match your setup because I've changed some of the node names and such to make it make more sense to me. 

Not sure if you've done this or not but you need to add the following to the end of your `CMakeLists.txt` file to be able to see the inertial sense node

**CMakeLists.txt**:
```CMakeLists.txt
# Allow ros2 to see the node
install(TARGETS ${PROJECT_NAME}_node
        DESTINATION lib/${PROJECT_NAME})
```

The basic command I use to run it is `ros2 run inertial_sense_ros2 inertial_sense_ros2_node "launch/params.yaml"` (when ran from within the `ros2` symlink directory as created by following the readme). I actually use a script to run it to add a namespace and give me flexibility.

**run_inertial_sense_node.sh**: located in my `src` directory alongside `inertial-sense-sdk` and `ros2`
```shell
#!/bin/bash

# Get the directory where the script is located
SCRIPT_DIR="$(dirname "$(realpath "$0")")"

# Set the relative path to the parameter file
RELATIVE_PATH="ros2/launch/params.yaml"

# Construct the absolute path to the parameter file
PARAMETER_FILE="$SCRIPT_DIR/$RELATIVE_PATH"

# Default namespace
namespace="inertial_sense"

# Default debug setting
debug_logging=""

# Use getopt to parse command-line options
while getopts "dn:" opt; do
  case "$opt" in
    d)
      debug_logging="--log-level debug"
      ;;
    n)
      namespace="$OPTARG"
      ;;
    \?)
      echo "Usage: $0 [-d] [-n <namespace>]" >&2
      exit 1
      ;;
  esac
done

# Remove parsed options from positional parameters
shift $((OPTIND - 1))

# Run the ROS 2 node with the determined namespace, debug logging, and absolute parameter path
ros2 run inertial_sense_ros2 inertial_sense_ros2_node "$PARAMETER_FILE" --ros-args -r __node:=$namespace -r __ns:=/$namespace $debug_logging
```

My params file follows closely to the example they gave. I added a few tweeks to mine, but the key for me to get it to work was figuring out the run command because you don't pass it as a param file to `--ros-args` like typical ros2 paramers. Instead you just pass it as a normal argument as shown above. 
I added `enable_imu_debug` and the `imu_raw` topic so you can ignore those. There are quite a few other things in the example params file, but I haven't used any of those so I don't know if they work or not. There was also a typo with the did_ins1 topic somewhere, but I don't remember where.

**params.yaml**:
```yaml

## My Stuff
enable_imu_debug: false
##

topic: "inertialsense"
port: [/dev/ttyACM0, /dev/ttyACM1, /dev/ttyACM2]
baudrate: 921600
enable_log: false
publishTf: true                                 # Publish Transform Frame (TR)
mag_declination: 0.0
ref_lla: [40.246266148,                           # Default is [0, 0, 0]. Note, this doesn't actually set the reference. Need to use the set_refLLA service to actually set it.
          -111.648571401, 
          1403.8386]

ins:
  rotation: [0, 0, 0]                           # Rotation in radians about the X,Y,Z axes from Sensor Frame to Intermediate Output Frame.  Order applied: Z,Y,X.
  offset: [0, 0 ,0]                             # X,Y,Z offset in meters from Intermediate Output Frame to INS Output Frame.
  navigation_dt_ms: 16                          # EKF update period.  IMX-5:  16 default, 8 max.  Use `msg/ins.../period` to reduce INS output data rate.
  INS output data rate.
  dynamic_model: 8                              # FIXME: these should be named types/srtring (PORTABLE, STATIONARY, AGV, UAV, etc)
  enable_covariance: false                      # Include covariance data in odom_ins_... messages
  messages:
    odom_ins_enu:
      topic: "odom_ins_enu"
      enable: true
    odom_ins_ned:                               # nav_msgs.msg.Odometry
      topic: "odom_ins_ned"
      enable: true
    odom_ins_ecef:
      topic: "odom_ins_ecef"
      enable: true
    did_ins1:  
      topic: "ins_eul_uvw_ned"
      enable: true
    did_ins2:
      topic: "ins_quat_uvw_lla"
      enable: true
    did_ins4:
      topic: "ins_quat_ve_ecef"
      enable: true
    inl2_states:
      topic: "inl2_states"
      enable: true

sensors:
  messages:  
    imu:              # Publish IMU angular rates and linear acceleration
      topic: "imu"
      enable: true
      period: 1
    imu_raw:          # Publish raw IMU data <-- I added this one
      topic: "imu_raw"
      enable: true
      period: 2       # 2 -> 500 Hz
    pimu:             # Publish preintegrated IMU delta theta and delta velocity
      topic: "pimu"
      enable: true
      period: 1
    magnetometer:
      topic: "mag"
      enable: true
      period: 1
    barometer:
      topic: "baro"
      enable: true
      period: 1

gps1:
  type: 'M8'
  antenna_offset: [0.8382, -0.0635, -1.36525]                     # X,Y,Z offset in meters in Sensor Frame to GPS 1 antenna
  gpsTimeUserDelay: 0.0
  messages:
    pos_vel:
      topic: "gps1/pos_vel"
      enable: true
      period: 1
    navsatfix:
      topic: "gps1/NavSatFix"                        # /navsatfix
      enable: true
    info:
      topic: "gps1/info"                        # inertialsense/gps1/info
      enable: true
      period: 5                                 # Publish every 1 second (200ms x 5)

gps2:
  enabled: true
  type: 'M8'
  antenna_offset: [-2.4511, -0.0635, -1.36525]                     # X,Y,Z offset in meters in Sensor Frame to GPS 1 antenna
  gpsTimeUserDelay: 0.0
  messages:
    pos_vel:
      topic: "gps2/pos_vel"
      enable: true
      period: 1
    navsatfix:
      topic: "gps2/NavSatFix"                        # /navsatfix
      enable: true
    info:
      topic: "gps2/info"                        # inertialsense/gps1/info
      enable: true
      period: 5                                 # Publish every 1 second (200ms x 5)

```

If you have any other questions or you want me to share my fork, just let me know. I'm happy to help :)