inertialsense/inertial-sense-sdk #865: Checksum calculation

perara, 2025-04-06:
We are trying to work with the IMU using SPI but are struggling with the checksum.  Seen from the dump below, we seem to get good data from the IMU, here is the ISB. At the end we see 0D 45 which is the checksum. We follow the way done in the docs by doing flecher16 on it except for the  checksum itself.

However, we get invalid checksum. Could anyone pinpoint to what is wrong here and if the data is valid? What i'm not too certain about the 00 00 00 00 which comes after the header (EF 49 04 3A 24 ). This does not align well with the IMU_1 data type...

Please advice

```
[INFO]  --------------------
[DEBUG] Read complete: 121 bytes (DR went LOW)
[VERB]  SPI Read Data (121 bytes):
0000: EF 49 04 3A 24 00 00 00 00 00 BC 66 FF 5A 00 02  | .I.:$......f.Z..
0010: B9 40 00 00 3F 00 95 36 79 39 8A FB 64 39 12 9F  | .@..?..6y9..d9..
0020: 72 3A DC AC 23 C0 0C 02 9C 3F 92 04 17 C1 00 00  | r:..#....?......
0030: 00 00 0D 45 00 00 00 00 00 00 00 00 00 00 00 00  | ...E............
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  | ................
0070: 00 00 00 00 00 00 00 00 00                       | .........
```

andrewpriddis, 2025-04-14:
What SDK version are you using? What firmware version are you using?

fdkz, 2025-05-05:
I had exactly the same strange zeros and checksum errors over SPI (using a custom packet decoder) with fw 2.3.2, fixed with updating the fw to 2.4.1.

[image]

Hephaestus-Maximus, 2026-10-06:
Thanks for the report. A user in this thread (@fdkz) saw the same checksum errors over SPI on firmware 2.3.2 and found they were fixed by updating to firmware **2.4.1**. Please update the IMX to 2.4.1 or later (the current release is 3.1.1) and recheck your checksum calculation.

Closing as resolved on that basis. If you still see invalid checksums on current firmware, please reopen this issue with your SDK and firmware versions and a raw packet dump.

closed 2026-10-06 by Hephaestus-Maximus